nodemcu/nodemcu-firmware #2814: [RFC] LuaRPC

ildar, 2019-06-28:
# Missing feature
RPC mechanism

### Justification
Due to the platform constraints it's sometimes useful to be able to distribute code among the NodeMCU and a Big Brother (e.g. PC), gluing with an RPC of a kind.
AFAIK `node` module 

eLua already has it with LuaRPC port.

### Workarounds
If someone has alternative ideas please share.

### Links

- http://www.eluaproject.net/doc/master/en_refman_gen_rpc.html
- https://github.com/jsnyder/luarpc/

TerryE, 2019-06-28:
@ildar, I am the committer with the focus on the Lua VM.  At the moment my top priorities are adding a Lua 5.3 alternative to 5.1 and converging the esp32 and 8266 variants.  

Whilst eLua was a core part of the initial NodeMCU implementation, and some of this functional additions have been added to the 5.3 lua core.  The rest we either don't use or have reimplement them to get better performacve on platform over the years.  The Lua 5.3 build has no eLua patch content at all.

One of this that we need to reflect is the reality that the vast majority of our app developers are WinX based and are even reluctant to use the command line in windows, so they don't use Lua on the Windows platform. 

Also the NodeMCU architecture is event driven non-blocking just like **Node.js** so we need to be very careful about implementing additional feature sets which cut across our basic architectural assumptions,  so I feel that it very unlikely that we would introduce something like LuaRPC in the near future sorry.

The reality is that something like this isn't going to get implemented and integrated into the project unless I do the work.  I personally use node.js / NodeRED on my SBC platforms to provide upper level control of my ESP devices and at the moment I use a simply JSON-based RPC stack to control them.   Once I've got the lua53 and bulk of the platform convergence work done, then one of the next things on my todo list is to develop a NodeRED module to wrap all of this so that you can seamlessly integrate NodeMCU devices into NodeRED.  Perhaps this might interest you.

ildar, 2019-07-01:
Terry, thanks so much for answering. I got curious: What's that JSON-based
RPC ? Where can I see it?

ildar, 2019-07-09:
it looks like https://github.com/lipp/tango/ is better suited for NodeMCU's
architecture. I'll try and test then report here.

ildar, 2019-07-12:
Ok, version 0.01 is at https://github.com/ildar/tango
It should work but has a long TODO list.
If having issues, please report right there or contact me in gitter.

closed 2020-07-25 by stale[bot]