nodemcu/nodemcu-firmware #2831: Build Problem at LOCK_IN_SECTION Macro

stevemac00, 2019-07-13:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Running MacOS 10.14.5 and Xcode 10.2.1
I have built several previous versions without problem. Today I also had problems with esp-open-sdk.git io I used 
```-b macos-10-14-sdk221 git@github.com:daviddpd/esp-open-sdk.git```
### Actual behavior
build errors
### Test code
```
luac.c:279:8: error: unknown type name 'uint'; did you mean 'int'?
../linit.c:78:7 and ../linit.c:64:14: 
error: argument to 'section' attribute is not valid for this target: mach-o section specifier requires a
      segment and section separated by a comma
static const LOCK_IN_SECTION(libs) luaL_reg LUA_LIBS[] = {
             ^
../../include/module.h:53:63: note: expanded from macro 'LOCK_IN_SECTION'
```
### NodeMCU version
2.2.1

### Hardware
8<------------------------ END BUG REPORT -------------------------------------

TerryE, 2019-10-28:
@stevemac00 Steve, any update on this and workarounds?

stevemac00, 2019-10-29:
Forgot about this. MacOS tends to have older code installed so I created a docker container to do my builds and that works fine. I'll close the issue.

closed 2019-10-29 by stevemac00