nodemcu/nodemcu-firmware #2843: ESP32: compilation error when compiling luac.cross

jpeletier, 2019-07-22:
### Expected behavior
`make` is successful on a clean repo clone.
### Actual behavior
compilation stops when complinig luac.cross. `make` fails with the attached log in commit `eaac369dece9ab74a736e8b42a21a4a655b3fe33`. If I go back to the previous commit, `7cb61a27d20fef3b868e95817089ee71b5b753fc`, `make` is succesful.

Error log:
```
CC build/lua/lzio.o
CC build/lua/lcode.o
CC build/lua/lfunc.o
CC build/lua/ldump.o
AR build/lua/liblua.a
make[2]: Leaving directory '/home/jm/src/nodemcu-firmware-esp32/build/lua'
make[2]: Entering directory '/home/jm/src/nodemcu-firmware-esp32/build/luac_cross'
make[3]: Entering directory '/home/jm/src/nodemcu-firmware-esp32/build/luac_cross'
[  dep] /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../base_nodemcu/linit.c
In file included from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lua.h:18:0,
                 from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../base_nodemcu/linit.c:12:
/home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/luaconf.h:305:10: fatal error: readline/readline.h: No such file or directory
 #include <readline/readline.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[  dep] /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../uzlib/crc32.c
[  dep] /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../uzlib/uzlib_deflate.c
[  dep] /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lzio.c
In file included from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lua.h:18:0,
                 from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lzio.c:12:
/home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/luaconf.h:305:10: fatal error: readline/readline.h: No such file or directory
 #include <readline/readline.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[  dep] /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lvm.c
In file included from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lua.h:18:0,
                 from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lvm.c:12:
/home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/luaconf.h:305:10: fatal error: readline/readline.h: No such file or directory
 #include <readline/readline.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[  dep] /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lundump.c
In file included from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lua.h:18:0,
                 from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lundump.c:11:
/home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/luaconf.h:305:10: fatal error: readline/readline.h: No such file or directory
 #include <readline/readline.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[  dep] /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/ltm.c
In file included from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lua.h:18:0,
                 from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/ltm.c:12:
/home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/luaconf.h:305:10: fatal error: readline/readline.h: No such file or directory
 #include <readline/readline.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[  dep] /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/ltablib.c
In file included from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lua.h:18:0,
                 from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/ltablib.c:12:
/home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/luaconf.h:305:10: fatal error: readline/readline.h: No such file or directory
 #include <readline/readline.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[  dep] /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/ltable.c
In file included from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lua.h:18:0,
                 from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/ltable.c:25:
/home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/luaconf.h:305:10: fatal error: readline/readline.h: No such file or directory
 #include <readline/readline.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[  dep] /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lstrlib.c
In file included from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lua.h:18:0,
                 from /home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/lstrlib.c:12:
/home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/luaconf.h:305:10: fatal error: readline/readline.h: No such file or directory
[... 417 lines cut ...]
/home/jm/src/nodemcu-firmware-esp32/components/luac_cross/../lua/luaconf.h:305:10: fatal error: readline/readline.h: No such file or directory
 #include <readline/readline.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** No rule to make target '/home/jm/src/nodemcu-firmware-esp32/build/luac_cross/luac.d', needed by 'build'.  Stop.
make[3]: Leaving directory '/home/jm/src/nodemcu-firmware-esp32/build/luac_cross'
/home/jm/src/nodemcu-firmware-esp32/components/luac_cross/component.mk:5: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/home/jm/src/nodemcu-firmware-esp32/build/luac_cross'
/home/jm/src/nodemcu-firmware-esp32/sdk/esp32-esp-idf/make/project.mk:552: recipe for target 'component-luac_cross-build' failed
make[1]: *** [component-luac_cross-build] Error 2
make[1]: Leaving directory '/home/jm/src/nodemcu-firmware-esp32'
Makefile:18: recipe for target 'all' failed
make: *** [all] Error 2

```

### Test code

Check out commit, run `make`

My system: Linux Ubuntu 18.04

### NodeMCU version
dev-esp32

chilipeppr, 2019-07-23:
I'm seeing the same error.

chilipeppr, 2019-07-23:
sudo apt-get install libreadline-dev

solved the error for me, but now i'm getting:

Generating esp32.project.ld
/bin/sh: 1: /home/john/nodemcu-firmware-cp2/components/base_nodemcu//add_rodata_ld.sh: not found
make[1]: *** [/home/john/nodemcu-firmware-cp2/components/base_nodemcu/Makefile.projbuild:13: /home/john/nodemcu-firmware-cp2/build/base_nodemcu/ld_patched] Error 127
make[1]: Leaving directory '/home/john/nodemcu-firmware-cp2'
make: *** [Makefile:19: all] Error 2

jpeletier, 2019-07-23:
@chilipeppr This would be a different issue. Do you mind filing a separate issue?
The proposed fix of installing `libreadline-dev` fixes this specific problem.

I am closing this since it is a documentation update what is needed, here is the PR: https://github.com/nodemcu/nodemcu-firmware/pull/2844

closed 2019-07-23 by jpeletier