nodemcu/nodemcu-firmware #2847: enduser_setup not working

HHHartmann, 2019-07-23:
# Expected behavior
the `POST /setwifi` call shpould parse the params

### Actual behavior
1. The end of the wifi_password param is searched to be & or space. In the post it is not anymore but the end of the message
2. if the headers and body are fragmented in several packets only the first one is analyzed and thus data is incomplete. This applies to the GET request as well but likely will nerver hit there.
At least fiddler (a windows web analyzer tool) does send headers and body in two packets.

(1.) works well on params in the URL as used in the GET request. They are terminated by '&' or 'space' 
``` http
GET /url?wifi_ssid=ssid&wifi_password=pass HTTP/1.1
```
The new POST locks loke this:
``` http
POST http://aaa.com/setwifi HTTP/1.1
Host: aaa.com
Content-Type: application/x-www-form-urlencoded
Accept-Encoding: gzip, deflate
Content-Length: 30

wifi_ssid=xxx&wifi_password=aa
```
So the wifi_password is terminated by the end of data here.

### Test code
compile with 
``` c
#define ENDUSER_SETUP_DEBUG_ENABLE 1
#define ENDUSER_SETUP_DEBUG_SHOW_HTTP_REQUEST 1
```
Example from the documentation
```Lua
enduser_setup.start(
  function()
    print("Connected to WiFi as:" .. wifi.sta.getip())
  end,
  function(err, str)
    print("enduser_setup: Err #" .. err .. ": " .. str)
  end,
  print -- Lua print function can serve as the debug callback
)
```

### NodeMCU version
Current dev build with the brand new enduser_setup

### Hardware
ESP8266 dev board

d-rez, 2019-07-24:
Looking at source this is by design as the code is only handling GET:

https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/enduser_setup.c#L947

HHHartmann, 2019-07-24:
Have a look on dev. It's a new design now.

HHHartmann, 2019-07-24:
Will fix 1. but the fragmentation part is above my `C` skills.

marcelstoer, 2019-07-24:
Interesting. Looks like my browser didn't fragment the packets then.

HHHartmann, 2020-04-18:
The missing part of this bug is covered in detail in #2931. So closing this issue

PR #2852 (2019-08-01): Fix enduser_setup default POST request
partially fixes #2847.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

**Rational**
the out of the box enduser_setup does not work as the param parsing fails.
It assumes to have a ' ' or '&' to terminate the params.
While this is true for the GET request (where the params are part of the URL which is followed by "&nb [... cut]
closed 2020-04-18 by HHHartmann