nodemcu/nodemcu-firmware #2849: Advice request for an i2c sniffing module (100kHz

d-rez, 2019-07-24:
8<------------------------ ADVICE REQUEST ------------------------------------
### Overview
I'm trying to have an ESP8266 listen in on i2c comms (100kHz) to "sniff" GPIOs status changes for an [AW9523B LED/GPIO controller](http://www.richwellgroup.com/Private/Files/20181025101009251%E2%88%AEDS_AW9523B_EN_Multi-Function%20LED%20Driver%20And%20GPIO%20Controller%20With%20IIC%20Interface_V1.1.1.pdf). 

I did some research and I know that Lua i2c module falls back to SDK's calls to handle i2c start/stop/read/write/address functions so I'll have to come up with another way. The rest of my code is already in NodeMCU Lua so I'd like to re-use it if possible. I'd like to avoid i2c expanders reading LED voltages directly (dirty + lots of cables + SMD elements aren't helping). 

### Idea
I figured the best way would be to use interrupts (I don't want to block the CPU needlessly as I need it to do other things). I came up with the (incomplete) code below after a few iterations as the simplest one.

Basically: have interrupts trigger on both rising and falling edge on sda and scl. Change global var bits one by one as scl cycles progress, detect start/stop in one interrupt and modify data variable in the other. Process data in a delegated task once readout is complete

### The Problem
Unfortunately Lua (?) is too slow with callbacks back and forth and I'm missing between tens and hundreds (in the previous version where I had interrupts on sda on a falling edge, and on scl on a rising edge) to tens of thousands edges (current version, interrupts on sda+scl on both rising and falling edge), which obviously kill the purpose (verified with the `eventcount `variable)

### Code

```
sda,scl=5,6
data = 0
cur_bit = 0
processing = false
sda_state,scl_state = 0,0

node.setcpufreq(node.CPU160MHZ) -- gotta go fast
gpio.mode(sda, gpio.INT)
gpio.mode(scl, gpio.INT)

function process_data() 
  -- do some processing here
  
  reset_data() -- just a func that resets globals
end

function scl_interrupt(level, when, eventcount)
    scl_state = level
    if (processing) then
        if (sda_state and scl_state) then
          data = bit.set(data,cur_bit)
        end
        cur_bit = cur_bit + 1
    end
end

function sda_interrupt(level, when, eventcount)
    sda_state = level
    if (scl_state) then
        if (not processing and not sda_state) then -- START contition met, SCL still UP
          processing=true
        end
        if (processing and sda_state) then -- stop condition reached, transmission over
          processing=false
          node.task.post(process_data)
        end
    end
end

gpio.trig(sda, "both", sda_interrupt)
gpio.trig(scl, "both", scl_interrupt)
```

### Summary and The Question
As above, it seems to me that my logic is correct but the Lua implementation is just too slow. Would I have luck in trying to dive into [writing my own C module ](https://github.com/nodemcu/nodemcu-firmware/wiki/%5BDRAFT%5D-How-to-write-a-C-module), or should I give up and just try to force my way with an i2c gpio expander and wire to all LEDs separately? :(

Any advice (or a nudge in the right direction) is highly appreciated. Thanks!

8<------------------------ END ADVICE REQUEST --------------------------------

pjsg, 2019-07-24:
I suspect that at 100kHz you will be right on the edge... However, I think that you only need to catch one edge of the SCL line and then just sample the SDA line at that time (in the C code). The issue may be that there will be periods of time that interrupts are disabled, and you will miss everything. Depending on your application, this may or may not be a problem.

d-rez, 2019-07-24:
Thanks for the idea. I didn't really think about just sampling sda on a single scl interrupt given that I had in mind to catch a full start-stop transmission (and as I checked with a logic analyzer, the bus has some bogus transmissions to 0x0 or 0x01 addresses every now and then which I'll be discarding). 

This approach would simplify things further, but require some smarter processing (and probably some timestamp analysis to detect pauses in clock ticks)

The matter of interrupts being disabled sometimes is also something I did not really consider, will try to take it into account

d-rez, 2020-04-08:
I'll be closing this, ended up just resolving my problem in an analog way (checking states ) instead of trying to listen in on what goes on in i2c. i2c was just too fast

Thanks for the advice!

closed 2020-04-08 by d-rez