nodemcu/nodemcu-firmware #2850: Add Lua table preallocation: table.new

BrickBee, 2019-07-24:
### Missing feature
Create a Lua table with an already allocated size instead of growing it incrementally.
`local t = table.new(512,0)  -- Array with 512 slots allocated, using 512 * 16 bytes of memory.`

### Justification
When inserting elements Lua tables double their size whenever they run out of space. This creates overhead and especially memory fragmentation in the already constrained device memory.
Preallocation solves this and also permits creating tables larger than what would be possible later on at run-time due to memory fragmentation.

### Workarounds
You can force Lua to create a table of a certain size by doing local t = {nil, nil, nil, ...} however this blows up the code size and is impractical.

I've patched my ltablib.c to be able to use that feature.
```
static int tnew (lua_State *L) {
  int a = luaL_checkint(L, 1);
  int h = luaL_checkint(L, 2);
  lua_createtable(L, a, h);
  return 1;
}

  {LSTRKEY("new"), LFUNCVAL(tnew)},
```

TerryE, 2019-11-09:
@BrickBee, apologies for a late response. 
> doing local t = {nil, nil, nil, ...} ... blows up the code size and is impractical.

Try doing a `luac -l` to compare this with `t=table.new(3)` and you will see that the former generates a 4 opcode sequence and the latter a 5 opcode sequence, so not only does this take up less space, the opcodes will also execute faster.

The doubling resizing algo is unavoidable in the case of keyed entries, so the only potential use where this would save the excess TValue slots.  But note that the unit TValue size has been 12 bytes not 16 since 2.1.0 IIRC and is now 8 bytes for Lua 5.3.

To use this we would also need to make some changes to the Lua GC as when this sweeps a table it can resize down the entry arrays.  Perhaps a better approach might be to have and `table.trim(t)` which the app could call if it is known that the table won't grow further, but again this would only help with tables will consecutive indexed entries.

BrickBee, 2019-11-17:
I fully agree with being very conservative when it comes to changing the Lua VM. However doing so is not required for the table.new functionality, _if_ the table resizing behavior wasn't changed. The code in the initial request merely makes the already available API function also accessible for the script.

> changes to the Lua GC as when this sweeps a table it can resize down the entry arrays

A preallocated table in vanilla Lua keeps its size during normal insert and delete operations within the allocated capacity. It'll only resize when the limit is exceeded. I've also quickly tested this with NodeMCU 3.0 and didn't see any diverging behavior. Thus to me it doesn't look like the GC needs to be modified.

> compare this with t=table.new(3)

Indeed this doesn't make much sense for small tables. In my case I'm using for bigger tables to buffer some sensor data while wifi isn't available. The TValue 16 to 12 reduction is already a great RAM saver for that by the way.

There are 24 opcodes generated when preallocating a 512 slot array using the "nil" method, as Lua generates 2 opcodes per 50 nil slots. For the hashtable there's one opcode per slot for the first 255 slots. Afterwards there are three opcodes per slot. Thus for "big" tables lots of opcodes are generated, let aside the hassle of writing the script for that.

table.new is a convenient and compact solution here. Of course the slots can also be allocated using a loop. Doing so will cause quite some memory fragmentation though due to the log(N) allocations. This sometimes prevented my scripts from reaching the desired table sizes.

TerryE, 2019-11-17:
```Lua
a=loadstring('return {%s}' % ('0,'):rep(n))()
```
9 opcodes :smile:

BrickBee, 2019-11-18:
That'd be an elegant solution if table.new wasn't available and RAM wasn't very limited. The string creation and loadstring cause memory fragmentation, resulting in less usable RAM.

`local t = {} for i = 1, 10 do t[i] = table.new(512, 0) print(i) end`
This will fit 6 512 slot tables into memory. Doing the same with loadstring instead of table.new only creates 4 tables during my test before the memory is too fragmented.

And yes, independent of that I'm clearly looking forward to 8 byte TValues for even more efficient memory use  :slightly_smiling_face:

nwf, 2020-11-17:
This seems like it might still be a worthwhile improvement if someone wants to make a PR of it.  In the interest of not further diverging from upstream Lua, tho' (however misguided that may be as a goal), I'd suggest that it be in `lnodemcu.c` and `node.new_table()` rather than in the Lua core.

closed 2022-04-30 by stale[bot]