nodemcu/nodemcu-firmware #2853: Can't compile in Docker after PR #2845 and PR #2844

vsky279, 2019-07-25:
### Expected behavior
After PR #2845 and PR #2844 (probably only one of them being toxic) I am not able to compile `dev` version with Docker (Windows 10).

### Actual behavior
```
CC app/user/user_main.c
In file included from user_main.c:25:0:
../../sdk-overrides/include/espconn.h:1:1: error: expected identifier or '(' before '.' token
 ../../app/include/lwip/app/espconn.h
 ^
user_main.c: In function 'user_process_input':
user_main.c:298:32: error: 'input_sig' undeclared (first use in this function)
     return task_post_low(input_sig, force);
                                ^
user_main.c:298:32: note: each undeclared identifier is reported only once for each function it appears in
user_main.c: In function 'user_init':
user_main.c:340:5: error: 'input_sig' undeclared (first use in this function)
     input_sig = task_get_id(handle_input);
     ^
make[2]: *** [.output/eagle/debug/obj/user_main.o] Error 1
make[2]: Leaving directory `/opt/nodemcu-firmware/app/user'
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory `/opt/nodemcu-firmware/app'
make: *** [.subdirs] Error 2
make: Leaving directory `/opt/nodemcu-firmware'
```

### Test code
No test code is needed. I just can't compile clean pull of `dev` version with the error above.

### NodeMCU version
`dev`

### Hardware
N/A

HHHartmann, 2019-07-25:
Can confirm

marcelstoer, 2019-07-25:
> PR #2844

How is that one related? It only changed documentation for the ESP32 branch.

marcelstoer, 2019-07-25:
@HHHartmann since the NodeMCU CI build is obviously ok do you think this may be related to trusty vs xenial (and therefore related to https://github.com/marcelstoer/docker-nodemcu-build/pull/69)?

vsky279, 2019-07-25:
> > PR #2844
> 
> How is that one related? It only changed documentation for the ESP32 branch.

Then it is not related. Of course. 
Sorry I did not go into detail. I just saw that after these two PRs compile fails.

HHHartmann, 2019-07-25:
seems to be that sdk-overrides/include/espconn.h is a link which does not get checked correctly out under windows as mentioned elsewhere. It contains 
```
../../app/include/lwip/app/espconn.h
```

HHHartmann, 2019-07-25:
After copying `app/include/lwip/app/espconn.h` to `sdk-overrides/include/espconn.h` it compiles.

marcelstoer, 2019-07-25:
> not get checked correctly out under windows

Ah, that explains why I cannot reproduce this on macOS.

HHHartmann, 2019-07-26:
Fixed on dev now

marcelstoer, 2019-07-26:
This was a bit of tumbleweed. The actual culprit was https://github.com/nodemcu/nodemcu-firmware/commit/98c2c0520d221d9fbbb2107d7273cd64af3535fd#diff-08924738929fb5bc35be7f4839a1f1c5 from @TerryE merged in #2843. 

Sorry I forgot to close this issue after I merged @jmattsson's fix.

PR #2856 (2019-07-26): Fix espconn.h override on Windows
Fixes #2853.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/*`.
closed 2019-07-26 by marcelstoer