nodemcu/nodemcu-firmware #2864: ADC receiving 1024

beckmx, 2019-07-29:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and **DELETE THE REST**.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Get readings of a voltage devider with GND and 3.3
### Actual behavior
ADC reading shows only 1024
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
if adc.force_init_mode(adc.INIT_ADC)
then
  node.restart()
  return -- don't bother continuing, the restart is scheduled
end
print(adc.read(0))
```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
branch: master
	commit: 68c425c0451f72fcabb1b2d6d31a8555f087371b
	SSL: false
	modules: adc,bit,cron,file,gpio,mdns,net,node,pwm,sjson,spi,tmr,uart,wifi

### Hardware
Describe which ESP8266 device you use and document any special hardware setup
required to reproduce the problem.
ESP12E single module

KT819GM, 2019-08-12:
As your example is working with at least latest master branch (as of 2019 08 12), you should consider checking hardware part.

ildar, 2019-08-12:
BTW note that max voltage shown by ADC is 1.0 V

jmd13391, 2019-08-12:
IMO, this is not a firmware issue nor is it an application developer issue...

- Many / Most ESP8266 development boards come equipped with an on-board 1% tolerance voltage divider connected between the ESP8266 module and the board's ADC input pin (A0). This built-in voltage divider increases the 1.0 volt max of the ESP8266 module's ADC to 3.3 volts... i.e. and adc.read(0) with 3.3 volts connected to the board's pin A0 will return a value of 1023.

- An adc.read(0) value of 1024 indicates a voltage out-of-range condition (i.e. > 3.3v). Remember: there is a +/-1% tolerance with the built-in dev board A0 pin voltage divider (if that's what you're using).

- Use of the ESP8266's built-in ADC should be limited to applications what do not require accurate or stable readings... this is because the ESP8266's ADC is neither. This accuracy / stability issue is the result of an internal design flaw in the ESP8266 chip (it is most noticeable during times of WiFi activity). Consider using an external ADC like the ADS1115 or ADS1015.

beckmx, 2019-08-12:
Actually, after a carefull read, I realized that the nodemcu and wemos board uses internal voltage dividers (resistors), and I created my own module without including those, so, my mistake, after creating my own voltage divider I could see the values

closed 2019-08-12 by beckmx