nodemcu/nodemcu-firmware #2870: redefinition of typedef ‘ROTable’

hsrzq, 2019-08-01:
Building firmware fail

When I run `make all`, this is output:
```
……
HOSTCC /home/zhangqi/projects/nodemcu-firmware/app/lua/luac_cross/luac.c
In file included from ../ltm.h:12,
                 from ../lstate.h:13,
                 from ../ldo.h:12,
                 from luac.c:22:
../lrotable.h:51: error: redefinition of typedef ‘ROTable’
../lobject.h:408: note: previous declaration of ‘ROTable’ was here
make[3]: *** [.output/host/release/obj/luac.o] Error 1
make[3]: Leaving directory `/home/zhangqi/projects/nodemcu-firmware/app/lua/luac_cross'
make[2]: *** [.subdirs] Error 2
make[2]: Leaving directory `/home/zhangqi/projects/nodemcu-firmware/app/lua'
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory `/home/zhangqi/projects/nodemcu-firmware/app'
make: *** [.subdirs] Error 2
make: Leaving directory `/home/zhangqi/projects/nodemcu-firmware'
```

hsrzq, 2019-08-01:
I FIXED it out by commented out `typedef const struct luaR_entry ROTable;` in "app/lua/lobject.h", but is it the right way?

TerryE, 2019-08-01:
@hsrzq Edward, that will work, but we are in the process of tidying all this up as part of preparations to move to Lua5.3 in a few months.  Have browse of the issues and PRs on this.

If you don't mind, I will close this because we are working here anyway.

closed 2019-08-01 by TerryE

hsrzq, 2019-08-02:
@TerryE OK, Thanks A Lot!