nodemcu/nodemcu-firmware #2871: wifi.sta.config() saves configurations randomly to flash

georeb, 2019-08-01:
Using `wifi.sta.config()` users can save up to 5 sets of WiFi credentials to flash for selection at a later date using `wifi.sta.setaplimit()`. However the order in which these 5 credentials are saved is crucial to an application that has an 'SSID connection attempt priority' order.

For example, I want NodeMCU to connect to a config stored in location 1 of flash before it tries the config stored in location 2 (should 1 fail to connect).

I have also noticed that whilst using `wifi.sta.config()` that although the `station_cfg.auto=false` option does indeed NOT connect to that given station, **it does kick you off the currently connected station!**

In addition, writing configs to flash using `wifi.sta.config()` will actually overwrite a station you are currently connected to! Perhaps it should be considered that **active** WiFi credentials should not be overwritten?

Furthermore, to be able to use this config store in the scenario mentioned above, the ability to write an already stored config to a different location. For example SSID A is stored in location 1 but this needed to be stored in location 3. Currently, NodeMCU assumes that you just want to overwrite SSID A with SSID A (hence the store position is not changed).

Even furthermore, I've noticed that a config write will sometimes overwrite an earlier saved config before utilising an empty storage slot!

As far as I can tell, the new configs are written to flash in no particular order. I may be missing something, but the general request here is that there is more control over the storage of WiFi configs if possible please?

I have tried the following code, even trying to execute `wifi.sta.changeap()` (wondering if it had to be used as a pointer) but this made no difference.

#### Add the first WiFi Config:

```Lua
station_cfg={}
station_cfg.ssid="NODE-AAAAAA"
station_cfg.pwd="password"
station_cfg.auto=false
wifi.sta.config(station_cfg)
```

#### Returns the following locations...

```Lua
1
2   NODE-AAAAAA
3
```

#### Add the second WiFi Config:

```Lua
station_cfg={}
station_cfg.ssid="NODE-BBBBBB"
station_cfg.pwd="password"
station_cfg.auto=false
wifi.sta.config(station_cfg)
```

#### Returns the following locations...

```Lua
1
2   NODE-AAAAAA
3   NODE-BBBBBB
```

#### Add the third WiFi Config:

```Lua
station_cfg={}
station_cfg.ssid="NODE-CCCCCC"
station_cfg.pwd="password"
station_cfg.auto=false
wifi.sta.config(station_cfg)
```

#### Returns the following locations...

```Lua
1   NODE-CCCCCC
2   NODE-AAAAAA
3   NODE-BBBBBB
```

### NodeMCU version

Latest MASTER branch

### Hardware

Bare ESP8266-12F

marcelstoer, 2019-08-02:
This is @dnc40085's area of expertise.

You'll find answers to some of your questions around https://github.com/nodemcu/nodemcu-firmware/blob/68c425c0451f72fcabb1b2d6d31a8555f087371b/app/modules/wifi.c#L858 Also, according to the Espressif reference the SDK doesn't allow you to choose the slot into which your configuration is stored: https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf#page=54

georeb, 2019-08-03:
As always, thank you for the clear and concise answer @marcelstoer :)

However, that disconnect call appears to be part of the `wifi.sta.clearconfig()` function, not the `wifi.sta.config()` function. Surely this is the correct one...? (or both are correct I suppose!)

https://github.com/nodemcu/nodemcu-firmware/blob/68c425c0451f72fcabb1b2d6d31a8555f087371b/app/modules/wifi.c#L1091

So I'm assuming that I just have to remove this line from my build?

Also, the Espressif reference mentions a 

> Wi-Fi Station configuration pointer

on page 53...?

Even if there is no way to set the location, maybe the slot allocation can be predicted, or is it purely random? Also the point I mentioned about overwriting configs **before** using blank slots, surely renders this unstable?

@dnc40085 Any advice would be very much appreciated, thanks.

georeb, 2019-08-03:
### FURTHER TESTING:

@dnc40085 @marcelstoer 
At least with limits of 1, 3 and 5 it is predictable now, but not very user friendly! It could be implemented into `wifi.c` but you'd have to read all slots and rewrite them in a particular order to preserve older configs. Limits of 2 and 4 are completely unstable!

Running the following code shows that it does predictably write to flash in a particular order...
(careful not to burn out your flash)
```Lua
wifi.sta.setaplimit(x)
node.restart()
```
then...
```Lua
x=0

checkconfigs = tmr.create()
checkconfigs:register(3000, tmr.ALARM_AUTO, function()

   local station_cfg = {}
   station_cfg.ssid = "myssidvalue_"..x
   station_cfg.pwd = "password"
   station_cfg.save = true
   station_cfg.auto = false
   wifi.sta.config(station_cfg)

   local w = wifi.sta.getapinfo()
   
   for i=1, (w.qty) do
      print(i, w[i].ssid, "\r\n")
   end

   print("\r\n")
   
   if x == 9 then
      x = 0
   else
      x = x+1
   end

end)

checkconfigs:start()
```
#### AP limit set to 2 write order:   2 - 2 - 2 - 2 - 2 - X - 2 - 1 - 1 - 1 - 1
UNSTABLE
#### AP limit set to 3 write order:   1 - 3 - 2 - 1 - 3 - 2
STABLE
#### AP limit set to 4 write order:   2 - 4 - 2 - 4 - 2 - 4 - 2 - X - X - X - 2 - 4
UNSTABLE (appears not to write to 1 or 3 ever!?)
#### AP limit set to 5 write order:   1 - 3 - 5 -2 - 4 - 1 - 3 - 5 -2 - 4
STABLE

X = FAILURE TO WRITE

georeb, 2019-08-10:
Any further thoughts on this @dnc40085 and/or @marcelstoer ?
I am keen to know if there are any plans to amend/solve this issue (if at all possible)? Thanks 🤞

closed 2020-08-19 by stale[bot]