nodemcu/nodemcu-firmware #2873: Fatal Exception 28 (LoadProhibitedCause)

NicolSpies, 2019-08-02:
### Expected behavior
I have been using TLS to successfully connect to the Telegram server over the last month. Using the latest dev build without any code change, I expected to be able to connect as before.
### Actual behavior
After a successful DNS resolve I get the following:
```
HTTP client: DNS request
HTTP client: DNS pending
HTTP client: DNS found api.telegram.org 149.154.167.220
client handshake start.
client handshake failed!
Reason:[-0x7080]
Fatal exception 28(LoadProhibitedCause):
epc1=0x4024d91b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000000c, depc=0x00000000

ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
```
I have increased the ```#define SSL_BUFFER_SIZE 6500``` just to be sure. Has this been experienced before? Maybe it is not related to the TLS handshake. @nwf ,any thoughts and pointers will be much appreciated.
### Test code
```Lua
   net.dns.resolve('https://api.telegram.org', function(sk, ip)
      http.get('https://api.telegram.org/bot724151041:AAHQN8AKdb4DOfjnu7msylNrP14ppnwqQ7s/sendMessage?chat_id=-1001170985013&text='..pl,nil,
        function(code, data)
          print(code, data)
        end)     
    end)
```
### NodeMCU version
```
NodeMCU 3.0.0.0 built with Docker provided by frightanic.com
	branch: dev
	commit: 49ac968bde4d64f73e06c132cf8840932eaf651a
	SSL: true
	Build type: float
	LFS: Size:                   0x20000
	modules: adxl345,bit,cron,crypto,file,gpio,http,i2c,net,node,ow,rtcmem,rtctime,sjson,sntp,struct,tls,tmr,uart,wifi
 build created on 2019-08-01 11:45
 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```
### Hardware
Standard ESP12-F

nwf, 2019-08-02:
0x7080 is `MBEDTLS_ERR_SSL_FEATURE_UNAVAILABLE`, as can be seen in `ssl.h`; you will want to capture a packet trace and/or rebuild with TLS debugging to diagnose what's going on there. 
There have been no TLS configuration changes on our end in some while (January's #2587).  It seems unlikely that telegram has moved their endpoint's configuration outside what we support; perhaps their certificate has just gotten larger and so cannot be processed locally.  Their ServerHello message appears to be `5170` bytes long.

I believe the consensus of the nodemcu developers is that while unfettered TLS access to the Internet would be delightful, we do not believe our beloved ESP8266 is in a position to reliably offer it.  You should proxy connections through a trust-worthy endpoint (whose TLS configuration) you control.  If you trust the local network, no TLS is necessary for the nodemcu<->proxy link; if you do not trust the local network, your proxy should speak with the nodemcu endpoint using TLSv1.2 using ECDSA certificates with short signature chains, perhaps even self-signed.

The load fault will need the attention of `gdb`, I suspect; see `https://nodemcu.readthedocs.io/en/master/modules/gdbstub/`.

NicolSpies, 2019-08-02:
@nwf, thanks for the reply. I have a feeling they their certificate has increased in size. I will try and confirm. 
I agree with your ESP8266 assessment, I dare to ask if you could confirm my feeling based on recent github posts that in the future the Lua effort on the ESP8266 could be available on the ESP32. If so our unfettered TLS access might come true with the extra resources available.

NicolSpies, 2019-08-02:
@nwf , with the risk of doing debugging on the forum, but I am sure this might be valuable to others as well. I do not spot anything strange except the process seems longer than the previous time. The failure seems to be ```ssl_tls.c:3922 mbedtls_ssl_read_record_layer() returned -28800 (-0x7080)``` right at the end.
```
HTTP client: hostname=api.telegram.org
HTTP client: port=443
HTTP client: method=GET
HTTP client: path=/bot724151041:AAHQN8AKdb4DOfjnu7msylNrP14ppnwqQ7s/sendMessage?chat_id=-1001170985013&text=test
HTTP client: DNS request
HTTP client: DNS pending
HTTP client: DNS found api.telegram.org 149.154.167.220
client handshake start.
TLS<2> (heap=10000): ssl_tls.c:6850 => handshake
TLS<2> (heap=10000): ssl_cli.c:3400 client state: 0
TLS<2> (heap=10000): ssl_tls.c:2574 => flush output
TLS<2> (heap=10000): ssl_tls.c:2586 <= flush output
TLS<2> (heap=10000): ssl_cli.c:3400 client state: 1
TLS<2> (heap=10000): ssl_tls.c:2574 => flush output
TLS<2> (heap=10000): ssl_tls.c:2586 <= flush output
TLS<2> (heap=10000): ssl_cli.c:777 => write client hello
TLS<3> (heap=10000): ssl_cli.c:815 client hello, max version: [3:3]
TLS<3> (heap=10000): ssl_cli.c:824 dumping 'client hello, random bytes' (32 bytes)
TLS<3> (heap=10000): ssl_cli.c:824 0000:  3f 11 ee f3 13 d5 76 bd 75 86 81 1e 0c dc 50 97  ?.....v.u.....P.
TLS<3> (heap=10000): ssl_cli.c:824 0010:  08 6e e9 b8 55 d8 6f dc 88 8c 5d 72 7f 3a 1c f9  .n..U.o...]r.:..
TLS<3> (heap=10000): ssl_cli.c:877 client hello, session id len.: 0
TLS<3> (heap=10000): ssl_cli.c:878 dumping 'client hello, session id' (0 bytes)
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c02c
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c030
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: 009f
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c0ad
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c09f
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c024
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c028
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: 006b
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c00a
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c014
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: 0039
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c0af
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c0a3
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c087
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c08b
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c07d
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c073
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c077
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: 00c4
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: 0088
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c02b
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c02f
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: 009e
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c0ac
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c09e
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c023
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c027
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: 0067
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c009
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c013
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: 0033
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c0ae
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c0a2
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c086
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c08a
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c07c
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c072
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: c076
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: 00be
TLS<3> (heap=10000): ssl_cli.c:925 client hello, add ciphersuite: 0045
TLS<3> (heap=10000): ssl_cli.c:937 client hello, got 40 ciphersuites (excluding SCSVs)
TLS<3> (heap=10000): ssl_cli.c:946 adding EMPTY_RENEGOTIATION_INFO_SCSV
TLS<3> (heap=10000): ssl_cli.c:995 client hello, compress len.: 1
TLS<3> (heap=10000): ssl_cli.c:997 client hello, compress alg.: 0
TLS<3> (heap=10000): ssl_cli.c:189 client hello, adding signature_algorithms extension
TLS<3> (heap=10000): ssl_cli.c:274 client hello, adding supported_elliptic_curves extension
[... 117 lines cut ...]
TLS<2> (heap=10000): ssl_tls.c:2516 in_left: 4021, nb_want: 5
TLS<2> (heap=10000): ssl_tls.c:2561 <= fetch input
TLS<3> (heap=10000): ssl_tls.c:3663 input record: msgtype = 22, version = [3:3], msglen = 4096
TLS<2> (heap=10000): ssl_tls.c:2358 => fetch input
TLS<2> (heap=10000): ssl_tls.c:2516 in_left: 4021, nb_want: 4101
TLS<2> (heap=10000): ssl_tls.c:2540 in_left: 4021, nb_want: 4101
TLS<2> (heap=10000): ssl_tls.c:2541 ssl->f_recv(_timeout)() returned 80 (-0xffffffb0)
TLS<2> (heap=10000): ssl_tls.c:2561 <= fetch input
[... cut ...]
TLS<1> (heap=10000): ssl_tls.c:3922 mbedtls_ssl_read_record_layer() returned -28800 (-0x7080)
TLS<1> (heap=10000): ssl_tls.c:4553 mbedtls_ssl_read_record() returned -28800 (-0x7080)
TLS<2> (heap=10000): ssl_tls.c:6860 <= handshake
client handshake failed!
Reason:[-0x7080]
TLS<2> (heap=13784): ssl_tls.c:7623 => free
TLS<2> (heap=34472): ssl_tls.c:7688 <= free
HTTP client: Disconnected with error: 16
HTTP client: Connection timeout
HTTP client: Calling disconnect
HTTP client: manually Calling disconnect callback due to error -12
HTTP client: Disconnected
http_status=-1
code: -1
```

nwf, 2019-08-02:
@NicolSpies... I don't mean to sound mean, but you simply must take time to get a deeper understanding of how TLS and mbedTLS work, especially on nodemcu, if you are going to insist upon connecting to other people's endpoints, as has been, I think fair to say at this point, thematic.  This will require that you be willing to read and understand the entirety of debug logs, some C code, and possibly some sections of RFCs, to understand what is going on, as well as turn knobs both exposed and perhaps not.  We simply cannot support your use cases, especially not gratis, so you must either adopt the recommended workarounds or be willing to support yourself.  I will open a PR to add yet another admonition to the `tls` docs as well as `http` and `mqtt` (and anyone else  using TLS).

I'm being harsh here because the line *immediately prior* to the error you highlight is 
```TLS<1> (heap=10000): ssl_tls.c:3327 TLS handshake fragmentation not supported```,
which should sound plausibly like a reason for an error message called `MBEDTLS_ERR_SSL_FEATURE_UNAVAILABLE`.
This, especially in combination with the line *immediately* above that,
```TLS<3> (heap=10000): ssl_tls.c:3265 handshake message: msglen = 4096, type = 11, hslen = 5165```,
is, perhaps obliquely, telling you precisely what is going on: mbedtls is prepared to operate on messages up to 4096 bytes in size, and has gotten one of 5165.  It is unwilling to deal with fragmented messages.  If you scroll further up, you'll see 
```TLS<3> (heap=10000): ssl_cli.c:453 client hello, adding max_fragment_length extension```.
This means that your build has retained
https://github.com/nodemcu/nodemcu-firmware/blob/4905381c004bdeaf744a60bb940c1906c2e052d4/app/include/user_config.h#L97
which is causing
https://github.com/nodemcu/nodemcu-firmware/blob/4905381c004bdeaf744a60bb940c1906c2e052d4/app/mbedtls/app/espconn_mbedtls.c#L299 to communicate this length limit to the remote endpoint.  Therefore, while you might have adjusted `SSL_BUFFER_SIZE` upwards, you are not really reaping the benefit of the extra heap allocated; you will need to undefine `SSL_MAX_FRAGMENT_LENGTH_CODE` and hope for the best.  (The TLS extensions are unable to express maximum fragment sizes larger than 4096, so absent the extension, most servers will assume that they can send you the full 16KiB the spec mandates.)

TerryE, 2019-08-03:
> I dare to ask if you could confirm my feeling based on recent github posts that in the future the Lua effort on the ESP8266 could be available on the ESP32. If so our unfettered TLS access might come true with the extra resources available.

That's what Johny and I are putting a lot of effort into at the moment and making solid progress.  😊
The esp8266 is still an effective workhorse for a lot of uses, but IMO the combined RAM / flash resources required for a Lua runtime + mbedTLS make it _extremely_ difficult to implement most applications within its resources constraints.

NicolSpies, 2019-08-04:
@nwf, thanks for the pointers, undefining the ```SSL_MAX_FRAGMENT_LENGTH_CODE``` allowed the  handshake message length  of ```hslen = 5165``` to be processed. I take note that this configuration allows servers as per RFC6066 to send 16Kb.

closed 2019-08-20 by NicolSpies