nodemcu/nodemcu-firmware #2875: The new SDK 3.0 exception handler does not interoperate with gdbstub

TerryE, 2019-08-03:
#Expected behaviour

I should be able to debug code with has byte reads from flash in it.  This worked because we added a fix in our exception handler to handle the standard byte and 16bit fetches from flash and to hand off everything else to GDB.  

### Actual behaviour

The new Espressif handler that I introduced with SDK 3.0 doesn't do this so hitting such unligned exceptions confuses the debugger: `Program received signal SIGEMT, Emulation trap.`  It looks like I will need to restore our old handler at lead if we have USE_GDB enabled.  What a PITA.

I've tagget Johny for his info.

nwf, 2019-08-03:
In all seriousness... how much of Espressif's code, esp. binary blob, do we actually use?  It seems like it's... not great code?  Could we get away from as much of it as possible? (*After* the Lua 5.3 thing lands, perhaps as part of figuring out how to run on both the 8266 and the 32... it might be nice to be viable on other platforms, too, and that would also mean detangling ourselves from Espressif.)

TerryE, 2019-08-03:
There exception handler does do stuff like handle unaligned word accesses such as a ..WW WW..` which ours doesn't -- though we don;t have any of these in our code.  I think that they added this to be able to move data into the  iRAM1 segment as data, but being instruction memory, throws the same sort of unaligned exceptions as using Flash for data.  I'll try switching our old handler back in to see if this fixes this issue.

jmattsson, 2019-08-04:
@nwf the entire wifi stack, unfortunately (or well, fortunately. uh, you know what I mean :D )

@TerryE if need be we could enhance our exception handler to support the unaligned word access too (or add a separate one, I don't recall off the top of my head if it's under the same exception code). And if we really want to make it sing, we could pull out the assembler version from the old rtos branch and retrofit it with gdbstub support - no time wasted on thunking to/from C on each exception. Harder to maintain, though (but also not something that should be subject to much change).

TerryE, 2019-08-06:
One side effect of this is that once you've initialised the GDB stub, if you hit an byte fetch from flash you die, so I've removed a couple more of these.

TerryE, 2019-09-17:
>  if you hit an byte fetch from flash you die

It isn't quite as bad as this.  If you step or next over an instruction containing a flash byte fetch then you die, but you can set a HW break following the offending access and execute the instruction(s) in H/W fine.

TerryE, 2019-10-28:
To be honest this has been an indirect advantage in that I've removed all of the unaligned exceptions in main VM paths because of this, and this has speeded up the runtime slightly as a result.

I am probably the only developer who uses the remote gdb enough to be hit by this, and it isn't really stopping me doing any development, so I can't justify spending any time on doing a fix.  So I will give it a week or so for any other contributors to raise a dissenting PoV, then I will close this.

nwf, 2019-10-28:
I'm more than OK with you closing this, but it'd be good if it made it into a list of known caveats of the gdbstub.

TerryE, 2020-07-04:
I been doing deep dive debugging with the elf-gdb tool for many months now:
-  You can set a HW break can continue to this and this work fine.
-  _However_ if you use `next` or `step` to step over an instruction that triggers an unaligned exception then this barfs the gdb stub.

To be honest this one isn't worth fixing, so I am going to close it.

closed 2020-07-04 by TerryE