nodemcu/nodemcu-firmware #2880: gpio.trig does not remove callback

stevemac00, 2019-08-06:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Documentation states "trigger is disabled when [callback] omitted"
### Actual behavior
Interrupt callback is still there. The way I worked around it was to have a dummy function (il1) to switch to but it is not omitted.
### Test code
``Lua
gpio.config({gpio={22},dir=gpio.IN, pull=gpio.PULL_UP})
function il1() end
function il0() print('low') gpio.trig(22,gpio.INTR_LOW) end
gpio.trig(22,gpio.INTR_LOW,il0)
```
### NodeMCU version
dev-esp32
powered by Lua 5.1.4 on ESP-IDF v3.3-beta1-791-g890a341 on ESP-IDF v3.3-beta1-791-g890a341 on SDK IDF

### Hardware
ESP-32
8<------------------------ END BUG REPORT -------------------------------------

joysfera, 2019-08-25:
Just FYI, that is wildly incompatible with the current ESP8266 implementation, where documentation says:

> The previous callback function will be used if the function is omitted.

For disabling the trigger there's a special "none" trigger type.

Couldn't we have the ESP32 gpio module more compatible with the ESP8266 one? Or perhaps it is already implemented so, just the documentation is behind?

stevemac00, 2019-08-25:
I found this by looking at C code.
gpio.INTR_NONE
Yes, the ESP8266 and ESP32 code is quite different in places but I was told the goal was to unite them at some point.

joysfera, 2019-08-25:
@stevemac00 there's a great chance the INTR_NONE will do what you need (and is almost compatible with "none" of NodeMCU for ESP8266).

tomsci, 2020-04-22:
Passing in `gpio.INTR_NONE` does kindasorta work in that it prevents the callback from being called, but does not unref the existing callback (or call `gpio_isr_handler_remove`).

Will open a PR for this (and to update the docs) as I just got hit by this too :)

PR #3072 (2020-04-27): esp32: Unref gpio.trig callbacks when type=INTR_DISABLE (#2880)
Fixes #2880

- [x] This PR is for the `dev` branch rather than for `master`. (_actually dev-esp32_)
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

As per my comment on the issue, this change documents how to use `gpio.INTR_DISABLE` to disable an interrupt trigger, and also ensures that the callback is unrefd (and `g [... cut]
commit f7b8cf01 (2020-04-27): Unref gpio.trig callbacks when type=INTR_DISABLE (#3072)

Fixes #2880
closed 2020-04-27 by marcelstoer