nodemcu/nodemcu-firmware #2883: GPIO.trig callback level not properly masked on dev-esp32

stevemac00, 2019-08-08:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and **DELETE THE REST**.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
interrupt callback parameter level is 0 or 256 instead of 0 or 1
### Actual behavior
high level is 256
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
function intr(pin,level)
  print(pin,level)
  gpio.trig(21,gpio.INTR_LOW,intr)
end
gpio.config({gpio={21},dir=gpio.IN, pull=gpio.PULL_UP})
gpio.trig(21,gpio.INTR_HIGH,intr)

-->21	0
-->21	256
```
### NodeMCU version
powered by Lua 5.1.4 on ESP-IDF v3.3-beta1-791-g890a341 on ESP-IDF v3.3-beta1-791-g890a341 on SDK IDF

### Hardware
ESP-WROOM-32
8<------------------------ END BUG REPORT -------------------------------------

tomsci, 2020-05-03:
Should this have been resolved by the merging of #3086 or is there something else I need to do?

marcelstoer, 2020-05-03:
Thanks @tomsci

PR #3086 (2020-05-02): esp32: Correct level value in gpio.trig() callback (#2883)
Fixes #2883.

- [x] This PR is for the ~dev~ `dev-esp32` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well.
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

This PR changes the `level` parameter passed to the `gpio.trig()` callback so that it matches the documentation, ie `0` or `1` rather than the current behaviour of `0` or `256`. **This [... cut]
closed 2020-05-03 by marcelstoer