nodemcu/nodemcu-firmware #2884: watchdog gets triggered during telnet transfer

stevemac00, 2019-08-09:
8<------------------------ BUG REPORT -----------------------------------------
I frequently get watchdog reboots when uploading via telnet using luatool.py. It works fine if I only send a small file but it reboots if I send "too many". I don't have any code to reproduce. It sounds similar to #2690. I can go all day if I don't "hammer" it with files but it fails as soon as I do things in a loop from luatool. I'm sorry I can't offer a repeatable problem and I ignored it the first few times but it's happened many times now. The same code works fine on an ESP8266.

Following is a dump.
```
[0;32mI (183359) event: sta ip: 172.27.35.6, mask: 255.255.255.192, gw: 172.27.35.1[0m
172.27.35.6	fe:ec:da:1d:4d:5b
[0;31mE (204949) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (204949) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (204949) task_wdt: Tasks currently running:[0m
[0;31mE (204949) task_wdt: CPU 0: main[0m
[0;31mE (204949) task_wdt: CPU 1: IDLE1[0m
[0;31mE (204949) task_wdt: Aborting.[0m
abort() was called at PC 0x400d2ec0 on core 0
```
### NodeMCU version
NodeMCU ESP32 built with Docker provided by frightanic.com
	branch: dev-esp32
	commit: 7ad6bc1be779b31fc9a3d8577a6273b1fd7f6f17
	SSL: true
	modules: adc,bit,dht,file,gpio,http,mqtt,net,node,sjson,tmr,wifi,time
 build created on 2019-08-04 10:37
 powered by Lua 5.1.4 on ESP-IDF v3.3-beta1-791-g890a341 on SDK IDF

### Hardware
ESP32 WROOM

8<------------------------ END BUG REPORT -------------------------------------

stevemac00, 2019-08-09:
I'm trying to dig deeper. There's something going on because there's a delay in watchdog abort. I'm trying to see output by changing from 0 to node.output(s_output,1) but I still don't see anything in terminal.

stevemac00, 2019-08-09:
Think I found the problem. In the luatool.py version I'm using there is a TCP transport and it only had a socket.close(). I changed it to:
```
self.socket.shutdown(socket.SHUT_RDWR)
self.socket.close()
```
and so far it's working. For an unknown reason this always worked on the ESP8266 but caused a delayed watchdog abort on ESP32.

closed 2019-08-09 by stevemac00

marcelstoer, 2019-08-10:
The good thing: that tool is open-source. The bad thing: it seems unmaintained and a PR to that effect may never get merged.

nwf, 2019-08-10:
> The good thing: that tool is open-source. The bad thing: it seems unmaintained and a PR to that effect may never get merged.

If it's a useful tool, perhaps we should take on its maintenance or at least include a fork in our repo?

stevemac00, 2019-08-10:
I've been using the tool since beginning and I have it modified with many options. It has conditional transfer based on chipid and a json file that keeps track of files and if they've changed since upload.

I haven't found anything I like better.
```
usage: luatool.py [-h] [-p PORT] [-b BAUD] [-f SRC] [-t DEST] [-c] [-d] [-v]
                  [-l] [-i] [-e] [-r] [-w] [-n] [--delay DELAY]
                  [--delete DELETE] [--ip IP] [--chkmissing]

ESP8266 Lua script uploader.

optional arguments:
  -h, --help            show this help message and exit
  -p PORT, --port PORT  Device name, default /dev/ttyUSB0
  -b BAUD, --baud BAUD  Baudrate, default 115200
  -f SRC, --src SRC     Source file on computer (directory for *.lua)
  -t DEST, --dest DEST  Destination file on MCU, default to source file name
  -c, --compile         Compile lua to lc after upload
  -d, --dofile          Run the Lua script after upload
  -v, --verbose         Show progress messages
  -l, --list            List files on device
  -i, --id              Query the modules chip id.
  -e, --echo            Echo output of MCU until script is terminated.
  -r, --restart         Restart MCU after upload
  -w, --wipe            Delete all lua/lc files on device. (No Warning)
  -n, --initlua         Rename 'init_app.lua' to 'init.lua'
  --delay DELAY         Delay in fractonal seconds each write (serial only).
  --delete DELETE       Delete a lua/lc file from device.
  --ip IP               Connect telnet (--ip IP[:port]) can be comma separated
                        list of ip
  --chkmissing          Check for missing files on flash
```