nodemcu/nodemcu-firmware #2885: [feature request] 'Panic' Error logging to file

chathurangawijetunge, 2019-08-13:
hi really appreciate if there is a module or any  way that we can log errors to a file (like in python). to direct errors in long time run.....?

nwf, 2019-08-13:
Work is in progress as part of a general upheaval of the foundations of nodemcu to be able to provide a `panic` handler.  See #2803 for discussion.

chathurangawijetunge, 2019-09-22:
Is there any way of capturing lua interpret's panic error message to a variable..?

TerryE, 2019-09-22:
@chathurangawijetunge,  We will be converting all module CBs to use a protected call variant as part of the Lua 5.3 development, hence they won't panic as such.  The errors will be delivered to an error reporter as a separate SDK task.  The default reporter is the print function, but users will have the option to replace this with their own custom reporter.
 
5.3 is currently in Alpha test, but a working `dev` build should be available in a month or so.

chathurangawijetunge, 2019-10-29:
Any luck in capturing errors

TerryE, 2019-10-29:
@chathurangawijetunge, as I said, this is included in the upcoming 5.3 version,

danjohn, 2020-02-09:
Funny, that this thread isn't closed after halve a year ;)

To get this relevant, we *do* need a mechanism to gather "last word" issues (over telnet) when the ESP runs into trouble (OOM mostly). By now, clients die a silent death without "whereisthebug" hints, when only connected via wifi...

I just read about "53", will this help when it's there?

nwf, 2020-02-09:
@danjohn You're welcome to dig into the tree and write some plans and/or code, just like the rest of us.  Noting elapsed time will not do anything except make us cross.

It is almost assuredly not possible to send text over telnet during an OOM situation, as the Espressif-provided network stack requires allocating packet buffers and posting tasks to internal queues.  The best you'll be able to do, reliably, anyway, is to see what went wrong at boot and report then.

Lua-induced exceptions ("panics") are a different matter and could well be reported in real time; there's a lot of new plumbing necessary to use `pcall` everywhere, tho', which has, as @TerryE notes, been rolled up into the 5.3 update that he's working on.

danjohn, 2020-02-09:
Y won't believe, I now got a new Composter(x64) from hell (company). Maybe really I can now take a seat near Pros by installing marcel's docker image, and "dig into the tree". If time permits.

Apart from "making cross" (sorry): OOM is issued from the lua machine, isn't? What about getting the (well tested) telnet code into ring0, leaving all mem reservations behind ("Panik" means: no future anyway), and spitting out all (maybe helpful) data to serial|wifi bevore dying?
The telnet code has to resettle into the C kernel in that case. Oh, I see.

Y're right, maybe I should shut up bevore having done anything to the project but consuming (and I did! thankfully). cu

TerryE, 2020-02-12:
At the moment all (bar 1 use) CBs in the module libraries use a call rather than a pcall to invoke any Lua CB.  If the Lua CB doesn't itself wrap error prone code in a `pcall()` then any error will be uncaught and result in a system panic, rebooting the CPU.  The major downside of this is that you don't get any traceback info or opportunity to intercept the error for logging purposes.

If the Lua application doesn't protect its code then the only safe thing to do is to exit the current task as the Lua stack gets unrolled, and in general ignoring errors is _a bad thing to do_.  However, having a catch-all CB for errors does seem a sensible option.  By default this catch-all posts a separate task to print the error traceback before restarting the CPU, but it would be pretty straight forward to allow the user to specify a custom application error catch all which could analyse the error or send it to a logger.    

I've added a new API call for both 5.1 and 5.3 to facilitate this, but we can't move forward until we merge the 5.3 patch.

chathurangawijetunge, 2020-06-27:
Thank you for the the new brake through.....
How to log to file only panic errors...?

TerryE, 2020-06-27:
`node.setonerror(function)`.  Sorry I missed pushing the updated `node.md`.  After a panic this function is posted which the error traceback as an argument.  By default this prints the error and restarts the ESP after a second (enough time  send the error over a telnet session). You can replace this bay any other logger, e.g. append to a logfile.  Note that in most circumstances it is best still to do a clean restart after processing the error.

chathurangawijetunge, 2020-06-28:
Sorry for the trouble i tried this with no luck....

function log_to_file(opipe)   
    local rec = opipe:read(1400)
    if rec and #rec > 0 then
       file.open("LOG.txt", "a+")
       file.write(rec)
       file.close()
    end
    return false
end
node.output(log_to_file, 1)

-- Creat a panic in 1 sec
tmr.create():alarm(1000, tmr.ALARM_SINGLE, function () gpio.write(x,0) end) 

Log file only writs ">" char ..

TerryE, 2020-06-28:
What are you trying to do here? This is nothing to do with the error handler, so why should it work?

HHHartmann, 2020-06-28:
@TerryE Terry you started above

> By default this prints the error and restarts the ESP after a second (enough time send the error over a telnet session)

so instead of sending it, it is written to a file here. 
As I understand this should work. 
Has it landed in dev already?

chathurangawijetunge, 2020-06-28:
> What are you trying to do here? This is nothing to do with the error handler, so why should it work?

What do u mean...? I'm trying to write to a file instead of tellnet

TerryE, 2020-06-28:
Output redirection and trapping panics are nothing to do with each other.  If you want to log a panic to file then do something like:
```Lua
  node.setonerror(function(e)
    local f = file.open("LOG.txt", "a+")
    f:write(rec); f:close()
    node.restart()
  end)
```

closed 2020-06-28 by TerryE

HHHartmann, 2020-06-28:
@TerryE Terry but what @chathurangawijetunge is doing is to capture stdout and write it to a file.
I would like to understand why after printing the message to stdout (in the default errorhandler) it does not show up in the capture function set by `node.output()` within the grace period of 1 sec.
Is there any task handling blocked?

TerryE, 2020-06-28:
> @TerryE Terry but what @chathurangawijetunge is doing is to capture stdout and write it to a file.

Maybe so but can we stick to #1010?  If there's something not working with output redirection then set up a test case for this to demonstrate and raise an issue.  This has nothing to do with error handling.  I am up to my eyeballs at the moment in a complex PR and I don't want to get sidetracked.  If this is an issue then let's handle it in something other than a feature request that is nothing to do with this and has been implemented.

HHHartmann, 2020-06-28:
by the way @chathurangawijetunge 
The example works for me. Are you sure you grabbed a build which includes the new feature?
does 
``` Lua
node.setonerror(function(e)     print("errormessase is today:",   e)  node.restart()   end)
```
work for you?

chathurangawijetunge, 2020-06-29:
> by the way @chathurangawijetunge
> The example works for me. Are you sure you grabbed a build which includes the new feature?
> does
> 
> ```lua
> node.setonerror(function(e)     print("errormessage is today:",   e)  node.restart()   end)
> ```
> 
> work for you?

Finally it works only in Dev not in master :-)