nodemcu/nodemcu-firmware #2887: Float/Int

danjohn, 2019-08-14:
Can I diffenentiate the underlying firmware flavour (int/float) in a simple way?
I'm switching from t2t bevore I release 1.0, and the LUA code should be smart enough to cope with that.

Do you think,(math.random()==0) is a safe try?
` function round(x) if (math.random()==0) then return x else return math.floor(x+.5) end`

 (found no buit-in constant to decide float/int)!
Sorry if offtopic.

HHHartmann, 2019-08-14:
There is a new functionality to differentiate int from float in the dev branch which will be merged to master soon.
See here https://nodemcu.readthedocs.io/en/dev/modules/node/#nodeinfo

marcelstoer, 2019-08-15:
Also note that there's no `math` module in NodeMCU: https://nodemcu.readthedocs.io/en/latest/lua-developer-faq/#how-is-nodemcu-lua-different-to-standard-lua

> ...which will be merged to master soon.

Therefore -> closing

closed 2019-08-15 by marcelstoer

jmd13391, 2019-08-19:
@marcelstoer 

> Also note that there's no math module in NodeMCU

Really?
Master Build on 07/21/19:

local function round(...)
  local v, b = ...
  local s = (v >= 0 and 1) or -1
  b = b or 1
  return **math.floor(v/b + s * 0.5)** * b
end

print(round(22.453,.01))

>22.45

marcelstoer, 2019-08-19:
> Really?

Yes, kind of. Depending on your firmware type there are a handful of functions linked in as per https://github.com/nodemcu/nodemcu-firmware/blob/master/app/lua/lmathlib.c#L317. However, we state in [the FAQ](https://nodemcu.readthedocs.io/en/latest/lua-developer-faq/#how-is-nodemcu-lua-different-to-standard-lua) that the `math` module be missing as to not raise any false expectations.

danjohn, 2019-08-19:
Gruezi, Servus.
The problem in #1 could evtly be catched by `function round(x) return math.floor(x+1/2) end` for int/float flavours... possible, but didn't try that by now.

Apart from this, two problems arised (maybe worth another thread?):
* Doc fails in explaining the built-in math functionality (enough for most apps, but undoc?)
* Int firmware can't "node.flashreload" a lua.cross-f-compiled image (states "header problem" IIRC)
(luac.cross -v = "Lua 5.1.4  Copyright (C) 1994-2011 Lua.org, PUC-Rio")

Have I catched a bug/usecase, or didn't I read all FAQs? 
;)

nwf, 2019-08-19:
@danjohn: I think you'll find that you'll have better luck with `node.flashreload` on an integer-type build if you use the `luac.cross.int` that's built next to the `luac.cross`. :)

danjohn, 2019-08-24:
@nwf: Thank you for pointing me to that!

I got my copy of `luac.cross` from an (don't remember) internet source, as my box is too poor to host a complete (docker) build chain. `luac.cross.int` wasn't included.

Can you point me to a binary source (Terry's won't help, because I need i386, he hosts only i64 binaries)? TY.

nwf, 2019-08-24:
I'm not aware of binary distributions, sorry.  I'd generally caution against running binaries you find scattered around the Internet, too.

danjohn, 2019-08-24:
Right, it seems we have the same opinion to avoid unknown "binaries off the internet". I'm sure, the binary wasn't from strange sources, an I straced it bevore use.

But, if you have a working buid chain:
 Can you please "make" a `luac.cross(.int)` binary, i386 (and provide it (email, ftp) to me)? 
Knowing you from this forum is legitimation enough to let me believe, that y're not a evil american IoT invader (hoping to be not too credulous :-).

That's all we would need, in our old school, out-timed, unprofessional lua programming shacks...

HHHartmann, 2019-08-24:
That will not be so easy. If I am not mistaken the standard binaries are 64 bit.
I had the same problem a while ago, trying to ramp up an online luac.cross service at the time Terry did. But he beat me and also did a much better job than i could have. 
Btw. You know his online service? 

However I saved the work on 386 binaries. Could have a look of it still works if you can't use the online service.

TerryE, 2019-08-24:
> Can you please "make" a `luac.cross(.int)` binary, i386

If you want a Debian-ready version, just download the same distro version as you are cloud building into `/tmp`, unzip, edit `user_config.h`, cd to the `luac-cross` for and make.  The binary is in the NodeMCU root folder so preserve this. If your Debian is 32-bit then your built luac.cross will be 32-bit as well.

 This is a 5 min job for you or anyone else _if they have a 32 bit system_, so why ask someone else to do it for you?