nodemcu/nodemcu-firmware #2889: Bug in gpio.trig()

chathurangawijetunge, 2019-08-15:
My Code is ....

local GPIO14 = 3
gpio.mode(GPIO14, gpio.INT, gpio.PULLUP)

y=0
function doorLocked(level, when, eventcount)
    gpio.trig(GPIO14, "none") 
    y=when-y
    print("Trg..",level, when, eventcount,y/1000)
    tmr.create():alarm(100, tmr.ALARM_SINGLE, function()
        gpio.trig(GPIO14, "down",  doorLocked)
    end)
end

gpio.trig(GPIO14, "down", doorLocked)
---------------------------------------------------------------------
get error:- 
Trg..	1	6827046	2	6827
Trg..	0	6962822	1	135
Trg..	0	7084608	1	6948
PANIC: unprotected error in call to Lua API (attempt to call a nil value)

chilipeppr, 2019-08-15:
The 2nd param is a constant, not a string. I think you mean gpio.INTR_DOWN not "down". And, it seems really odd you are calling yourself over and over recursively and recreating the trigger callback each time?

chathurangawijetunge, 2019-08-15:
what do u mean by "gpio.INTR_DOWN" there is no such.. 
in the documentation 
_type ="up", "down", "both", "low", "high", which represent rising edge, falling edge, both edges, low level, and high level trigger modes respectivey. If the type is "none" or omitted then the callback function is removed and the interrupt is disabled._

chathurangawijetunge, 2019-08-15:
I'm trying to use one switch for different functions (count the number of switch pushers ) code is 

Master_Sw = 3
Master_ctn=0
local mytimer = tmr.create()

local function Master_pressed(a,b,c)
   gpio.trig(Master_Sw, "none")
   Master_ctn=Master_ctn+1
   mytimer:alarm(150, tmr.ALARM_SINGLE, function()
      mytimer:alarm(300, tmr.ALARM_SINGLE, function()
          if Master_ctn>0 then
             print("pressers",Master_ctn)
          end                 
          Master_ctn=0 
      end)          
      gpio.trig(Master_Sw, "down",Master_pressed)
   end)
end          
gpio.trig(Master_Sw, "down", Master_pressed)

but some times only i get

pressers	1
pressers	1
pressers	6
pressers	5
pressers	10
pressers	1
pressers	1
pressers	2
pressers	2
pressers	1
pressers	6
pressers	1
pressers	2
pressers	4
pressers	8
PANIC: unprotected error in call to Lua API (attempt to call a nil value)

where is the error......?

marcelstoer, 2019-08-15:
> what do u mean by "gpio.INTR_DOWN" there is no such..

Hadn't just ignored our issue template and filled in all the required information instead @chilipeppr wouldn't have had to wonder whether you are on ESP32 or ESP8266.

chathurangawijetunge, 2019-08-15:
thank you for the feed back.
i am on ESP8266

NodeMCU custom build by frightanic.com
	branch: master
	commit: 68c425c0451f72fcabb1b2d6d31a8555f087371b
	SSL: false
	modules: file,gpio,mqtt,net,node,rtctime,sntp,tmr,wifi
 build created on 2019-08-14 01:47
 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)

nwf, 2019-08-16:
It's possible this is related to #2553 (sometimes function refs go missing there, too) or perhaps we have a #2881-style bug in `gpio` or `tmr`.

chathurangawijetunge, 2019-08-24:
Any Fix on this issue......?

nwf, 2019-08-24:
We surely would have commented if appropriate.  You are welcome to investigate the issue yourself or offer to hire someone to work on issues at your pace, otherwise we work on things at our pace and if issues do not affect our workflows, well, we'll get to them when we get to them.

vsky279, 2019-09-04:
Hi,

I have not tested your code but maybe the critical part is disabling and enabling the gpio trigger. It seems it is a bug. If it is the case then you can simple avoid it. Try something like this:
```Lua
local GPIO14 = 3
gpio.mode(GPIO14, gpio.INT, gpio.PULLUP)

local last = 0
local delay_tmr = tmr.create()
function doorLocked(level, when, eventcount)
    if when - last > 100 then
      print("Trg..",level, when)
      delay_tmr:alarm(100, tmr.ALARM_SINGLE, function()
        --
      end)
      last = when
    end
 end
gpio.trig(GPIO14, "down", doorLocked)
```
It seems to me that the functionality is the same. Do you get these unprotected errors?

closed 2020-09-12 by stale[bot]