nodemcu/nodemcu-firmware #2892: TLS handshake heap usage higher under SDK 3.0.1 compared to SDK 2.2.1

NicolSpies, 2019-08-20:
### Expected behavior
It is expected that the heap usage, during the TLS handshake, when comparing SDK 3.0.1 and SDK 2.2.1, should be relatively similar.
### Actual behavior
- When using `tls.setDebug(4)` to compare the available heap during the respective TLS handshake processes, it is observed that both SDKs start with about the same available heap. 
- SDK 3.0.1 progressively uses more heap to a maximum of `1.4Kb`, 60% through the handshake process.
- During the last 30% the difference in heap used by the two SDKs reduces to end at about the same amount of heap available.
### Test code
Similar test code and environment are used for both tests executed using the different builds as indicated.
### NodeMCU version
The two builds used are:

` build 2019-08-20 14:34 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
`
`build created on 2019-08-20 17:08 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
`
### Hardware
Standard Hardware

NicolSpies, 2019-08-21:
This report is for information only as it has been noticed and mentioned before when the switch over to SDK 3.0.0 took place.

nwf, 2019-08-21:
It would be perhaps informative, I suppose, if you could post the debug logs from both firmware versions.

NicolSpies, 2019-08-21:
Debug log for Telegram TLS handshake using `build 2019-08-20 14:34 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)` :
Heap starts at 22760 kB runs down to 9392 Kb
```
client handshake start.
TLS<2> (heap=22760): ssl_tls.c:6850 => handshake
TLS<2> (heap=22760): ssl_cli.c:3400 client state: 0
TLS<2> (heap=22760): ssl_tls.c:2574 => flush output
TLS<2> (heap=22760): ssl_tls.c:2586 <= flush output
TLS<2> (heap=22760): ssl_cli.c:3400 client state: 1
TLS<2> (heap=22760): ssl_tls.c:2574 => flush output
TLS<2> (heap=22760): ssl_tls.c:2586 <= flush output
TLS<2> (heap=22760): ssl_cli.c:777 => write client hello
TLS<3> (heap=22760): ssl_cli.c:815 client hello, max version: [3:3]
TLS<3> (heap=22760): ssl_cli.c:824 dumping 'client hello, random bytes' (32 bytes)
TLS<3> (heap=22760): ssl_cli.c:824 0000:  bf 0e fb a1 62 d3 f5 dc 7f 55 b9 bc f1 05 08 6f  ....b....U.....o
TLS<3> (heap=22760): ssl_cli.c:824 0010:  d9 24 e2 cd 5e 79 26 81 4f 95 38 21 37 42 6f bf  .$..^y&.O.8!7Bo.
TLS<3> (heap=22760): ssl_cli.c:877 client hello, session id len.: 0
TLS<3> (heap=22760): ssl_cli.c:878 dumping 'client hello, session id' (0 bytes)
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c02c
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c030
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: 009f
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c0ad
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c09f
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c024
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c028
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: 006b
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c00a
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c014
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: 0039
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c0af
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c0a3
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c087
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c08b
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c07d
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c073
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c077
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: 00c4
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: 0088
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c02b
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c02f
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: 009e
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c0ac
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c09e
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c023
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c027
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: 0067
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c009
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c013
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: 0033
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c0ae
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c0a2
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c086
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c08a
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c07c
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c072
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: c076
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: 00be
TLS<3> (heap=22760): ssl_cli.c:925 client hello, add ciphersuite: 0045
TLS<3> (heap=22760): ssl_cli.c:937 client hello, got 40 ciphersuites (excluding SCSVs)
TLS<3> (heap=22760): ssl_cli.c:946 adding EMPTY_RENEGOTIATION_INFO_SCSV
TLS<3> (heap=22760): ssl_cli.c:995 client hello, compress len.: 1
TLS<3> (heap=22760): ssl_cli.c:997 client hello, compress alg.: 0
TLS<3> (heap=22760): ssl_cli.c:189 client hello, adding signature_algorithms extension
TLS<3> (heap=22760): ssl_cli.c:274 client hello, adding supported_elliptic_curves extension
TLS<3> (heap=22760): ssl_cli.c:339 client hello, adding supported_point_formats extension
TLS<3> (heap=22760): ssl_cli.c:521 client hello, adding encrypt_then_mac extension
TLS<3> (heap=22760): ssl_cli.c:555 client hello, adding extended_master_secret extension
TLS<3> (heap=22760): ssl_cli.c:588 client hello, adding session ticket extension
TLS<3> (heap=22760): ssl_cli.c:1074 client hello, total extension length: 60
TLS<2> (heap=22760): ssl_tls.c:2867 => write record
[... 903 lines cut ...]
TLS<4> (heap=10256): ssl_tls.c:1734 dumping 'IV used' (12 bytes)
TLS<4> (heap=10256): ssl_tls.c:1734 0000:  43 c1 63 a0 e9 92 8f 04 f1 4b bf d4              C.c......K..
TLS<4> (heap=10256): ssl_tls.c:1735 dumping 'TAG used' (16 bytes)
TLS<4> (heap=10256): ssl_tls.c:1735 0000:  89 3b b6 50 29 ef dd 94 14 5d a6 2e 75 f5 49 c9  .;.P)....]..u.I.
TLS<2> (heap=10256): ssl_tls.c:2198 <= decrypt buf
TLS<4> (heap=10256): ssl_tls.c:3860 dumping 'input payload after decrypt' (16 bytes)
[... cut ...]
TLS<2> (heap=10256): ssl_cli.c:3400 client state: 15
TLS<2> (heap=10256): ssl_tls.c:2574 => flush output
TLS<2> (heap=10256): ssl_tls.c:2586 <= flush output
TLS<3> (heap=10256): ssl_tls.c:5370 => handshake wrapup
TLS<3> (heap=10256): ssl_tls.c:5343 => handshake wrapup: final free
TLS<3> (heap=12264): ssl_tls.c:5363 <= handshake wrapup: final free
TLS<3> (heap=12264): ssl_tls.c:5425 <= handshake wrapup
TLS<2> (heap=12264): ssl_tls.c:6860 <= handshake
client handshake ok!
HTTP client: Connected
```

NicolSpies, 2019-08-21:
Debug log for Telegram TLS handshake using `build 2019-08-20 14:34 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)` :
Heap starts at 21976 kB runs down to 7488 Kb

```
client handshake start.
TLS<2> (heap=21976): ssl_tls.c:6850 => handshake
TLS<2> (heap=21976): ssl_cli.c:3400 client state: 0
TLS<2> (heap=21976): ssl_tls.c:2574 => flush output
TLS<2> (heap=21976): ssl_tls.c:2586 <= flush output
TLS<2> (heap=21976): ssl_cli.c:3400 client state: 1
TLS<2> (heap=21976): ssl_tls.c:2574 => flush output
TLS<2> (heap=21976): ssl_tls.c:2586 <= flush output
TLS<2> (heap=21976): ssl_cli.c:777 => write client hello
TLS<3> (heap=21976): ssl_cli.c:815 client hello, max version: [3:3]
TLS<3> (heap=21976): ssl_cli.c:824 dumping 'client hello, random bytes' (32 bytes)
TLS<3> (heap=21976): ssl_cli.c:824 0000:  b8 f1 3a 47 d4 c4 a4 05 07 36 77 dd 97 e1 24 2a  ..:G.....6w...$*
TLS<3> (heap=21976): ssl_cli.c:824 0010:  19 5d 9f 57 09 dd ee c8 04 f1 ed fe ee 03 b6 87  .].W............
TLS<3> (heap=21976): ssl_cli.c:877 client hello, session id len.: 0
TLS<3> (heap=21976): ssl_cli.c:878 dumping 'client hello, session id' (0 bytes)
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c02c
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c030
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: 009f
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c0ad
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c09f
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c024
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c028
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: 006b
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c00a
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c014
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: 0039
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c0af
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c0a3
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c087
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c08b
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c07d
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c073
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c077
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: 00c4
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: 0088
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c02b
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c02f
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: 009e
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c0ac
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c09e
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c023
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c027
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: 0067
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c009
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c013
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: 0033
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c0ae
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c0a2
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c086
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c08a
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c07c
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c072
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: c076
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: 00be
TLS<3> (heap=21976): ssl_cli.c:925 client hello, add ciphersuite: 0045
TLS<3> (heap=21976): ssl_cli.c:937 client hello, got 40 ciphersuites (excluding SCSVs)
TLS<3> (heap=21976): ssl_cli.c:946 adding EMPTY_RENEGOTIATION_INFO_SCSV
TLS<3> (heap=21976): ssl_cli.c:995 client hello, compress len.: 1
TLS<3> (heap=21976): ssl_cli.c:997 client hello, compress alg.: 0
TLS<3> (heap=21976): ssl_cli.c:189 client hello, adding signature_algorithms extension
TLS<3> (heap=21976): ssl_cli.c:274 client hello, adding supported_elliptic_curves extension
TLS<3> (heap=21976): ssl_cli.c:339 client hello, adding supported_point_formats extension
TLS<3> (heap=21976): ssl_cli.c:521 client hello, adding encrypt_then_mac extension
TLS<3> (heap=21976): ssl_cli.c:555 client hello, adding extended_master_secret extension
TLS<3> (heap=21976): ssl_cli.c:588 client hello, adding session ticket extension
TLS<3> (heap=21976): ssl_cli.c:1074 client hello, total extension length: 60
[... 907 lines cut ...]
TLS<4> (heap=8968 ): ssl_tls.c:1735 0000:  a9 41 28 2d a6 79 21 4e bf 10 d2 41 e8 9b 51 24  .A(-.y!N...A..Q$
TLS<2> (heap=8968 ): ssl_tls.c:2198 <= decrypt buf
TLS<4> (heap=8968 ): ssl_tls.c:3860 dumping 'input payload after decrypt' (16 bytes)
TLS<4> (heap=8968 ): ssl_tls.c:3860 0000:  14 00 00 0c 00 34 53 3c 83 df 79 e7 f7 75 b4 0c  .....4S<..y..u..
TLS<3> (heap=8968 ): ssl_tls.c:3265 handshake message: msglen = 16, type = 20, hslen = 16
TLS<2> (heap=8968 ): ssl_tls.c:3937 <= read record
TLS<2> (heap=8968 ): ssl_tls.c:5626 <= parse finished
TLS<2> (heap=8968 ): ssl_cli.c:3400 client state: 14
[... cut ...]
TLS<2> (heap=8968 ): ssl_cli.c:3400 client state: 15
TLS<2> (heap=8968 ): ssl_tls.c:2574 => flush output
TLS<2> (heap=8968 ): ssl_tls.c:2586 <= flush output
TLS<3> (heap=8968 ): ssl_tls.c:5370 => handshake wrapup
TLS<3> (heap=8968 ): ssl_tls.c:5343 => handshake wrapup: final free
TLS<3> (heap=11048): ssl_tls.c:5363 <= handshake wrapup: final free
TLS<3> (heap=11048): ssl_tls.c:5425 <= handshake wrapup
TLS<2> (heap=11048): ssl_tls.c:6860 <= handshake
client handshake ok!
HTTP client: Connected
```
```

```

closed 2020-08-16 by NicolSpies