nodemcu/nodemcu-firmware #2895: luac(.cross) options for Lua53

TerryE, 2019-08-24:
We have adopted a policy of only changing the standard Lua 5.3 code base when there is a compelling reason to do so.  One area where this policy has triggered a major review is that the NodeMCU Lua 5.1 code is based on the  `eLua` code for saving and loading Lua in compiled format.  Given that the Intel compatible, most ARM implementations, and the ESPs are little-endian twos complement binary with IEEE FP, there _very_ little need to may any material change here at all for Lua53.  Yes, we have added ROTable support, but these are transparent at the compiled Lua LC format (since you can't have array constants anyway in Lua 5.3).  

Hence at the moment the only changes that I've included relate to range checking of integer constants to conform to the ESP integer sizes, and to introduce a compact encoding for integer constants in the dump format: given that nearly all integer constants in code are small so fit into a single byte

To be honest this last optimisation drops the uncompressed code size by less than 10%, but ake no material difference when GZ compressed, so I am even considering backing this out. 

So I have been pondering: why not just support loading the compiled output from standard luac as well?  This would mean that as an alternative to having to make luac.cross, users could just download the binary version (for example [here](http://luabinaries.sourceforge.net/download.html) for Windows) and use that.  Or even going so far as to forget about having a NodeMCU Lua LC format altogether and just adopt the host LC format.  

The one issue that this doesn't address is how to create LFS images.  Or does it?  At the moment lua-5.3 generates a very specific format for compiler output of multiple Lua files that looks like this for source a.lua, b.lua, c.lua:
```
main <(luac):0,0> (7 instructions at 0x55bc89d33070)
0+ params, 2 slots, 1 upvalue, 0 locals, 0 constants, 3 functions
    1   [1] CLOSURE     0 0 ; a.lua:main
    2   [1] CALL        0 1 1
    3   [1] CLOSURE     0 1 ; b.lua:main
    4   [1] CALL        0 1 1
    5   [1] CLOSURE     0 2 ; c.lua:main
    6   [1] CALL        0 1 1
    7   [1] RETURN      0 1
constants (0) locals (0) upvalues (1):
    0   _ENV    1   0

main <path/a.lua:0,0> (N instructions)
0+ params, X slots, Y upvalue, Z local, K constants, F functions
...
<and so on for b.lua and c.lua>
```
If we went this way, we would drop the `flashimg.c` module, and rewrite the `lflash.c` loader to load an LC file of this format into LFS. Because of RAM limitations, this would take two passes with a reboot at the end of each pass:
1.   To build the ROstrt for all TStrings used in the LC file and to write this to the LFS region.
2.   To load the Proto hierarchies followed by the ROTable index table into the LFS region.

We could also allow the option of passing a Lua pattern as the compile file name, and this would tick off another user wish-list item: the ability to create LFS images on ESP only.  (Albeit with the PITA size limitation for doing on ESP compiles.  

No double Johny will want to retain the ability to convert such an LS file to absolute format, but this could be done quite simply by a standalone host C exe or even a Python script. 

I would still want to keep a version of the host luac complete with the `-e` option for convenience and testing purposes, but for most developers using this would jut be one alternative.
 
There are a few wrinkles and variants to this approach but I would appreciate the thoughts of **the committers and any other readers with karma**, before deciding on an implementation path.

jmattsson, 2019-08-25:
Wouldn't this play mayhem with things like the the integer-only build, and differently sized ints/floats in 5.3?

TerryE, 2019-08-25:
My current intent is to _**drop**_ the integer only build and substitute an `int` / `float` TValue instead.

TerryE, 2019-08-25:
It is really quite hard to know what to do for the best here especially given the general lack of input and view form the other committers; both approaches have their relative strengths:

-  Keeping the current model involves less work and "no change" is a conservative approach that still leaves us with use and learning barriers for our Win10-only users who want to develop ESP/ Lua apps without getting into command-line use and building `luac.cross` for themselves. Yes, we could easily provide a native Win10 `luac.cross` executable through the Travis Windows support, but this requires another committer to do this work.

-  Some variant of this approach would at least give this `luac` options to such developers and incidentally also allow them to build smaller LFS images on-ESP.  The challenge that I have is keeping the LFS image loader within the RAM availability of the ESP8266.  We need a two or possibly three pass approach because of these RAM limitations:  (explode the the GZ compressed format;) build and load the ROstrt and reboot; add the Protos and ROTable index.

Anyway, I've now got to the point of a full clean compile of the firmware with both the `LUA=51` and `LUA=53` variants less LFS support for Lua53.  I want to carry on debugging the `luac.cross -e` environment for now.  I will make a call on this after I've got some more considered views.

jmattsson, 2019-08-26:
I find the idea of being interoperable with a bog-standard .lc file appealing. The less customised stuff we need to rely on, the less headaches in terms of maintenance I'd expect, and lower barrier of entry. As you mention, yes, I'd very much like to have a way of generating absolute LFS images, but I'm fine with this being an advanced use case which takes a bit more effort to use (it'll all get automated anyway).

As presented it seems there are significant advantages to be had from what you're suggesting, and no serious downsides. I'm still not entirely clear on what an `int/float` TValue change would mean in terms of impact to backwards compatibility of .lua files - my gut is warning me about hazards for integers in the 2^32 - 2^53 range, but that might be unwarranted.

TerryE, 2019-09-17:
> I find the idea of being interoperable with a bog-standard .lc file appealing.

I have been considering options for LFS implementation in Lua53.  One of the things that I dislike about the `app/lua` implementation is that there are two variants for dumping and loading Lua bytecode: 

1. `ldump.c` + `lundump.c`
2. `lflashimg.c` + `lflash.c`.

Wouldn't it be preferable to unify these into a single load / unload mechanism?  This is possible, but only if I make some changes to the internal ordering of the LC file format.  The issue here is that any `Proto` can embed other Proto definitions internally, creating a Proto hierarchy.  Hence the load algorithm looks like:
```
load_proto(p)
  load some bits
  foreach subp in p
    load_proto(subp)
  load some more bits
end
```
Because of this each Proto's resources get scattered by being interleaved with those for its sub-protos hierarchy.  This also means that address references need to written non-serially, all of which is bad news for writing serially to the LFS region.  There is a simple method of avoiding this which is to reverse the order of the proto hierarchy tree walk, so that the lowest protos in the hierachy are loaded first: any proto now refers to other protos that are already loaded.
```
dump_proto(p)
  foreach subp in p
    dump_proto(subp)
  dump proto content
end
```
This also enables the proto's resource to be allocated as a single allocation unit and written to serially, and the same code can be used for loading LC into RAM and into LFS.  There are two slight complications, which are straightforward to address once identified as potential issues.

-  The load process must be GCable on error (for example if we are out of memory during the load)
-  The current format embeds string constants in each proto as a <len><bytes> definition.  We need to hoist all such strings to the head of the load image as a vector of <len><bytes>.  The string constants inside the protos are now just an index into this vector.  This approach allows the 

The downside of this restructuring is that the NodeMCU load format is internally reordered v.v. standard `luac` output.  If we want to enable an option for using standard `luac` on the host then the easiest work around is to write a format converter.  And since we already have a host dependency on Python being installed then probably the simplest implementation will be to write this in Python.

TerryE, 2019-10-28:
PR #2912 implements these discussions.  We already have an overlap with #2917, so let us handle any residual discussions there.

closed 2019-10-28 by TerryE