nodemcu/nodemcu-firmware #2899: gpio.trig's callback reports wrong level sometimes

joysfera, 2019-08-30:
### Expected behavior
gpio.trig's callback reports level of first interrupt even if several of them were "elided" (quoted from the documentation of `gpio.trig`).

### Actual behavior
whenever the `eventcount` is greater than 1 it means that several interrupts were elided. At that moment the `level` seems to contain the level of the last interrupt, not the first one. This makes the callback for "both" edges largely unusable.

### Test code
copied from the documentation and modified to use "both" edges and to work around bouncing by ignoring interrupts that come too soon since the previous one:
```Lua
do
  -- use pin 1 as the input pulse width counter
  local pin, pulse1, du, now, trig = 1, 0, 0, tmr.now, gpio.trig
  gpio.mode(pin,gpio.INT,gpio.PULLUP)
  local function pin1cb(level, pulse2, eventcount)
    local since = pulse2 - pulse1
    if since > 250000 then
      print( level, since / 1000,  eventcount)
    end
    pulse1 = pulse2
  end
  trig(pin, "both", pin1cb)
end
```
Now when you use a switch connected to given pin and keep turning it on and off slowly (with a delay > 0.25 second) the print should return different `level` each time, I believe. Unfortunately, whenever the `eventcount` is 2 or more the `level` often contains value of different IRQ (the latter one) than what triggered the callback originally. This makes it hardly usable.

IMHO it should always return the level of the very first IRQ it caught instead of the last one - the level of the last one can easily be obtained by direct reading of gpio current state, if anybody needed it (for the switch multiple interrupts together are usually caused by a bouncing so one can/should ignore all the subsequent interrupts).

### NodeMCU version
2.2.1-master_20181207

### Hardware
Sonoff Mini DIY

joysfera, 2019-08-30:
Here's test output of the code above. Since I used a push button the output (those four lines) equal to four button actions: push - release - push - release. Since the `level` is printed in the first column it should have printed `0` (push) - `1` (release) - `0` (push) - `1` (release). Instead I've got 0 - 1 - 0 - **0**:

```
0	976	1  
1	821	1
0	849	1
0	753	2
```

As you can see in the last step the release of the button got "lost" - the `level` returned (first column) is `0` instead of `1`  even though the button was released - most probably because the `eventcount` is 2, so the push button bounced a bit during the release. I take this as the evidence that the `level` contains the level of the latter (or even last) interrupt, not the first one.

joysfera, 2019-08-30:
Oh, the more I test it the worse it looks. Now it seems like the `level` cannot be trusted even if the `eventcount` is 1. Here's another push - release - push - release sequence:

```
0	1514	1
1	1180	1
0	1390	1
0	1217	1
```

The last line - `level` is 0, should have been 1, `eventcount` is 1.

Please note that reading the gpio directly in the callback will not give good results because the button is bouncing at that moment. The only chance I can see now is to catch the interrupt, then wait a bit (say 200 ms) and then read the gpio directly - hoping, that the bouncing finished in the meantime...

vsky279, 2020-01-31:
I think you are experiencing  bouncing issue. There's nothing to be fixed in the NodeMCU itself. You need to implement some debouncing algorithm in your implementation.

You can search for "debounce" in (closed) issues (e.g. #479). The implementation you will need actually depends on your application.

In one of my application I use the following routine:
```Lua
local falling
local function tick(level, when) -- waits for raising edge following falling edge in less than given time
  if level==0 then
    if not falling then falling=when end
  else
    if falling then 
      local width=(when-falling)
      width = ( (width < 0) and ((width + 0x7FFFFFFE) + 1) or width) /1000 -- workaround for #1691
      print(string.format("tick width: %d ms", width))
      if 30<width and width<120 then -- avoid bouncing 
        print("tick")
      end
      falling=nil
    end
  end
end
```
I hope it helps.

I suggest closing this issue. Thank you.

joysfera, 2020-01-31:
@vsky279 I don't think your routine can recognize and handle both falling and raising events so it's not really addressing the issue I reported above.

vsky279, 2020-01-31:
Yep, as said the routine needs to be adjusted to your needs. This one detects short (but not too short) signal. 
I don't think this is anything that can be fixed in NodeMCU...

joysfera, 2020-01-31:
I reported that the values passed to the callback of `trig()` were incorrect and/or largely unusable. Can you prove me wrong, please? If not, then suggesting to close this issue seems weird.

vsky279, 2020-01-31:
Of course I can't. I don't have your signal source. 
But look at the implementation `gpio.c` module:
https://github.com/nodemcu/nodemcu-firmware/blob/71a182caa7841cbb478ed90ede526dc881943c80/app/modules/gpio.c#L35
There's nothing hidden to be fixed.

The `gpio_intr_callback_task` is callback called by the SDK with given parameter that it passes to the registered Lua function.

The solution depends really on what you need.

If you need to act upon signal with higher frequency you will need to go into C code and play with `platform_gpio_register_intr_hook()`. Inspiration to be found e.g. in `rotary` or `dcc` modules sources.

vsky279, 2020-01-31:
Also have a look #2849

HHHartmann, 2020-02-01:
@vsky279 Your analysis is correct so far, but the callback is not initiated from the sdk directly. The primary callback for from the SDK seems to be implentated in platform.c.

https://github.com/nodemcu/nodemcu-firmware/blob/71a182caa7841cbb478ed90ede526dc881943c80/app/platform/platform.c#L247-L250

In there the current signal state together with the current time and the missed count is being sent to gpio.c.

In platform.c the level (and the timestamp) would have to be saved if there are no pending (already counted) interrupts. That way timestamp and level of the first interrupt would be delivered as @joysfera requestes.

vsky279, 2020-02-01:
@HHHartmann Gregor, of course you are right. I was a bit quick. I remember reading this part of code when dealing with the `dcc` module.

pjsg, 2020-02-01:
As I recall, the level of the pin is grabbed pretty quickly after the interrupt is handled. *However* there is no guarantee that the value on the pin is the same as it was when the interrupt was triggered. I suppose that the code could be changed for the edge triggered and level triggered interrupts to always return the 'right' value -- but that leaves the both-edge interrupt.....

joysfera, 2020-02-01:
Well, maybe I could describe my original intention - I wanted to use a rocker switch with NodeMCU but handle not only simple switch but also pulses generated by momentary spring return rocket switches. With a rocker switch the first interrupt's edge is what is important (I thought), the remaining bunch of interrupts are electrical bounces. That's why I wanted to use the `level` parameter to find out what user did with the switch. And then I found out that in current NodeMCU the `level` did not return the first edge so it is unreliable/unusable parameter (in my eyes).

pjsg, 2020-02-01:
Probably what you want is (for a momentary switch) do the following:

* Set it to interrupt on DOWN. Read the status, if already down, move to
the next step.
* When interrupt happens, disable the interrupt and set a timer for 20
milliseconds
* When timer goes off, set the edge interrupt for UP. Read the status, if
already UP, move to the next step
* When the UP interrupt goes off, disable the interrupt and set a timer for
20 milliseconds
* Go to the first step.

This will give you the push and release that I think you are looking for.

HHHartmann, 2021-01-31:
So do we need to fix this or not?

joysfera, 2021-01-31:
probably not.

HHHartmann, 2021-01-31:
Then we should close it

closed 2022-04-30 by stale[bot]