nodemcu/nodemcu-firmware #2900: rst cause:2, boot mode:(3,6)

532910, 2019-09-06:
The following code
```lua
d = require( 'ds18b20' )

local function R()
	d:read_temp( readout, 4 )
end

function readout( temp )
	print( temp )
end

tmr.create():alarm( 1 * 2000, tmr.ALARM_SINGLE, R )
```
causes

```rst cause:2, boot mode:(3,6)```

on esp8622 and esp8285  with the 6-modules (file, gpio, node, ow, tmr, uart) build from https://nodemcu-build.com/ 

The workaround I found:

```lua
d = require( 'ds18b20' )

local function R()
	print( "OK" )
	d:read_temp( readout, 4 )
end

function readout( temp )
	print( temp )
end

tmr.create():alarm( 1 * 2000, tmr.ALARM_SINGLE, R )
```

ds18b20.lua is
https://github.com/nodemcu/nodemcu-firmware/blob/master/lua_modules/ds18b20/ds18b20.lua

vsky279, 2019-09-08:
Do you have some other (random) resets?
Is your ESP properly powered?
Do all the examples from the [documentation](https://nodemcu.readthedocs.io/en/dev/lua-modules/ds18b20/) work fine for you?

Evonic4, 2019-09-08:
tmr.register(0, ...
dont work

532910, 2019-09-08:
I have not random or other resets at all.
My ESP is  properly powered.
All examples works fine?

Could you explain how your questions is related to the fact, that adding `print` before `read_temp` makes it work?

vsky279, 2019-09-08:
> tmr.register(0, ...
> dont work

You're right. This needs to be amended.

vsky279, 2019-09-08:
> All examples works fine.

Great.
 
> Could you explain how your questions is related to the fact, that adding `print` before `read_temp` makes it work?

I thought it might be some timing issue impacting current draw and corresponding voltage drop. Postponing readout could have spread the current peak to longer time period and lower maximum current. Just a guess.

532910, 2019-09-08:
You don't need the ds18b20 itself to reproduce this issue. Could you try to run both examples?

532910, 2019-09-08:
> tmr.register(0, ...
> dont work

It's not clear, could you explain.

HHHartmann, 2019-09-09:
The interface using indexes had been removed some releases ago.
You need to use timer= tmr.create ()  and timer:register ...

See https://nodemcu.readthedocs.io/en/master/modules/tmr/ for details and examples

532910, 2019-09-09:
@HHHartmann `timer:register` isn't mandatory:
example from https://nodemcu.readthedocs.io/en/master/modules/tmr/
```lua
if not tmr.create():alarm(5000, tmr.ALARM_SINGLE, function()
  print("hey there")
end)
then
  print("whoopsie")
end
```

HHHartmann, 2019-09-09:
OK, I tried it with following firmware:

```
NodeMCU 3.0.0.0 
	branch: dev
	commit: 0b90495517c8e2b34dbb3a2a99455ac3c4f05f1c
	release: 2.0.0-master_20170202 +437
	release DTS: 
	SSL: false
	build type: float
	LFS: 0x40000
	modules: adc,bit,color_utils,crypto,dht,encoder,enduser_setup,file,gpio,http,i2c,l3g4200d,mdns,net,node,ow,pipe,pwm,rtcfifo,rtcmem,rtctime,sjson,sntp,spi,struct,tmr,uart,wifi,wifi_monitor,ws2812,ws2812_effects
 build created on 2019-08-14 19:56
 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```
which is a current dev (without Terrys last commit)

I got the exact same results.
I also tried with executing some code before and got weird error messages which might or might not be relevant:
```
> dofile("script1.lua")
> PANIC: unprotected error in call to Lua API (ds18b20.lua:120: bad argument #1 to '?' (table or rotable expected, got nil))

```
and 
```
> dofile("script1.lua")
> PANIC: unprotected error in call to Lua API (ds18b20.lua:118: attempt to index a @! value)

```
The corrosponding line are
```
117  local function conversion(self)
118    local sens = self.sens
119    local powered_only = true
120    for _, s in ipairs(sens) do powered_only = powered_only and s:byte(9) ~= 1 end
121    if powered_only then
```

With the print statement everything works fine.

To me this looks like some sort of memory corruption.
@TerryE maybe this rings a bell for you?

HHHartmann, 2020-06-28:
Whoopsie, this was on my TODO.

Just checked with current build but _without_ attached device and that seems to work now.
Can't remember how I tested above with/without device attached.

#532910 can you confirm that it works with current firmware?

532910, 2020-06-29:
1. You don't need to attach ds18b20.
2. It's still reproducible for me with the build from nodemcu-build.com or git master.

Exact steps:

`% git clone https://github.com/nodemcu/nodemcu-firmware`
`% cat user_modules.h | grep -v '^//\|^$'`:
```H
#ifndef __USER_MODULES_H__
#define __USER_MODULES_H__
#ifndef LUA_CROSS_COMPILER
#define LUA_USE_MODULES_FILE
#define LUA_USE_MODULES_GPIO
#define LUA_USE_MODULES_NODE
#define LUA_USE_MODULES_OW
#define LUA_USE_MODULES_TMR
#define LUA_USE_MODULES_UART
#endif  /* LUA_CROSS_COMPILER */
#endif  /* __USER_MODULES_H__ */
```
`make`

`esptool/esptool.py erase_flash`
`esptool/esptool.py write_flash -fm dout 0x00000 0x00000.bin`
`esptool/esptool.py write_flash -fm dout 0x10000 0x10000.bin`

`% cat init.lua`
```Lua
dofile( 'rst.lua' )
```
`% cat rst.lua`
```Lua
d = require( 'ds18b20' )

local function R()
--	print( "OK" )
	d:read_temp( readout, 4 )
end

function readout( temp )
	print( temp )
end

tmr.create():alarm( 1 * 2000, tmr.ALARM_SINGLE, R )
```

`nodemcu-uploader/nodemcu-uploader.py upload ds18b20.lua init.lua rst.lua`

nwf, 2020-06-30:
There appears to be some confusion in `ds18b20.lua` in which state is carried partially through globals and partially not, with tasks posted referencing both global and upval state.  This risks all kinds of confusion, and so...

Since you are running `d:read_temp` on a *timer*, it would be helpful to know if the reset you see is on the _first_ or _a subsequent_ invocation of your `R` function.

sonaux, 2020-06-30:
I think issue is not with the ds18b20 module, but in garbage collector or something.
Test with no sensor attached:
```NodeMCU 3.0.0.0 
	branch: dev
	commit: 1f2e5bba4ab38c499d41c57a1e74e5c405fd3868
	release: 3.0-master_20200610 +6
	release DTS: 202006160719
	SSL: false
	build type: float
	LFS: 0x30000 bytes total capacity
	modules: <all modules not related to specific hardware>
 build 2020-06-25 01:54 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

node.setonerror(print)
dofile'init.lua'
Heap:	39016
 ?:0: attempt to index a thread value
stack traceback:
	?: in function <?:120>
	(tail call): ?
```
but if I change GC behavior:
```
node.egc.setmode(node.egc.ON_MEM_LIMIT, 24096) --lower value causes 'not enough memory' error
dofile'init.lua'
Heap:	38216
 table: 0x3fff5288
```
or if I change `tmr.create():alarm(...)` to

```
t = tmr.create()
t:alarm( 1 * 2000, tmr.ALARM_SINGLE, R )
```

or change `tmr.ALARM_SINGLE` to `tmr.ALARM_AUTO`  then it also works ok.

With sensor attached it behaves similar, but sometimes works without tuning.
Renaming variable `temp` and function `readout` does not change anything.
Changing `print( temp )` to `print(sjson.encode(temp))` does not throw an error and shows `[]` or temperature value if sensor is connected.

nwf, 2020-06-30:
@sonaux Since you can reproduce the issue, could you answer my question above?

>  it would be helpful to know if the reset you see is on the first or a subsequent invocation

If it is a bug with the use of tasks inside `ds18b20`, as I suspect it might be, that could help narrow it down.

sonaux, 2020-06-30:
@nwf Error is only on first `dofile'init.lua'`, on second and next runs ok.

without sensor:
```
node.setonerror(print)
dofile'init.lua'
Heap:	38944
> ?:0: attempt to index a nil value
stack traceback:
	?: in function <?:120>
	(tail call): ?
dofile'init.lua'
Heap:	29536
> table: 0x3fff0c68
dofile'init.lua'
Heap:	29392
> table: 0x3fff0a38
dofile'init.lua'
Heap:	29432
> table: 0x3fff0bb8
dofile'init.lua'
Heap:	29424
> table: 0x3fff0c68
dofile'init.lua'
Heap:	29336
> table: 0x3fff0a38

```

with sensor attached:
```
node.setonerror(print)
dofile'init.lua'
Heap:	38944
> bad argument #2 to '?' (function expected, got lightfunction)
stack traceback:
	[C]: in function '?'
	?: in function <?:55>
	(tail call): ?
dofile'init.lua'
Heap:	29600
> table: 0x3fff0be0
dofile'init.lua'
Heap:	29600
> table: 0x3fff0a68
dofile'init.lua'
Heap:	29560
> table: 0x3fff0be0
```

with sensor, but error is the same as without (I checked, table is not empty):
```
node.setonerror(print)
dofile'init.lua'
Heap:	38944
> ?:0: attempt to index a nil value
stack traceback:
	?: in function <?:120>
	(tail call): ?
dofile'init.lua'
Heap:	29448
> table: 0x3fff0c68
dofile'init.lua'
Heap:	29176
> table: 0x3fff0a68
dofile'init.lua'
Heap:	29192
> table: 0x3fff0c68
dofile'init.lua'
Heap:	29160
> table: 0x3fff0bb8
```

And I tried latest Lua 5.3 build from dev branch, works ok.

TerryE, 2020-07-01:
@sonaux Natalia, great to see you trying out the new error handling.  Thanks :+1: and to the error:
```
Heap:	38944
> bad argument #2 to '?' (function expected, got lightfunction)
stack traceback:
	[C]: in function '?'
	?: in function <?:55>
	(tail call): ?
``` 
This one is definitely a bug. As part of the changes to makes the C API compatible between 5.1 and 5.3 we "buried" the distinction of `function` and `lightfunction` and this involve changing changing a few references from `ttype(o)` to `basettype(o)` -- that is masking off the high bit in the type which differentiates the  `function` and `lightfunction` sub-type.  It looks like we've missed one. 

I need to double check the cases.  I seem to have missed one.  _**Action** for TerryE_

The `?` chars in the traceback are because the loaded function has its debug info including line numbers stripped out.  See this example to see how this works:
```Lua
> =file.getcontents'x.lua'  -- test file that I'v e previously put
  function f(a) local s=1/a end
  function b(a) local res=f(a); return res end
  function c(a) return b(a) end
  function xx() a = 'hello'c(a) end

>  dofile'x.lua';print(xx())  -- this compiles x.lua with debug info so the traceback has useful info
Lua error: 	x.lua:1: attempt to perform arithmetic on a string value
stack traceback:
	x.lua:1: in function 'f'
	x.lua:2: in function <x.lua:2>
	(tail call): ?
	x.lua:4: in function 'xx'
	stdin:1: in main chunk
	[C]: ?
	[C]: ?
> -- Note that c() does a tailcall which is processed almost like a goto so you loose the line info 
> -- for this frame in the call stack.  Now compile to strip the debug info and retry
> node.compile'x.lua'
> dofile'x.lc';print(xx())
Lua error: 	?:0: attempt to perform arithmetic on a string value
stack traceback:
	?: in function 'f'
	?: in function <?:2>
	(tail call): ?
	?: in function 'xx'
	stdin:1: in main chunk
	[C]: ?
	[C]: ?
> 
```
You can see the unrole.  In your example the top routine tailcall was the 750mSec alarm CB at line 146 with then called `readout(self)` and the error was somewhere in a C api call in this. 

I need to check and document the strip debug options in `luac.cross`.  IMO, unless LFS region space is getting very tight, we should compile LFS modules with full debug as this only increases module size by maybe 20% and has no RAM or runtime impact, but makes debugging a lot easier.  Another _**Action** for TerryE_

532910, 2021-07-01:
Still actual

TerryE, 2021-09-26:
I am not going to work on this, so unless you @532910 are going to provide the fix / PR, this falls into the don't know / won't fix category

closed 2021-09-26 by TerryE