nodemcu/nodemcu-firmware #2901: Long interrupt processing time

StasBronchukov, 2019-09-07:
### Expected behavior
Fast interrupt
### Actual behavior
From the moment a signal arrives to the start of its processing, 0.8522 millisecon pass
### Test code
For the test, I use a logic analyzer.
Pins 4 and 19 are shorted. Pins 32 and 33 are connected to a logic analyzer.
The z () function determines the delay between switching from low to high.
By the z () function we find out the delay between switching from low to high - 
it is 50 microseconds in the aisles, and then it can be ignored.
By the q () function we find out the delay between the arrival of the signal and its processing - 
it amounts to about 0.86 milliseconds, which is extremely high.
```Lua
gpio.config({gpio={4,32,33}, dir=gpio.OUT},{gpio={19}, dir=gpio.IN})

gpio.write(4,1)
gpio.write(32,0)
gpio.write(33,0)

gpio.trig(19,gpio.INTR_DOWN,function() gpio.write (32,1) end)
function q()
gpio.write(33,1)
gpio.write(4,0)
end

function z()
gpio.write(32,1)
gpio.write(33,1)
end
```
I read about such an experiment on the Internet, but micropyton was used there, 
and as far as I remember, it turned out much faster. Also in that article we worked 
directly with registers and got values close to switching low to high pins.
For me, this time is critical.
My task is to count the pulses and the time between them. I decided to try to write 
a module that would work with interrupts directly, but unfortunately I am not very strong 
in this area and after 2 days of trying to figure it out, I decided to ask for smarter people.
What are the possible solutions to this problem?

### Hardware
Esp32

nwf, 2019-09-08:
852 microseconds is not a whole lot of time, figuring that IRQs reach Lua by... triggering a C routine, which posts to a task queue, which is then popped and the Lua interpreter invoked.  You might be able to do better, but it will require that you spend time working in the run-time layer of NodeMCU.

If you need to count events accurately, you will be much better served by use of hardware counters, which may need to be external to the ESP32.

HHHartmann, 2019-09-08:
I assume you are aware of the pulse count module?
https://nodemcu.readthedocs.io/en/dev-esp32/modules/pulsecnt/
Maybe that can help you.

StasBronchukov, 2019-09-08:
It seems that this is what you need. I have not seen this library before. Thank.
By the way, I found this article. In it, when using the register, 550 nanoseconds are obtained.
[https://habr.com/ru/post/448702/](url)

closed 2020-09-17 by stale[bot]