nodemcu/nodemcu-firmware #2903: broke what previously worked

Evonic4, 2019-09-08:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
>>expired

### Actual behavior
init.lua:1: attempt to call field 'create' (a nil value)

### Test code
https://nodemcu.readthedocs.io/en/master/modules/tmr/#tobjstart
`local mytimer = tmr.create()
mytimer:register(5000, tmr.ALARM_SINGLE, function (t) print("expired"); t:unregister() end)
mytimer:start()`

### NodeMCU version
2.2.1-master_20190405

### Hardware
ESP8266 device v3

8<------------------------ END BUG REPORT -------------------------------------

nwf, 2019-09-09:
It's hard to fathom how that could be true, but in any case, 20190405 was a while ago; could you try with a newer release?

HHHartmann, 2019-09-09:
Actually that is the latest release as the current one is not yet announced properly.

Evonic4, 2019-09-09:
need fix

HHHartmann, 2019-09-09:
I just checked on my device and it works (as expected)
@Evonic4 can you give more information about your firmware?
If you built from frightanic or docker please give the startup message.
Or can you please try a frightanic build (https://nodemcu-build.com/) or docker build from clean sources?

Evonic4, 2019-09-10:
and how many modules did you include, perhaps this is due to incompatibility?
I collected 16 modules
 (https://nodemcu-build.com/)

Evonic4, 2019-09-11:
still this error

NodeMCU 0.9.5 build 20150318  powered by Lua 5.1.4
Waiting ...
lua: init.lua:2: attempt to call field 'register' (a nil value)
>

nwf, 2019-09-11:
@Evonic4: No matter how many times you say "need fix", any fix will not arrive faster.  If you are so desperate for a fix, I can be hired at reasonable professional contracting rates; you're welcome to contact me out of band to arrange such.

You have not provided anyting useful in terms of feedback.  Please, for example, articulate *which* 16 modules you selected in the node builder.

`20150318` is a *very*, *very* long time ago and we certainly no longer support that version.  Are you sure you have flashed your device with the version you claimed in the original report, (`2.2.1-master`)

Evonic4, 2019-09-11:
here is the feedback to you

1. modules
cron crypto file gpio http i2c net node pcm pwm rtctime spi **tmr** uart websocket wifi.

2. last release used
https://github.com/nodemcu/nodemcu-firmware/releases
Latest release **3.0-master_20190907**    KARL!

3. 
No need to relate to the project through the sleeves just because it is free. The project is used and needed by people!
If you want to earn money, then you are not here, here are true connoisseurs of free art.
And I don’t want to slap you in the dirty ass, just understand: no need to shit in free projects, they are no less important than paid ones.
I hope the community will present a solution or roll back to a working version and start development again from it.

HHHartmann, 2019-09-11:
From the startup message you supplied it is clearly visible, that the image you are using is ancient (from 2015).
Looking at the code of that version, there has not been any create at that time. So that's why your code fails.

This is most likely the default image which your device was shipped with. You cannot build it with the online service.

So you need to get that image flashed on your device first.

Current versions show a startup message similar to this:
```
NodeMCU 3.0.0.0 
	branch: dev
	commit: 0b90495517c8e2b34dbb3a2a99455ac3c4f05f1c
	release: 2.0.0-master_20170202 +437
	release DTS: 
	SSL: false
	build type: float
	LFS: 0x40000
	modules: adc,bit,color_utils,crypto,dht,encoder,enduser_setup,file,gpio,http,i2c,l3g4200d,mdns,net,node,ow,pipe,pwm,rtcfifo,rtcmem,rtctime,sjson,sntp,spi,struct,tmr,uart,wifi,wifi_monitor,ws2812,ws2812_effects
 build created on 2019-08-14 19:56
 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```

And by the way, I would appreciate you staying polite in your comments. nwf has committed quite some time to this project. Pro bono that is! 
So he has some karma here, which you don't.

Evonic4, 2019-09-11:
I very much apologize if I was a little polite, this happened spontaneously and in a fit it is not clear where the critical substance came from in my display of literature)

I need to make a video about how to make a set of modules at https://nodemcu-build.com
I think this is superfluous and users should not do this to get you comprehensive information, can check this resource for collecting your devops?

Evonic4, 2019-09-11:
tried to assemble again
nodemcu-master-16-modules-2019-09-11-16-16-12-55-integer

16 modules: cron crypto file gpio http i2c net node pcm pwm rtctime spi tmr uart websocket wifi.

all the same

code
mytimer = tmr.create (0)
mytimer: register (5000, tmr.ALARM_SINGLE, function (t) print ("expired"); t: unregister () end)
mytimer: start (0)

conclusion
lua: 1: attempt to call field 'create' (a nil value)

earlier in the assembly:
nodemcu-master-16-modules-2018-09-12-12-17-26-55-integer

everything worked
I no longer have the opportunity to engage in testing, because it’s busy and I’ll write it myself better...

closed 2019-09-11 by HHHartmann

nwf, 2019-09-11:
There is no bug here; reporter is likely not flashing module correctly.

Full transcript of verification is as follows:
```
drogon% wget https://nodemcu-build.com/builds/nodemcu-master-16-modules-2019-09-11-20-22-59-integer.bin                                                 
--2019-09-11 21:28:42--  https://nodemcu-build.com/builds/nodemcu-master-16-modules-2019-09-11-20-22-59-integer.bin
Resolving nodemcu-build.com (nodemcu-build.com)... 217.26.52.40
Connecting to nodemcu-build.com (nodemcu-build.com)|217.26.52.40|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 634880 (620K) [application/octet-stream]
Saving to: ‘nodemcu-master-16-modules-2019-09-11-20-22-59-integer.bin’

nodemcu-master-16-modules-2019-09-11-20-22-59-integer.bin                       100%[====================================================================================================================================================================================================>] 620.00K  2.24MB/s    in 0.3s    

2019-09-11 21:28:42 (2.24 MB/s) - ‘nodemcu-master-16-modules-2019-09-11-20-22-59-integer.bin’ saved [634880/634880]

drogon% sha1sum nodemcu-master-16-modules-2019-09-11-20-22-59-integer.bin                                                                               
b1e452f7a8b2e1666f100809fb89720ffb10a42d  nodemcu-master-16-modules-2019-09-11-20-22-59-integer.bin
drogon% ~/ee/esp/esptool/esptool.py write_flash --flash_size 32m --flash_mode dio --verify 0x0 nodemcu-master-16-modules-2019-09-11-20-22-59-integer.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.4.1-dev
Trying /dev/ttyUSB0 ... 
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:3d:72:25
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0240
Compressed 634880 bytes to 423887...
Wrote 634880 bytes (423887 compressed) at 0x00000000 in 37.3 seconds (effective 136.0 kbit/s)...
Hash of data verified.

Leaving...
Verifying just-written flash...
(This option is deprecated, flash contents are now always read back after flashing.)
Flash params set to 0x0240
Verifying 0x9b000 (634880) bytes @ 0x00000000 in flash against nodemcu-master-16-modules-2019-09-11-20-22-59-integer.bin...
-- verify OK (digest matched)
Hard resetting via RTS pin...
drogon% kermit -l /dev/ttyUSB0
C-Kermit 9.0.302 OPEN SOURCE:, 20 Aug 2011, for Linux+SSL+KRB5 (64-bit)
 Copyright (C) 1985, 2011,
  Trustees of Columbia University in the City of New York.
Type ? or HELP for help.
(/home/nwf/ee/esp/tmp/idiot/) C-Kermit>set baud 115200
/dev/ttyUSB0, 115200 bps
(/home/nwf/ee/esp/tmp/idiot/) C-Kermit>set flow-control none
(/home/nwf/ee/esp/tmp/idiot/) C-Kermit>set carrier-watch off
(/home/nwf/ee/esp/tmp/idiot/) C-Kermit>connect
Connecting to /dev/ttyUSB0, speed 115200
 Escape character: Ctrl-\ (ASCII 28, FS): enabled
Type the escape character followed by C to get back,
or followed by ? to see other options.
----------------------------------------------------
```
press reset, and then...
```
NodeMCU 3.0.0.0 
	branch: master
	commit: 310faf7fcc9130a296f7f17021d48c6d717f5fb6
	release: 3.0-master_20190907
	release DTS: 201909070945
	SSL: false
	build type: integer
	LFS: 0x0
	modules: cron,crypto,file,gpio,http,i2c,net,node,pcm,pwm,rtctime,spi,tmr,uart,websocket,wifi
 build 2019-09-11 20:23 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
lua: cannot open init.lua
> mytimer = tmr.create()
> =mytimer
userdata: 3fff1540
> mytimer:register(5000, tmr.ALARM_SINGLE, function (t) print("expired"); t:unregister() end)
> mytimer:start()
> expired
```

I will not be holding my breath for either an apology or that "better" version.

Evonic4, 2019-09-12:
different languages)

[image]

[image]

[image]

[image]

Evonic4, 2019-09-12:
dibils...