nodemcu/nodemcu-firmware #2907: crypto.fromHex feature request

georeb, 2019-09-09:
### Missing feature

`crypto.fromHex`

The missing counterpart to `crypto.toHex`

Provides an ASCII char representation of a (hex) Lua string. Each hex byte in the input string is represented as two hex characters.

For example: "48656c6c6f576f726c64"

### Justification
The use of encrypt and decrypt is for purposes where the original value is required at a later date. Using toHex is very useful for storing in the file structure, but when this is later required, the hex data string must be converted back into the original char string.

### Workarounds
Currently using this code:

```Lua
--First encrypt the data

    local hexstring = crypto.toHex(crypto.encrypt("AES-CBC","mykey","data"))

    file.open("mydata.txt","w+")
    file.write(hexstring)
    file.close()

--Then decrypt the data WITHOUT the use of 'crypto.fromHex'

    file.open("mydata.txt","r")
    local dat = file.read()
    file.close()

    local output

    for i=1,#dat,2 do
        if not output then
            output=string.char("0x"..string.sub(dat,i,i+1))
        else
            output=output..string.char("0x"..string.sub(dat,i,i+1))
        end
    end

    return crypto.decrypt("AES-CBC", "mykey", output):match("(.-)%z*$")

--Or decrypt the data WITH the use of 'crypto.fromHex'

    file.open("mydata.txt","r")
    local output = crypto.fromHex(file.read())
    file.close()

    return crypto.decrypt("AES-CBC", "mykey", output):match("(.-)%z*$")
```

nwf, 2019-09-28:
I think we should get deprecate then remove `crypto.toHex` and `crypto.toBase64` and redirect to the `encoder` module (https://nodemcu.readthedocs.io/en/master/modules/encoder/), which provides the inverse functions as well.

georeb, 2019-10-02:
Didn't spot the `encoder.fromHex` function. Thanks!

closed 2019-10-02 by georeb