nodemcu/nodemcu-firmware #2909: Fix startup banner issues

marcelstoer, 2019-09-12:
This sample startup banner reveals we've got a couple of issues:

```
NodeMCU 3.0.0.0 
	branch: dev
	commit: 0b90495517c8e2b34dbb3a2a99455ac3c4f05f1c
	release: 2.0.0-master_20170202 +437
	release DTS: 
	SSL: false
	build type: float
	LFS: 0x40000
	modules: adc,bit,color_utils,crypto,dht,encoder,enduser_setup,file,gpio,http,i2c,l3g4200d,mdns,net,node,ow,pipe,pwm,rtcfifo,rtcmem,rtctime,sjson,sntp,spi,struct,tmr,uart,wifi,wifi_monitor,ws2812,ws2812_effects
 build created on 2019-08-14 19:56
 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
```

- "NodeMCU 3.0.0.0" - having 4 digits here is superfluous I argue.
- "NodeMCU 3.0.0.0" vs. "SDK 3.0.1-dev" - the since our versions always claim to represent the SDK version it should be 3.0.1 rather than 3.0.0.0. We have a mix of concepts here.
- "release: 2.0.0-master_20170202 +437" - on `dev` we can't really use the [output `git describe`](https://github.com/nodemcu/nodemcu-firmware/blob/master/tools/update_buildinfo.sh#L7). It appears that if you have a long-running (feature) branch Git can't find it's way back to the recent tag. We would have to delete and recreate `dev` after every release.

HHHartmann, 2019-09-16:
> Does `git commit` behave any better if you use `--all` instead of `--tags`?

Not really. For master it remains the same. For other branches it just basically prints the branch name.

nwf, 2019-09-16:
Maybe what we should do, and I recognize this is kind of terrible, is merge `dev` to `master`, tag the release, and then `merge` master back to `dev`.  That should let `git describe` figure out what's going on.  By way of example, I rebased my local `dev` branch atop the official `dev` and a merge of `master` and then ran `git describe` and got `3.0-master_20190907-12-gae9b8179`.

marcelstoer, 2019-09-16:
> merge `dev` to `master`, tag the release, and then `merge` master back to `dev`

Yes, that'll have the same effect as re-creating `dev` (by delete and create) after tagging. That feels even more terrible but it would avoid potential merge commits on `dev`. You'd get those if we had commits on `master` that are not on `dev` (we do). That is unless you rebase first before you merge - and deal with the conflicts.

It's true that we should have made sure that `master` and `dev` are identical after every release. I can amend https://github.com/nodemcu/nodemcu-firmware/blob/master/CONTRIBUTING.md#handling-releases by this:
- In your local clone (not fork!) do
  - `git checkout master && git pull && git checkout dev && git merge master --ff`
  - fix any potential merge conflicts
  - verify that `git describe` displays the tag you just created
  - git push

nwf, 2019-09-16:
We can, rather than truly merging back to `dev`, just graft on `master` as another parent to a post-fork commit.  I forget the command to do that, but it's absolutely possible.

HHHartmann, 2020-06-28:
Since we did that now and dev branch release messages is something like "3.0-master_20200610 +15" we can close this.
The other issues with versioning does not only affect the startup banner but is deeper I think and should have its own issue.

The 3.0.0 comes from ESP_SDK_VERSION_STRING which is defined in the SDK. Can't really do anything about that. The 3rd "0" is defined and added in user_version.h - could remove thet
But that should be done in its own issue and with greater consensus.

So closing this one.

closed 2020-06-28 by HHHartmann