nodemcu/nodemcu-firmware #2916: RFC: rework LED strip modules

HHHartmann, 2019-09-16:
Currently we have several LED strip modules for different Hardware.
While some only support writing out the data to the strip there are several features for the WS2812. I would like to make these features available to all led strips.

1. It would be nice to abstract the buffer used for the WS2812 to be HW independent and adapt all HW modules to be able to work with these buffers.

2. I would also like to address the different color orders of the strips. AT least RGB and GRB are needed, maybe more. The order would be given when writing to a strip, or when initializing it. (whatever we define)

3. Since usually colors are given in RGB I would like to move the color definition in `color utils Module` and `WS2812 effects Module` to RGB instead of GRB. So for the logic in the buffers we would use RGB

Or course a white channel will also be available, Just wanted to keep it short here.

I would especially like to hear your opinion on 3. since that would probably have to be a breaking change, or we could duplicate functions and then have `color_utils.hsv2grb` and `color_utils.hsv2rgb`.

### Justification
Make available the buffer manipulation routines for all LED modules, not only WS2812

### Workarounds
Do it yourself completely or recode the buffer to a string usable by the other modules.
No workaround for the `WS2812 effects Module`.

nwf, 2020-05-23:
I have not yet tested beyond "it compiles", and it is premature to PR, my `pixbuf` split.  If someone wants to grab https://github.com/nwf/nodemcu-firmware/tree/led-strip-refactor and try it out, tho', that'd be grand.  I'll try to keep it up to date when I put more time into it, but right now I'm getting up out of this chair. :)

HHHartmann, 2020-05-24:
I recently also stumbled across LED strips supporting WWA (WarmWhite/ColdWhite/Amber) so how can/should we deal with that?

 1. Ignore it, so people can still use the buffer but have to remember the mapping RGB <-> WWA
 2. Have the buffer "know" what type it is (RGB GRB RGBW WWA) and have effects and color utils check if they can handle it

nwf, 2020-05-24:
Hm; that's an interesting question.  I was, relatedly, pondering whether we should have a "generic" pixbuf that's basically just a string of bytes to be manipulated ("1 byte per LED").

How about something like this?
```
enum pixbuf_type {
  PIXBUF_TYPE_WILD /* just bytes */,
  PIXBUF_TYPE_GRB /* 3-color: green, red, blue order */,
  PIXBUF_TYPE_GRBW /* 4-color: GRB + white */,
  PIXBUF_TYPE_WWA /* 3-color: warm white, cold white, amber */
  // ...
};

typedef struct pixbuf {
  enum pixbuf_type type;
  size_t npix;
  uint8_t values[0];
} pixbuf;

static bool pixbuf_type_compat (pixbuf *lhs, pixbuf *rhs, enum pixbuf_type *out) {
  if (lhs->type == PIXBUF_TYPE_WILD) {
    if (out) {
      *out = rhs->type;
    }
    return true;
  }

  if (rhs->type == PIXBUF_TYPE_WILD) {
    if (out) {
      *out = lhs->type;
    }
    return true;
  }

  if (lhs->type == rhs->type) {
    if (out) {
      *out = lhs->type;
    }
    return true;
  }

  return false;
}

static bool pixbuf_size_bytes (pixbuf *p) {
  switch (p->type) {
  case PIXBUF_TYPE_WILD: return p->npix;
  case PIXBUF_TYPE_GRB: return p->npix * 3;
  case PIXBUF_TYPE_GRBW: return p->npix * 4;
  case PIXBUF_TYPE_WWA: return p->npix * 3;
  // ...
  default: return 0;
  }
}
```

The various functions can use `pixbuf_type_compat` for pairwise checks and check equality of `pixbuf_size_bytes()` as required.

closed 2021-01-07 by marcelstoer