nodemcu/nodemcu-firmware #2918: Problem with makefile

poorandunlucky, 2019-09-19:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

For `make` to build targets without errors.

### Actual behavior

`make` hangs-up on 334; `make -i` does manage to build (complete? valid?) binaries.

```
extracting: ESP8266_NONOS_SDK-e4434aa730e78c63040ace360493aef420ec267c/third_party/include/user_config.h
PRUNE libmain.a libc.a
Makefile:334: recipe for target '/home/poorandunlucky/git/esp8266/nodemcu/sdk/.pruned-3.0-e4434aa' failed
make: *** [/home/poorandunlucky/git/esp8266/nodemcu/sdk/.pruned-3.0-e4434aa] Error 2
```

With `make -i`:

```
make: Entering directory '/home/poorandunlucky/git/esp8266/nodemcu'
PRUNE libmain.a libc.a
/bin/sh: 1: Syntax error: "(" unexpected

Files /tmp/buildinfo.h and app/include/buildinfo.h differ
```

### Test code

git clone -o nodemcu https://github.com/nodemcu/nodemcu-firmware.git nodemcu
cd nodemcu
branch local
checkout local
make[ -i ]

### NodeMCU version

3.0-master_20190907

### Hardware

N/A

8<------------------------ END BUG REPORT -------------------------------------

HHHartmann, 2019-09-19:
try `git clone --recurse-submodules ...`

poorandunlucky, 2019-09-19:
I used --recursive, but omitted it from my pseudo-commands the example above...

poorandunlucky, 2019-09-19:
So out of curiosity, I tried using `--recurse-submodules` instead of the (I guess deprecated but kept-for-compatibility) `--recursive, and it yielded the same results...  `git clone` fetched nodemcu, ucg and u8g2, `make` fetched the toolchain from pfalcon, and `make` hung-up on line 334 as above, and that line is `rm -f $(SDK_DIR)/lib/liblwip.a $(SDK_DIR)/lib/libssl.a $(SDK_DIR)/lib/libmbedtls.a`.

TerryE, 2019-09-21:
Gregor if this relates to `buildinfo.h` then this one is down to you, old chap.  If you find a fix then post it here as a diff if you don't want the hassle of raising a PR -  I will roll it into my next commit.

HHHartmann, 2019-09-21:
@TerryE I just had a clean build of current master. I only omitted the 
```
git branch local
git checkout local
```
steps.

@poorandunlucky does your dev environment contain windows?

poorandunlucky, 2019-09-21:
@HHHartmann It does, actually, but I don't see how it could be its fault...  I'm using the Windows Subsystem for Linux (WSL) which allows you to run a full command-line Linux distribution from the PowerShell command line interface, Ubuntu, in my case, but other distros are available in the Microsoft app store.

I can't say I understand how the WSL works, though, or even Ubuntu, for that matter, I'm more a FreeBSD person, but from the userland it seems like a normal Linux distribution, I can't say I've ever had to jump any hoops or do anything special because it's running in WSL...  I can't say I understand the makefile, either...  I've never programmed anything complex enough to warrant their usage, so never really got to it...  I don't understand why those files would get deleted, seems like crypto libs, too, so...

HHHartmann, 2019-09-21:
recently we have had trouble with windows lineends several times.
Do you do the checkout under windows powershell  or inside the WSL?
If not WSL please try from there. Use `git status` inside WSL. Every file should be marked as changed if the lineends don't match.

poorandunlucky, 2019-09-21:
No, it’s true that git is set to normalize line endings to \n on my system, but:

`poorandunlucky@M6500:~/git/esp8266/nodemcu$ git status
On branch local
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

        modified:   app/include/u8g2_displays.h
        modified:   app/include/u8g2_fonts.h
        modified:   app/include/user_config.h
        modified:   app/include/user_modules.h
        modified:   app/include/user_version.h

no changes added to commit (use "git add" and/or "git commit -a")`

Also, for the sake of conversation…  PowerShell is what replaces the old Dos 6.22-based command line interpreter, it’s not like SSH… it’s kinda like an OO shell…  I’ve only used its expanded features a few times though, the commands are hard to remember and it’s hard to navigate…  It’s probably a lot of help, and a welcome addition to Windows people, though….

HHHartmann, 2019-09-22:
I'm on windows, so I know powershell, but haven't made friends with it either.
Im using marcels docker image to build my firmware. 

I wonder why you have changed files on your build. Maybe it's related to this different configuration that your build fails.

HHHartmann, 2019-09-22:
@TerryE terry I don't think that it is related to `buildinfo.h` as you suggested.
the main error is related to removing some libraries. Even if the `sh (` error is caused by my script it would not break the build. So I'm out.

@poorandunlucky It just came to my mind that you could have some sort of access right problems. But that's just a wild guess.

poorandunlucky, 2019-09-22:
Oh those are the configuration files, you're supposed to change those, but even on the master branch it's the same error, I just wanted to prove having checked about the line endings...

I'm just not sure what the problem is, I don't know what that PRUNED stuff does, and make seems to clean-up after the error?  I don't know what that part of the makefile is for, either, like what purpose it serves...  I was hoping someone else who knows what they're doing would have an easier time figuring out what's going on...

It's there a way to turn on verbosity to maximum?  Are there comments about this target?  I don't think it was in the last release, or at least I haven't had problems with it before...  I could build the last release just fine, did so many, many times...

I've upgraded one of my MCUs to 3.0.0, and it boots and behaves just fine, just have to set the partition map for the LFS there, but it seems to work...  Not really using TLS or networking at this time, though...

If there's anything I can do to try and figure out what's going on, please let me know, and I'll perform commands, and return log files as desired...  Also since it's crypto-related, maybe try to find out who put it there?  Maybe somebody took an opportunity to create a security hole while they were at their friend's place when he had the code open, or something?

I'll try to poke around it a bit more later, the other day I think I got it working, but I was very tired...  I logged-out, and back in, and I might've gotten it to work, but it only worked once, not twice in a row, and not after logging out and back in again, so I'm not 100% I didn't put a -i after the "make"...  I guess it's good learning anyway...  : ]

poorandunlucky, 2019-09-23:
I think I got it...  The problem is the Makefile's syntax...  Whoever wrote it used $(VAR) instead of ${VAR}, and $(VAR) is (exactly?) the same as \`VAR\`, so it tries to execute commands...  I think one of the variables tried to execute something in my $PATH...

IMO this is insecure, and obviously unpredictable...  A very bad habit, at any rate...

The files that it tries to delete on Makefile:334 to save space do get deleted on disk, so the problem is right after, probably the $(summary) command...

If I just leave the basic stuff in my $PATH, like /bin:/sbin:/usr/local/bin:/usr/local/sbin, etc, make runs just fine, no errors, but you can't just go tell people what they can, and can't have in their $PATH... that's insane....  What if someone has a script named "summary" in /usr/local/sbin that prints a summary of the systems statuses?

I apologize if I may seem a bit salty here, but I was hoping this would be tasty, and epic journey of learning, but it appears to be a syntax error, and I'm a bit dissapoint, and kinda frustrated...  : \

TerryE, 2019-09-23:
@poorandunlucky Patrick,  thanks for drilling into this. Much appreciated.

> Whoever wrote it used $(VAR) instead of ${VAR}, and $(VAR) is (exactly?) the same as `VAR`, so it tries to execute commands...

Which make engine are you using?  The NodeMCU scripts are optimised around GNU make.  The *standard* method in GNU make is to use the `$(x)` form though the `${x}` also works for many uses (see [Ref: 6.1 Basics of Variable References](https://www.gnu.org/software/make/manual/html_node/Reference.html#Reference) etc.)

poorandunlucky, 2019-09-26:
GNU Make 4.1...

I thought Make was just a collection of shell scripts (kinda like FreeBSD's rc scrpts), didn't think there was a binary interpreter behind it...  Then I guess there was just something in my $PATH that made that happen, but I think it should be documented behaviour that $(VAR) allows program execution even in Make...

Depending how big the diff is between the 2.1.1 and 3.0.0 Makefiles I might try to figure out more precisely what was going on... as I said I don't know much about Makefiles, and I guess it's time I learn, but I have a bit of cleaning-up to do first, which actually involves improving my workflow, so...  Guess I got a date  : (

But yeah, if anyone has this error on Ubuntu under the Windows Subsystem for Linux (WSL) you should clean-up your $PATH as it includes your Windows environment as well...  I guess they wanted a closer integration of the two than most people actually end-up using in real life...

Thanks to everyone who participated for their time  : ]

closed 2019-09-26 by poorandunlucky