nodemcu/nodemcu-firmware #2920: Why LUA_USE_MODULES_WIFI_MONITOR in user_config.h?

marcelstoer, 2019-09-21:
By coincidence I found out just now that we're having `LUA_USE_MODULES_WIFI_MONITOR` in `user_config.h` at https://github.com/nodemcu/nodemcu-firmware/blob/master/app/include/user_config.h#L138. This to me looks like a mistake as all modules should be enabled/disabled through `user_modules.h` exclusively, no (#2913 aside)?

HHHartmann, 2019-09-21:
This seems to be a mistake, especially since it is also included in user_modules.h
Was introduced by https://github.com/nodemcu/nodemcu-firmware/commit/4f21224d563b5481362d0eda0045908cf485fb27#diff-1a214f0fbc7aa550af9869070422db14 where a lot changed in that file, so maybe an oversight to not remove it before commit.
@TerryE ?

TerryE, 2019-09-21:
Definitely. The wedding place for it and easy to fix.

commit 81e213ac (2019-09-21): Remove superfluous module def

Fixes #2920
commit 08ea0c8d (2020-06-09): Remove superfluous module def

Fixes #2920
closed 2019-09-21 by marcelstoer