nodemcu/nodemcu-firmware #2922: Slow display ILI9341 UCG

StasBronchukov, 2019-09-22:
### Expected behavior
Fast display
### Actual behavior
The display is slow. Is this normal or is it possible to speed up the work?
https://youtu.be/0XaDDxurCfY
### NodeMCU version
Branch 51912d5
### Hardware
ESP32

StasBronchukov, 2019-09-22:
Sorry, branch dev-esp32, commit 51912d5

f4grx, 2019-09-26:
Same behaviour here.
SPI speed seems to be hardcoded to 10 MHz in components/platform/ucg_nodemcu_hal.c
I changed it to 40 MHz and got a faster display refresh.

It is not currently configurable, but it could be done by changing the ucg lua module, for example by passing a SPI DEVICE instead of a SPI BUS to the screen driver init functions.
This would allow user's lua to setup the clock speed.

f4grx, 2019-10-03:
After looking a bit at this: This is too much ram for the esp32. This lib is just slow. It helps to only draw small zones.

closed 2020-10-11 by stale[bot]