nodemcu/nodemcu-firmware #2924: ESPNOW

poorandunlucky, 2019-09-23:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

ESPNOW

### Justification

Well it's one of the big selling points for the ESP MCUs...  Oftentimes wifi is just overkill, or you want to implement an ecosystem of devices that doesn't use wifi per se, or that communicate between each other, and only a base station connects to wifi...  idk, really, it's just one of the things I was hoping to be able to play with, and I wanted to know if there were any plans on implementing it in NodeMCU...

### Workarounds

Well you can use the SDK, but you have to learn a whole new firmware, and it's not as pleasant to use as NodeMCU...

8<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2020-07-31:
https://stackoverflow.com/questions/63168628/is-there-a-way-to-use-esp-now-protocol-using-lua linking here.

closed 2021-09-07 by stale[bot]

jmattsson, 2024-03-07:
@poorandunlucky This is "a bit" late for this issue, but I've just raised a PR with a new `espnow` module, if you're interested in testing it.

poorandunlucky, 2024-03-07:
well that's very nice...

how hard was it to put the code in a module?

i really love the Lua firmware, I think it's very elegant...

Patrick Dorion
📱 +1 438 345-4575

jmattsson, 2024-03-08:
It was pretty straight forward to map the API, it was mainly a matter of time and "grunt work".

poorandunlucky, 2024-03-08:
Yeah, that's pretty much what I gathered...  When I saw the project hadn't
been updated in quite some time, I took another look at the codebase,
especially the modules to figure out if I wanted to keep using the
firmware, or move on to something else...  I really like the simplicity,
and elegance of it, and being able to run stuff from SPIFFS to save RAM, i
don't know if that's unique to this, or not, but when I saw how simple the
modules really were, I decided there was no harm in sticking with it, and
worst case it would just help transition me to the ESP Non-OS IDF framework
that I originally wanted to use, but that was too complicated, and still
is, to some extent...

Thanks for letting me know, and for helping the project along!  Nodemcu is
definitely the most elegant piece of software I ever came across...

Patrick Dorion
📱 +1 438 345-4575

PR #3644 (2024-03-27): Added espnow module with documentation.
Fixes #2924, at least for the ESP32 range.

- [x] This PR is for the `dev` branch rather than for the `release` branch.
- [x] This PR is compliant with the [other contributing guidelines](/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

This is a pretty straight-forward module implementation of the regular ESP-NOW API. Tested on ESP32-C3, but should be board agnostic.

@pjsg, quick review if you have the time, please?
@marcelsto [... cut]