nodemcu/nodemcu-firmware #2925: adc.readvdd33()

lmercucci, 2019-09-23:
<------------------------ BUG REPORT -----------------------------------------
adc.readvd33() not working.... 
in my init.lua have:

if adc.force_init_mode(adc.INIT_VDD33) then
  node.restart()
  return -- don't bother continuing, the restart is scheduled
end
print("System voltage (mV):", adc.readvdd33(0))

bud

=adc.readvdd33()  
=adc.readvdd33()  
65535
<------------------------ END BUG REPORT -------------------------------------

<------------------------ FEATURE REQUEST ------------------------------------
<------------------------ END FEATURE REQUEST --------------------------------

marcelstoer, 2019-09-25:
According to our documentation (and the code) `readvdd33()` doesn't take any parameters. Yet, the code above and in the documentation calls it with 0.

HHHartmann, 2019-09-25:
For 
```
NodeMCU 3.0.0.0 
	branch: dev
	commit: 0b90495517c8e2b34dbb3a2a99455ac3c4f05f1c
```
calling it with no argument yields the same result.

But for 
```
NodeMCU 2.2.0.0 built with Docker provided by frightanic.com
	branch: dev
	commit: 5514475ab0b152eaa994495e6d034123ae811786
```
it returns 3015 which is what I would expect

TerryE, 2019-09-26:
Doing a quick `git diff 5514475a 0b904955 -- ../modules/adc.c` shows that `adc_sample()` hasn't changed (except removing a trailing whitespace on the return).  It's essentially a wrapper for `system_get_vdd33()` which only works if byte 107 of esp_init_data_default.bin is set as 0xFF, and this in turn breaks the read() function.  
`xxd -s 107  -l 1  sdk/esp_iot_sdk_v3.0-e4434aa/bin/esp_init_data_default_v05.bin` returns 0, so it shouldn't work by default.  If we want it to work by default we need to override byte 107.

marcelstoer, 2019-09-26:
> If we want it to work by default we need to override byte 107.

Hhm, that's what `adc.force_init_mode()` i.e. [adc_init107()](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/modules/adc.c#L28) does - or used to do. However, `git blame` shows _it_ hasn't changed in years, either.

If the test script keeps rebooting the device it means that `adc_init107()` detects that byte 107 isn't 0xFF but it fails to overwrite it.

HHHartmann, 2019-09-26:
`adc.force_init_mode` returns true for the first call only.

TerryE, 2019-09-26:
Oops.  Definitely  bug.  I missed [adc.c:L132](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/adc.c#L32).  The Phy Data sector is no longer at END-4.  We need to read its address from the PT
```
  partition_item_t pt = {0,0,0};
  system_partition_get_item(SYSTEM_PARTITION_PHY_DATA, &pt);
```

TerryE, 2019-11-17:
I have fixed this in my working directory and will add this fix to my #2912 next commit.

Michal78S, 2019-12-13:
Please clarify this: adc.readvdd33(0) returns system voltage in milivolts, but adc.read(0) returns what?
Voltage on ADC pin in milivolts or direct convertion resulst in range of 0-1024?

ildar, 2019-12-13:
>
>  adc.read(0) returns what?
> Voltage on ADC pin in milivolts or direct convertion resulst in range of
> 0-1024?
>
Both are the same. That's because adc measures only up to 1000 millivolts.

Michal78S, 2019-12-13:
Are You siure?
I got as below:
mV  | ADC
---- |-----
0     | 14
100 | 114
500 | 525
963 | 1000
986 | 1024
So 14 is ADC offset, and then when substracted offset I got exacly 977 for 1V. It is the same when divide 1000mV/1024.

jmd13391, 2019-12-14:
@Michal78S The ESP8266 ADC has inherent stability issues as a result of an internal hardware design flaw. You may find some of [2864](https://github.com/nodemcu/nodemcu-firmware/issues/2864#issuecomment-520437766) and [esp8266/Arduino:2070](https://github.com/esp8266/Arduino/issues/2070#issuecomment-224891523) relevant.

rptnscz, 2020-01-05:
> According to our documentation (and the code) `readvdd33()` doesn't take any parameters. Yet, the code above and in the documentation calls it with 0.

There are two examples in the doc https://nodemcu.readthedocs.io/en/master/modules/adc/
The example in adc.force_init_mode() chapter shows 'readvdd33(0)' with '0' as a parameter.
But in adc.readvdd33() the syntax is without any parameters. 

Anyway both doesn't work in the branch master. 

@TerryE: how can i checkout your fix of the bug? I guess there are no change in adc.c module.

marcelstoer, 2020-01-05:
@rptnscz maybe this one https://github.com/nodemcu/nodemcu-firmware/pull/2912/commits/32e161b131400a0cd215fddd50733d26dd98ff93#diff-1925935f8d24083ba3e86383f92b7ea8 It's a bit unfortunate that many fixes went into that branch (rather than dev) but that's just how it is. Please be patient.

TerryE, 2020-05-29:
Fixed in #3075

closed 2020-05-29 by TerryE