nodemcu/nodemcu-firmware #2926: dev-esp32: UCG modules "does not work"

f4grx, 2019-09-26:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
I should be able to use the ucg module

### Actual behavior
I am not

### Test code
```Lua
bus = spi.master(spi.HSPI, {sclk=14, mosi=13})
disp = ucg.ili9341_18x240x320_hw_spi(bus, 27, 26, 25)
```
### NodeMCU version
dev-esp32 commit d20778ed0930856e39d4223b1d6ace5dc551d582

### Hardware
Using a wroom-32 DevkitC
TJCTM24024-SPI (ILI9341) screen module on the above mentioned pins
ucg and correct screen driver activated in config

8<------------------------ END BUG REPORT -------------------------------------

I see the module is available and populated

```Lua
> for k,v in pairs(ucg) do print(k,v) end
"ili9341_18x240x320_hw_spi"	lightfunction: 0x400d6f7c
"font_7x13B_tr"	userdata: 0x3f40f275
"font_helvB08_hr"	userdata: 0x3f40eda1
"font_helvB10_hr"	userdata: 0x3f40e704
"font_helvB12_hr"	userdata: 0x3f40df54
"font_helvB18_hr"	userdata: 0x3f40d46e
"font_ncenB24_tr"	userdata: 0x3f40c523
"font_ncenR12_tr"	userdata: 0x3f40be49
"font_ncenR14_hr"	userdata: 0x3f40b57c
FONT_MODE_TRANSPARENT	1
FONT_MODE_SOLID	0
DRAW_UPPER_RIGHT	1
DRAW_UPPER_LEFT	2
DRAW_LOWER_RIGHT	8
DRAW_LOWER_LEFT	4
DRAW_ALL	15
__metatable	romtable: 0x3f404c68
```

however I cannot access the fields:
```Lua
print(ucg.ili9341_18x240x320_hw_spi)
nil
print(ucg["ili9341_18x240x320_hw_spi"])
nil

disp = ucg.ili9341_18x240x320_hw_spi(bus, 27, 26, 25)
stdin:1: attempt to call field 'ili9341_18x240x320_hw_spi' (a nil value)
stack traceback:
	stdin:1: in main chunk

```
I dont understand why the key is present in pairs() but not when accessed directly.

Note that I am fully able to use other modules. I have tested: ow, dht, mqtt, wifi, without any problem.
This issue seems specific to ucg.

f4grx, 2019-09-26:
I see that the ucg module defines __index in the module's table, which is not done by ow, for example.

HHHartmann, 2019-09-26:
The '"' around ili9341_18x240x320_hw_spi look strange. The print(k,v) does not ad them.
So maybe try

``` Lua
print(ucg["\"ili9341_18x240x320_hw_spi\""])
```

Non't really know much about the configuration of ucg but it seems as if the configuration is wrong or some script working with the config file, not removing the '"' at some stage.

f4grx, 2019-09-26:
OMG, you are right:
```Lua
>print(ucg["\"ili9341_18x240x320_hw_spi\""])
lightfunction: 0x400d71ec
```Lua

f4grx, 2019-09-26:
Here is a fix. Thanks a lot for this finding, @HHHartmann !
This should be applied to the dev-esp32 branch:

```diff
diff --git a/components/modules/ucg.c b/components/modules/ucg.c
index 28d27e4..70cc160 100644
--- a/components/modules/ucg.c
+++ b/components/modules/ucg.c
@@ -734,7 +734,7 @@ LROT_END(lucg_display, NULL, 0)

 LROT_BEGIN(lucg)
 #undef UCG_DISPLAY_TABLE_ENTRY
-#define UCG_DISPLAY_TABLE_ENTRY(binding, device, extension) LROT_FUNCENTRY( #binding, l ## binding )
+#define UCG_DISPLAY_TABLE_ENTRY(binding, device, extension) LROT_FUNCENTRY( binding, l ## binding )
   UCG_DISPLAY_TABLE

   // Register fonts
```

HHHartmann, 2019-09-26:
@f4grx also great finding the place in the code. 
Care to do the PR yourself?

f4grx, 2019-09-26:
This is the commit that broke UCG. The lua function names were strings before this change, and the stringification operator in the C macro was forgotten where it should have been removed.

https://github.com/nodemcu/nodemcu-firmware/commit/eaac369dece9ab74a736e8b42a21a4a655b3fe33

u8g2 has the same problem, which is now included in my previous patch.

PR #2927 (2019-09-27): Fix the ug8 and ucg modules broken by eaac369.
Fixes #2926 

* [x] This PR is for the dev branch rather than for master.
* [x] This PR is compliant with the other contributing guidelines as well.
* [x] I have thoroughly tested my contribution.
* [x] The code changes are reflected in the documentation at docs/*.

https://github.com/nodemcu/nodemcu-firmware/commit/eaac369dece9ab74a736e8b42a21a4a655b3fe33 introduced changes to the way Lua ROM tables are built. Previously, lua function names were strings, now they are symbols (probably stringified by the macros).
But the ucg and ug8 modules use ma [... cut]
closed 2019-09-27 by marcelstoer