nodemcu/nodemcu-firmware #2928: wifi.startsmart() broken on master branch 310faf7

KT819GM, 2019-09-28:
### Expected behavior
Calling wifi.startsmart() will start auto configuration when `#define WIFI_SMART_ENABLE` is uncommented in user_config.h
### Actual behavior
`stdin:1: attempt to call field 'startsmart' (a nil value)
`
### Test code
```Lua
wifi.setmode(wifi.STATION)
wifi.startsmart(0,
    function(ssid, password)
        print(string.format("Success. SSID:%s ; PASSWORD:%s", ssid, password))
    end
)
```
### NodeMCU version
commit 310faf7fcc9130a296f7f17021d48c6d717f5fb6 (HEAD -> master, tag: 3.0-master_20190907, origin/master, origin/HEAD)

### Hardware
Generic ESP-07S

marcelstoer, 2019-09-29:
Would be good to know when it last worked. [As it looks](https://github.com/nodemcu/nodemcu-firmware/commits/master/app/modules/wifi.c) we only added 4 commits to `wifi.c` by @TerryE and @jmattsson in the last release.

KT819GM, 2019-09-29:
Seems error sit in wifi.c `#include "smart/smartconfig.h"`
smartconfig.h is included in sdk, not in smart folder so should be `#include "smartconfig.h"`. Sadly, can't check today as esptouch does not work anymore with ios 13 (but I'm not getting error when starting smartconfig anymore). 
What's more strange to me, that I can compile firmware with this file missing at my other pc (can't on current).

KT819GM, 2019-09-29:
Okay, got other phone to test. With:
```C
#ifdef WIFI_SMART_ENABLE
#include "smart/smart.h"
#include "smartconfig.h"
```
in wifi.c all works okay. Thank you for pointing out. Now left to find out how to deal with enduser rework.

KT819GM, 2019-09-29:
And now I saw #2923 which would be same problem, but as I said I could compile it somehow on one of mine pc's (ubuntu, standard 'make -j8') with file missing.

marcelstoer, 2019-09-29:
Great! Thanks for the analysis. I just checked the other modified `#include`s in #2842 for the same type of error. The rest seems sane.

The fix is now on `dev`.

commit 04287acb (2019-09-29): Fix invalid smartconfig include

Fixes #2928, #2923
commit da53d193 (2020-06-09): Fix invalid smartconfig include

Fixes #2928, #2923
closed 2019-09-29 by marcelstoer