nodemcu/nodemcu-firmware #2932: MQTT drops connection for no apparent reason

georeb, 2019-10-02:
### Expected behavior

Stay connected to MQTT broker, subscribe, publish and send keep-alive packets without randomly disconnecting!

### Actual behaviour

Intermittent disconnects with no rhyme nor reason.

Disconnects on a message receive
Disconnects on a publish
Disconnects on sending a keep-alive packet

**Error code from NodeMCU**
MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY -0x7880
The peer notified us that the connection is going to be closed.

**Error code from Azure**
404104
The connection was closed by the client, but IoT Hub doesn't know why.

One of them is lying.

I'm connected with 27k heap (using LFS) so memory shouldn't be a problem (and doesn't error)

### Test code
Connecting to Microsoft Azure IoT Hub (a pretty standard requirement for any IoT device)

```Lua
az=mqtt.Client("mytestdevice",60,"mytesthub.azure-devices.net/mytestdevice/?api-version=2018-06-30","SharedAccessSignature xxx")

az:on("message", function(client,topic,message)
    az:publish("devices/mytestdevice/messages/events/","GOT MESSAGE!",0,0)
end)

az:connect("mytesthub.azure-devices.net",8883,1,

function(client)
    print("connected")
    az:subscribe("devices/mytestdevice/messages/devicebound/#",0)
end)
```
### NodeMCU version
Master branch

### Hardware
ESP8266-12S

KT819GM, 2019-10-09:
Can you please be more specific which branch are you using now? Have you tried with clean mosquitto?
Anyway, will try to make on current dev 043046d2ab06fdefdae2a097c7c0f01cdb8da2fb some tests to see if it behaves in the same way.

georeb, 2019-10-13:
@KT819GM Sorry, I am using NodeMCU 2.2.1-master_20190405 build

I have tried connecting to Mosquitto with NodeMCU and its flawless, however not a fair test, as Mosquitto does not support secure connections.
Any support you can offer is much appreciated, I am pulling my hair out with this!

I have to say, I am disappointed with the lack of support from @joysfera @NicolSpies @pjsg on this issue. For the novice, this is a massive learning curve and I was hoping that an expert could contribute. It's not something a lot of people are able to fix themselves.

NicolSpies, 2019-10-13:
@georeb , I strongly suspect 27Kb heap is not enough for a successful TLS azure handshake as has been discussed in detail in previous posts. For Telegram I need the heap to be at least 35Kb with a `SSL_BUFFER_SIZE` of 5.5Kb and `SSL_MAX_FRAGMENT_LENGTH_CODE` not defined. 
To confirm this build an image with the TLS debug switches set in `user_config.h` as detailed in TLS related discussions on this list.
Using `tls.setDebug()`, detailed connection information will be provided to enable to debug what is going on during the TLS connection.

georeb, 2019-10-13:
@NicolSpies I am **connected** with 27Kb heap, not attempting to **connect** with 27Kb heap.

The handshake is successful and I am connected. The issue intermittently arises on a message receive, publish or most commonly, sending a keep-alive packet. There is no pattern, just completely spontaneous.

I have debugged as you've described and the only useful information from that debug is above:

> MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY -0x7880

I'd expect to see a `EM:xxxx` if there was memory resource deficiency.

joysfera, 2019-10-13:
@georeb , sorry to disappoint you with the lack of support. Couple of notes from my own experience:

- MQTT connection is very stable as long as I 'ping' = send an MQTT message every minute. This is to simulate keep-alive that did not work correctly when I started using MQTT on NodeMCU.

- Mosquitto has TLS support, I use it extensively with secured connections

- last but not least: I gave up completely on NodeMCU and TLS. This is simply a no go. No matter how much free heap I had it always failed.

NicolSpies, 2019-10-13:
@georeb , @joysfera, I believe us three have discussed and identified this problem in #1406 and #1811 three years ago. It sounds to me as the same MQTT problem identified back then and not a TLS issue.

georeb, 2019-10-13:
No problem @joysfera I understand that this is all open source, therefore no-one is obliged to progress this, however, this is all only as powerful as the support users are provided with. I get unjustifiably frustrated sometimes. Oh well!

> MQTT connection is very stable as long as I 'ping' = send a MQTT message every minute. This is to simulate keep-alive that did not work correctly when I started using MQTT on NodeMCU.

Your 'work around' solution may work for Mosquitto but isn't a viable solution for when using Azure, as users are charged for every message they process, which, being Microsoft, gets quite expensive.

> Mosquitto has TLS support, I use it extensively with secured connections

Sorry, my mistake, I wasn't aware of this. It's been at least 2 years since I last used it.

> last but not least: I gave up completely on NodeMCU and TLS. This is simply a no go. No matter how much free heap I had it always failed.

I disagree, I use many different TLS aspects of the NodeMCU firmware and have no issue usually, just MQTT. I've learnt that you just have to code as cleanly as possible.

@NicolSpies I agree that there is a bug in the MQTT module somewhere, rather than being a TLS issue, but where and why has this been ignored for 3 years!?

This is also backed up (as I mentioned in #1811) by the following debug I got in 2017 and I still get now:

```
client's data invalid protocol
Reason:[-0x7880]
```

`client's data invalid protocol` would suggest that NodeMCU is sending a malformed packet, again as suggested by @joysfera and I would have commented on #1811 if I could, however the stale bot closed the _OPEN_ issue (another gripe of mine!)

Where can we go from here? We have a clear and known issue thanks to @joysfera sniffing around with Wireshark, so why are we just ignoring this problem?

joysfera, 2019-10-13:
NodeMCU MQTT module got some fundamental improvements about a year ago, IIRC. For example a receiving message fragmented by TCP is now correctly concatenated at the NodeMCU side. This didn't work previously at all. I assume the change needed to implement this could also fixed most (if not all) other MQTT issues I've been having originally. I don't go back and test old things due to lack of time and also no possibility to upgrade firmware in units that have been deployed already.

FYI, I use the 20181207 release in most of the nodes I deployed but I have also several older firmwares (2017, 2016 and earlier) including those that infamously corrupt memory and do other nasty things. Nevertheless they all seem to work rather stable (knocking on wood!).

FYI2, I may have the largest NodeMCU deployment of you all. Also, my software is rather complex and mostly does not fit into memory so it tends to reboot when say timer fires and web request arrives at the same time.

And I cannot update old firmware because there isn't any OTA (I didn't go for the rboot thingy), so it's a pain in the butt when NodeMCU breaks compatibility with old firmware (which happens surprisingly often). That's also why I am generally not much interested in NodeMCU progress because I simply cannot use it.

I also have an interesting experience trying to run my Lua software on ESP32. It forced me to do things I was hoping would never have to do.

Anyway, the MQTT without TLS seems to do what I need, most of the time, using the workarounds I explained earlier/above. Wish you better luck with Azure.

georeb, 2019-10-13:
@NicolSpies Any thoughts?

NicolSpies, 2019-10-13:
Compiling a image with all the debug switches and using the tls.debug(3) command provides reams of debug information of all the TLS messages exchanged between the 8266 and the server. I have posted a number such debugs on the list.

Without the debug logs it is impossible to say what the problem could be. I had TLS related problems in the past not related to low heap that could only be spotted by fine-combing the debug log line by line.

georeb, 2019-10-13:
@NicolSpies Okay, that sounds promising and I'll comb through as much debug output as required if it helps fix the issue for me and others...

Currently my debug output gives this:

```
...
enter mqtt_socket_sent.
sent1, queue size: 1
send_if_poss, queue size: 0
sent2, queue size: 0
leave mqtt_socket_sent.
Reason:[-0x7880]
...
```

PLEASE NOTE

- This is the output by defining `DEVELOP_VERSION` in `user_config.h` only.
- The `Reason:[-0x7880]` **ALWAYS** occurs after a `leave mqtt_socket_sent`.

How do I enable this `tls.debug(3)` command that you mention please?

NicolSpies, 2019-10-14:
@georeb I assume you have included  `#define DEVELOP_VERSION` in `user_config.h` before building the image. Simply include `tls.setDebug(4)` in init.lua to get debug info similar to the extract I included in #2892. Using `4` provides the maximum debug feedback.

georeb, 2019-10-15:
@NicolSpies WOW that really outputs a lot of info, thanks for that!! I'll start sifting :)

However, using `tls.setDebug(4)` I am getting watchdog timer reset. But when using `tls.setDebug(3)` it hangs in there until the end. Why would this be? Too much debug?!

This is the exact point that the watchdog barks...

```
...

TLS<3> (heap=13672): ssl_cli.c:2911 value of 'ECDH: Q(X)' (383 bits) is:
TLS<3> (heap=13672): ssl_cli.c:2911  53 5f 28 42 fc da 7a a1 1e 92 b5 52 1d 20 8a 7c
TLS<3> (heap=13672): ssl_cli.c:2911  a6 94 6d 94 c3 a8 b6 8f f2 7e e1 2a 67 53 a2 67
TLS<3> (heap=13672): ssl_cli.c:2911  3e 89 23 4b fd 9f 93 2d 0b 59 c0 77 3f 0a a9 95
TLS<3> (heap=13672): ssl_cli.c:2911 value of 'ECDH: Q(Y)' (384 bits) is:
TLS<3> (heap=13672): ssl_cli.c:2911  ec 7d 3c 45 52 61 21 81 0c 49 65 a5 e6 67 c0 00
TLS<3> (heap=13672): ssl_cli.c:2911  b5 eb d9 fa f4 b5 6a 5e ad 65 45 b8 56 be 37 17
TLS<3> (heap=13672): ssl_cli.c:2911  c7 b4 94 da f6 6e ef 5f 49 57 a3 ec 16 3f ad b7

ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
```

NicolSpies, 2019-10-15:
@georeb, You need to step away a bit to see the big picture and follow the logical process from the start to see what it was busy with. 
A clue is `ssl_cli.c:2911` that refers to the mbedtls module and I believe 2911 is the line number. If you google this you will find the following code on that line:
```
            mbedtls_ssl_send_alert_message( ssl, MBEDTLS_SSL_ALERT_LEVEL_FATAL,
                                            MBEDTLS_SSL_ALERT_MSG_HANDSHAKE_FAILURE );
            return( MBEDTLS_ERR_SSL_PK_TYPE_MISMATCH 
```
This refers to a handshake failure. You will be able to confirm this reading the debug info from the start to see if this is still during the initial TLS handshake process. I suspect it is and the failure happens before the TLS handshake is confirmed to be established. If this is so, the problem is not related to MQTT.

georeb, 2019-10-16:
@NicolSpies I think I may not have explained myself. The watchdog timer reset is unrelated to this issue. It occurs every time I try to connect to Azure. I have got around this by issuing `tls.setDebug(4)` _after_ a successful Azure connection. Weird.

[Extract from a successful PUBLISH event](https://gist.github.com/georeb/d2b4e1db8fc34eafad151b7b51d52813)

[Extract from an unsuccessful PUBLISH event and immediate disconnect](https://gist.github.com/georeb/d87788c6757a47fd6424057b5fd5a711)

To save you the time sifting through, below are the inconsistencies between a **successful** and **unsuccessful** PUBLISH event. Remember that nothing has been changed between these two publish attempts. I am sending exactly the same payload to exactly the same topic...

Successful PUBLISH event

```
TLS<4> (heap=26600): ssl_tls.c:1385 dumping 'computed mac' (32 bytes)
TLS<4> (heap=26600): ssl_tls.c:1385 0000:  d8 56 01 5e dc 01 05 3b c6 32 1d 95 e3 76 95 ed  .V.^...;.2...v..
TLS<4> (heap=26600): ssl_tls.c:1385 0010:  ac 61 9b 75 5a 3a 48 a1 e8 28 85 e8 30 77 05 8d  .a.uZ:H..(..0w..
TLS<3> (heap=26600): ssl_tls.c:1552 before encrypt: msglen = 112, including 16 bytes of IV and 15 bytes of padding
```

Unsuccessful PUBLISH event

```
TLS<3> (heap=26432): ssl_tls.c:1552 before encrypt: msglen = 80, including 16 bytes of IV and 15 bytes of padding
TLS<3> (heap=26432): ssl_tls.c:1597 using encrypt then mac
TLS<4> (heap=26432): ssl_tls.c:1604 dumping 'MAC'd meta-data' (13 bytes)
TLS<4> (heap=26432): ssl_tls.c:1604 0000:  00 00 00 00 00 00 00 49 17 03 03 00 50           .......I....P
```

The difference appears to show a payload encryption mismatch...

NicolSpies, 2019-10-16:
@georeb , try the same but with the QoS in the publish command set to 1 and 2 respectively to see if it makes a difference.

georeb, 2019-10-16:
Okay, I'll give that a try, although, please see a quote from Azure below...

> The following is a list of IoT Hub implementation-specific behaviours:
> 
> IoT Hub does not support QoS 2 messages. If a device app publishes a message with QoS 2, IoT Hub closes the network connection.
> 
> IoT Hub does not persist Retain messages. If a device sends a message with the RETAIN flag set to 1, IoT Hub adds the x-opt-retain application property to the message. In this case, instead of persisting the retain message, IoT Hub passes it to the backend app.
> 
> IoT Hub only supports one active MQTT connection per device. Any new MQTT connection on behalf of the same device ID causes IoT Hub to drop the existing connection.

NicolSpies, 2019-10-16:
@georeb, the more I investigate, the more I run into old post of more or less the same issue. The fact that it is still around indicates to me that nobody is trying to do what you are or have accepted that it can not work. All I can suggest is to try and find a pattern that could give a clue to why the problem is intermittent. More eyes on the problem would also help even if it is only to suggest what to try.

NicolSpies, 2019-10-16:
@georeb, the problem could be caused by MQTT, TLS or Azure. It would help to have eyes on the server side as well. I do not know if a local azure setup is possible. You could try another secure MQTT server to see if the fault persists. You know it is fine in a unsecured connection. So that could maybe rule MQTT out. It does work but occasionally breaks down, could it be connection instability. Run as stand alone code using LFS to ensure lots of heap. Hope this helps

georeb, 2019-10-23:
I think I may have found the culprit...

An `if` statement starting at line 1338 in [ssl_tls.c (line 1338)](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/mbedtls/library/ssl_tls.c#L1338) is randomly flicking between true and false, resulting in a successful and unsuccessful MQTT publish, respectively.

I have tried to get an output of the values within the `if` statement, but with no luck (I'm not fluent in the language). Any idea of a suitable string I can insert into this file?

Iv'e tried...

```c
MBEDTLS_SSL_DEBUG_MSG( 1, ( "MODE = ", mode ) );
MBEDTLS_SSL_DEBUG_MSG( 1, ( "MBEDTLS_MODE_STREAM = ", MBEDTLS_MODE_STREAM ) );
MBEDTLS_SSL_DEBUG_MSG( 1, ( "MBEDTLS_MODE_CBC = ", MBEDTLS_MODE_CBC ) );
MBEDTLS_SSL_DEBUG_MSG( 1, ( "ssl->session_out->encrypt_then_mac = ", ssl->session_out->encrypt_then_mac ) );
MBEDTLS_SSL_DEBUG_MSG( 1, ( "MBEDTLS_SSL_ETM_DISABLED = ", MBEDTLS_SSL_ETM_DISABLED ) );
```

Can you shed any light on this @TerryE and/or @nwf ? I'm assuming that an incorrect cipher mode is being selected, resulting in Azure rejecting the published data and kicking the device offline...

NicolSpies, 2019-10-23:
@georeb, accolades for your determination not to give up. You have been doing a lot of digging to try and isolate the area where the problem could occur. In the past I did the same to enable the hardcore developers to give their expert opinion without the hours of trying to find a pattern that could give a clue of the problem. @nwf and @TerryE provided excellent guidance that allowed me to fix the problem.

georeb, 2019-10-23:
@NicolSpies Haha, I don't like giving up!

> enable the hardcore developers to give their expert opinion without the hours of trying to find a pattern that could give a clue of the problem

Exactly my mentality! I have the time to dig, just need @TerryE and @nwf  to keep me heading in the right direction...

In addition to my last comment:

It appears that, on an unsuccessful publish/keepalive, as suspected, I am also getting:

> MBEDTLS_ERR_SSL_INVALID_MAC - 0x7180 - Verification of the message MAC failed.

...followed by...

> client's data invalid protocol

...then...

> MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY - 0x7880 - The peer notified us that the connection is going to be closed.

Which falls in line with the fact that the `ssl_encrypt_buf` function is to blame. Sometimes a MAC is added to the encryption, sometimes not. I cannot see why this would be skipped occasionally...

georeb, 2019-10-24:
MbedTLS Support have replied with this:

> If this if statement sometimes returns true and sometimes return false, it probably means that either mode or ssl->session_out->encrypt_then_mac is changed between every run. This shouldn’t happen, as this is depending on the negotiated ciphersuite (the mode), or whether encrypt_ten_mac was enabled or not.

And then more worryingly, this:

> If this has changed during your session, this may hint on some memory corruption in your system.

georeb, 2019-10-24:
Right, so, by implementing extra debug info, I have determined the following:

**Constants (that we already knew):**

```c
MBEDTLS_MODE_STREAM = 7
MBEDTLS_MODE_CBC = 2
MBEDTLS_SSL_ETM_DISABLED = 0
```

**On a successful MQTT publish:**

```c
mode = 2
ssl->session_out->encrypt_then_mac = 0
```

**On an unsuccessful MQTT publish:**

```c
mode = 2
ssl->session_out->encrypt_then_mac = 1701013878
```

**So that’s the discrepancy that's causing the IF statement to return false;** `mode` appears to always be `2` but the number returned by `encrypt_then_mac` appears to be a random integer of length 10...

@TerryE and @nwf I would really appreciate your input on this, I've done the leg work here, so I'm hoping a quick glance is all you need to get to the bottom of this...?!

joysfera, 2019-10-24:
FYI, the unsuccessful mac value is `eciv` in ASCII. Not sure if it means anything or if it's just a random portion of memory containing say Lua source code.

TerryE, 2019-10-25:
@georeb I agree that this needs some diagnostic time with someone who knows their way around using the gdbstub interface, but unfortunately I am up to my eyes in alligators at the moment trying to get the Lua53 release out.  Perhaps @nwf or @HHHartmann can help?

nwf, 2019-10-25:
Skimming mbedTLS's source, `encrypt_then_mac` is only ever deliberately set to `MBEDTLS_SSL_ETM_ENABLED` (`1`) or `MBEDTLS_SSL_ETM_DISABLED` (`0`).  If you're seeing another value there, I suspect memory corruption (as suggested by @joysfera).

There is no particularly good way to debug such things; perhaps printing out more of the session structure could give a hint as to who's overflowing some adjacent buffer.  If the gdbstub understands watchpoints (@TerryE?), then that's another option as well.

TerryE, 2019-10-26:
@georeb, @HHHartmann, picking up @nwf's point:I do understand the gdbstub and
-  Yes, it could be used to diagnose this issue and
-  Yes, I do understand how to diagnose this.

_But_ I worked out all of this by trial and error and looking at the sources.  I also feel that it is a mistake for me to be the single point of all such knowledge on the project, hence #2731.  If anyone has basic C development skills, preferably has used `gdb`, has either native Linux or docker build capability and wishes to diagnose this problem then why don't they volunteer and work with me through #2731 with primary aim of documenting a "How to do remote debugging for C library developers and maintainers", and along the way track down this issue?

TerryE, 2019-10-28:
Re #2731: @georeb @NicolSpies @joysfera @nwf @HHHartmann, any volunteers?

georeb, 2019-10-28:
Using `#undef MBEDTLS_SSL_ENCRYPT_THEN_MAC` did the trick!

Thank you @NicolSpies for suggesting to use `tls.setDebug(4)` otherwise I may never of found the solution!

This fixed my unique issue however, it doesn't explain why `encrypt_then_mac` is fluctuating...

> MbedTLS Support: "This strengthens my assumption that there is some memory corruption in your system, that over wrote your session context. In your application, have freed all the variables that you have allocated, after use? Of course, this [`#undef MBEDTLS_SSL_ENCRYPT_THEN_MAC` ] would probably only work around the issue, however it will not fix your memory corruption or memory leak in your system."

Perhaps this potential **memory corruption** is something worth investigating...

NicolSpies, 2019-10-28:
@georeb , great news. To confirm, the random disconnection does not happen anymore ?

georeb, 2019-10-28:
> Re #2731: @georeb @NicolSpies @joysfera @nwf @HHHartmann, any volunteers?

Happy to contribute where I can, however I'm a novice at C.

NicolSpies, 2019-10-28:
@georeb, I would like to try it as well, would you mind to share your code with me offline

nwf, 2020-10-28:
I'm not sure this is stale; we may still have some un-diagnosed memory corruption.

closed 2022-04-30 by stale[bot]