nodemcu/nodemcu-firmware #2940: Having memory leak problems when using gpio pulse

peterdownunder, 2019-10-08:
### Expected behavior
Memory should not leak on successive call to pulse.build
### Actual behavior
Memory leaks about 4k for every 100 calls in the sample below
### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```
local function IntervalTimerRestart (timer)
    return function()
        print ('done:' .. node.heap())
        timer:start()
    end
end

local function IntervalTimerAction (actions, timer)
    return function()
        for id, a in next, actions do
            actions[id] = nil
        end
        table.insert(actions, { [7] = gpio.LOW, delay=500 })
        table.insert(actions, { [7] = gpio.HIGH, delay=500, loop=1, count=10})
        pulsar = gpio.pulse.build(actions)
        pulsar:start(IntervalTimerRestart (timer))
    end
end

gpio.mode(1, gpio.OUTPUT, gpio.PULLUP)
gpio.mode(7, gpio.OUTPUT, gpio.PULLUP)

gpio.write(1, gpio.LOW)
gpio.write(7, gpio.LOW)

node.egc.setmode(node.egc.ALWAYS)
local stepAction = {}
mytimer = tmr.create()
mytimer:alarm(1000, tmr.ALARM_SEMI, IntervalTimerAction(stepAction, mytimer))
```
### NodeMCU version
Building a local version of NodeMcu off master version 2.2.1

### Hardware
Describe which ESP8266 device you use and document any special hardware setup
required to reproduce the problem.
Standard ESP 12 (Amica) dev board

pjsg, 2019-10-10:
This is very odd -- I've stared at the implementation of gpio.pulse and it looks plausible to me. I'm going to have to dig deeper!

peterdownunder, 2019-10-10:
I did notice that there has been work done on pulse in the dev branch, none releated to memory.

Is it worth me trying on the dev branch

pjsg, 2019-10-10:
I doubt it -- the differences between the current dev and master are just a
rename of a library function.

nwf, 2020-10-06:
This seems like it's still a bug.

PR #3308 (2020-10-13): Don't free a ref of zero -- bad things happen...
Fixes #2940.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

It turns out that a `luaL_unref(L, LUA_REGISTRYINDEX, 0)` has the rather surprising effect of throwing away the f [... cut]
closed 2020-10-13 by marcelstoer