nodemcu/nodemcu-firmware #2941: I2C: No ACK on last byte

poorandunlucky, 2019-10-09:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

I would like inhibit I2C ACK on last read/written byte, as my silicon increments registers on each read/written byte, and this causes the chip to hang/wait indefinitely in TX mode after reading.

### Justification

Incrementing addresses seems to be a pretty common feature for a lot of silicon...  MicroPython's I2C library seems to incorporate ways to do just that, too...

### Workarounds

Haven't found one, but there's probably one...  If there is please let me know...

8<------------------------ END FEATURE REQUEST --------------------------------

sonaux, 2019-10-10:
I think it's possible, I'll look if I can do it.

poorandunlucky, 2019-10-11:
Thanks Natalia...  I looked in the SDK source, and that's where the byte gets set and sent, I think you'd have to create another function, or edit the function, but those functions have prototypes in several files, and uses a lot of structs and pointers, and I'm not that good with C yet...  I don't think you could do it in Lua itself....

sonaux, 2019-10-21:
Main document on I2C (UM10204 I2C-bus specification and user manual Rev. 6 - 4 April 2014) states:
>There are five conditions that lead to the generation of a NACK:
>1-4. Receiver cannot receive data right now or not responding or does not know what to do with received data.
>5. A master-receiver must signal the end of the transfer to the slave transmitter.

It turns out that variant 5 is actually implemented in module `i2c`: 
```
// Lua: read = i2c.read( id, size )
static int i2c_read( lua_State *L )
{
...
  for( i = 0; i < size; i ++ )
    if( ( data = platform_i2c_recv_byte( id, i < size - 1 ) ) == -1 )
      break;
    else
      luaL_addchar( &b, ( char )data );
  luaL_pushresult( &b );
  return 1;
}
```
Expression `i < size - 1` evaluates to 0 (NACK) when last requested byte is received. 
It should work, only I had stuffed all my IoT toys in a box long time ago and cannot check this with breadboard and logic analyzer right now.

Of course, if you're using some C module, sending NACK on last received byte should be implemented in that module, because i2c driver and platform i2c wrapper contains only basic functions.

poorandunlucky, 2019-11-11:
Well I leaned C (and showed some intuition this time around), but it's going to take me a lot more time than anticipated to make heads and tails of the SDK for reasons you're probably all already familiar with, so wanting to resume playing with my electronics I went back to NodeMCU with a newfound respect, and appreciation of it... it's just so pleasant to use, and well documented, and I already had a (pretty nice if I may say so myself) environment going with it, so yesterday I hooked up the basic logic analyzer I bought in the interval (a LHT00SU1 if anyone's interested), and NodeMCU does indeed send out a NACK after the last read byte.

I did everything calm and proper yesterday, with the I2C standard, the datasheets, looked at the traces on my demoboard, put caps everywhere, and went through my code with the manual, and I managed to read single bytes, multiple bytes, and also perform multiple read operations one after the other which was the problem...

I think the problem was the order of operations, I think at some point I deviated from the order in the guide trying to debug, but what the problem was before that I can't recall...  Maybe it was something else that was connected to the bus, I had an OLED giving temperature from a thermistor on the board last time, too, but the essential is that it works now...

That SDK is something else, honestly....

Thank you..

Patrick Dorion
📱 +1 438 345-4575

closed 2019-11-11 by poorandunlucky