nodemcu/nodemcu-firmware #2944: Temperature influence to the BME680 air pressure and humidity readout

Michal78S, 2019-10-16:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior:
The air pressure, humidity and gas resistance are compensated during ambient temperature changes.
### Actual behavior:
The air pressure drops or grown when temperature increase or decrease. Please look at attached screen. I have checked two sensor. Both behaved the same. Then I have checked it in Arduino with ClosedCube library problem still existed, but when checked with Adafruit library the pressure was stable, and changed as weather forecast shows. I suposse that NodeMCU and ClosedCube library comes from the same source with bug.

### Test code
ClosedCube lib: https://github.com/closedcube/ClosedCube_BME680_Arduino
Adafruit lib: https://github.com/adafruit/Adafruit_BME680 - works OK
### Hardware
ESP12

8<------------------------ END BUG REPORT -------------------------------------

vsky279, 2019-10-16:
It's true that I had no chance to test the library in some real application. So there might be some flaws. When I was writing the module I didn't see any other implementation for inspiration. I will try to check the Adafruit lib and find potential bug in the module.

Though it might take same time.

Michal78S, 2019-10-17:
[image]
It looks like that. Blue is temperature, green is air pressure.

vsky279, 2019-11-17:
The implemented module is using driver [version 3.5.1](https://github.com/BoschSensortec/BME680_driver/tree/2a51b9c0c1899f28e561e6701caa22cb23201cfc) from Bosch while Adafruit version is using more recent driver.

So updating the module to latest version might solve the issue.

Michal78S, 2019-11-19:
Thanks for checking, but Adafruit works OK when ClosedCube and NodeMCU occurs temperature influences.

vsky279, 2019-11-23:
I confirm the issue with the existing driver. The Bosh driver update solves the issues. I will post a PR soon as soon as I clean it a bit.

PR #2969 (2019-12-09): Sensor driver for BME680 sensor - update to version 3.5.9
Fixes #2944.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/*`.

Solves the issue where the air pressure, humidity and gas resistance are not compensated during ambient temperature changes.

Updates the Bosh BME680 driver from versi [... cut]
closed 2019-12-09 by marcelstoer