nodemcu/nodemcu-firmware #2945: Terry's LFS Lua Cross-Compile Web Service IS NOT WORKING

chathurangawijetunge, 2019-10-18:
Terry's LFS Lua Cross-Compile Web Service IS NOT WORKING

https://blog.ellisons.org.uk/article/nodemcu/a-lua-cross-compile-web-service/

HHHartmann, 2019-10-19:
@TerryE Terry seems as if your server refuses connection. Might need some care by you.

chathurangawijetunge, 2019-10-19:
What do u mean

TerryE, 2019-10-19:
Sorry, it's a dedicated RPi that's dropped out for some reason maybe SDcard failure.  It's no longer visible on my network hub.  I need to plug it in to HDMI display and keyboard to diagnose. Don't have time today.  Maybe tomorrow.

chathurangawijetunge, 2019-10-19:
Can you please tell me how to run the same in my raspberry pi.
As I'm in a really difficult situation in compiling LFS .img
If you can show me the way I will setup a raspberry pi even for public use

chathurangawijetunge, 2019-10-19:
If u can send a img of rip ad card image really appreciate

TerryE, 2019-10-19:
> Can you please tell me how to run the same in my raspberry pi.

See our online documentation: [Compiling Lua on your PC for Uploading](https://nodemcu.readthedocs.io/en/dev/compiling/#compiling-lua-on-your-pc-for-uploading) and in more detail in [LFS:Choosing your development life-cycle](https://nodemcu.readthedocs.io/en/dev/lfs/#choosing-your-development-life-cycle).  

Buildiing luac.cross and the host tools only needs the standard _build essential_  tools and you just download our distro zip, cd to the `app/lua/luac_cross` directory and do a make.  This produces the executable in seconds.  You can then just run the executable from the command line.

TerryE, 2019-10-19:
You can also happily build your firmware on the RPi so long as have the xtensa toolchain installed.  One of our contributors supplies the [linux-x86](https://github.com/jmattsson/esp-toolchains/releases/download/linux-x86_64-20190731.0/toolchain-esp32-linux-x86_64-20190731.0.tar.xz).  But you can also build this yourself if needed.  IIRC, this takes about 15-30 mins on an RPi3 and half of that on an RPi4.   I'll looking into adding the RPi versions to Johny's release.  But you don't need any of this to do luac compiles.

chathurangawijetunge, 2019-10-20:
Thank for info....
Look forward till lfs sight (pi)  re start......

chathurangawijetunge, 2019-10-21:
Any luck on restarting the pi server......?

chathurangawijetunge, 2019-10-21:
Server is up and running Thank you

closed 2019-10-21 by chathurangawijetunge

bantugam, 2020-06-14:
Hi Terry,
unfortunately your wonderful webservice is not working currently. A couple of days it started to deliver .img files that were not accepted by the ESP. I figured out that uploading 3 times the same .zip file I got a different .img file each time. Today I noticed that an upload is not possible at all anymore. Do you have a chance to look into it? I really appreciate your commitment!! Thank you!! BR Joerg
Message: "No zip file uploaded"  (I tried Chrome and MS Edge as browsers)