nodemcu/nodemcu-firmware #2954: lua_examples/luaOTA crashes when entering _init.lua

sognetic, 2019-11-03:
### Expected behavior
I've been trying to use the luaOTA example from lua_examples to do deployment. 
Expected behaviour is thus: Not crashing.
.
### Actual behavior
Code immediately panics when entering _init.lua with
```
Starting Provision Checks

PANIC: unprotected error in call to Lua API (luaOTA/_doTick.lua:1: attempt to index global 'self' (a nil value))
```

### Test code
MCVE is simply uploading the code in luaOTA (minus the server component) to the ESP

### NodeMCU version
master
commit: 310faf7fcc9130a296f7f17021d48c6d717f5fb6

### Hardware

NodeMCU board Ver. 0.1 (?)

TerryE, 2019-11-03:
Thanks @sognetic.  I'll take a look in the next day or so.

sgjava, 2019-12-06:
I'm seeing same issue with:
```
	branch: dev
	commit: 9a2579d97160e981cd084ff483f0eeac8210da37
	release: 
	release DTS: 201911261112
	SSL: false
	build type: float
	LFS: 0x0
```

```
Starting Provision Checks
?:0: attempt to index global 'self' (a nil value)
stack traceback:
	?: in function '_doTick'
	?: in function '_init'
	?: in function <?:64>
```

adeltc, 2020-01-15:
luaOTA seems behind the current version of the firmwork. 
there is a couple of issues. For example, the new api implementation of the timer doesn't match the latest update. we can see that the line 35 in the _doTick.lua still provide the timer id as first argument which the new version of the firmware doesn't take it. There is couple of other small issue that I fixed. I have a workable version. Should I make a push request ?

HHHartmann, 2020-01-15:
@adeltc Please go ahead making a push request.

adeltc, 2020-01-15:
ok the pull request is done : https://github.com/adeltc/nodemcu-firmware/pull/1

HHHartmann, 2020-01-17:
@adeltc thanks for the findings and for creating a pull request.

There are some things which went the wrong way though.
To be able to merge your PR (pull request) it has to be made against the nodemcu/nodemcu-firmware repository, not against your fork of it. After pushing your branch to your forked repository when browsing to https://github.com/nodemcu/nodemcu-firmware there is already a button asking if you would like to do a PR. Be aware to make the PR against the dev branch, not master.

When updating a branch with the newest commits from the dev branch use rebase instead of merge. If you use merge all the commits from other contributors will show up and it will be difficult to see your changes.

I just saw that your repo does not seem to be a fork of nodemcu/nodemcu-firmware, so you would have to fork nodemcu/nodemcu-firmware first and then create a branch off of dev, push your changes and create the PR.
 
When creating the PR online you can also see the changes it contains. If there is anything but yours, something is wrong.

adeltc, 2020-01-21:
Really sorry about this mistake. I'm currently fixing it.

adeltc, 2020-01-21:
Hopefully this one would be the right one : https://github.com/nodemcu/nodemcu-firmware/pull/3015

adeltc, 2020-01-22:
Even that I based my new branch on dev branch, the pull request has another based branch configuration that I didn't see. I realised it was based on master. I just made antoher pull request with dev as base branch : https://github.com/nodemcu/nodemcu-firmware/pull/3016

PR #3016 (2020-04-26): #2954: adapt LuaOTA to the new firmware api for the timer and some other sma…
Notice : This pull request is based on dev branch as the luaota_improvement branch also based on dev

Fixes #2954

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested m [... cut]
closed 2020-04-26 by marcelstoer