nodemcu/nodemcu-firmware #2955: linenoise (readline-like line editing) support for nodemcu

wilhelmy, 2019-11-03:
### Linenoise input

### Justification
Using an input library would make it easier to use the serial line interactively.
I'm currently working on adding linenoise because it's small.

I would appreciate some help with the following linker problem (I don't think this really belongs into the issue tracker but I don't know what would be a more appropriate way to discuss it, sorry)

```
make[2]: Entering directory '/home/mw/src/nodemcu/firmware/app/mqtt'
make[2]: Leaving directory '/home/mw/src/nodemcu/firmware/app/mqtt'
xtensa-lx106-elf-gcc -L/home/mw/src/nodemcu/firmware/sdk/esp_iot_sdk_v3.0-e4434aa/lib -L/home/mw/src/nodemcu/firmware/sdk/esp_iot_sdk_v3.0-e4434aa/ld  -Wl,--gc-sections -Wl,-Map=mapfile -nostdlib -T../ld/nodemcu.ld -Wl,@../ld/defsym.rom -Wl,--no-check-sections -Wl,-static -u WIFI_module_selected -u UART_module_selected -u TMR_module_selected -u SPI_module_selected -u PIPE_module_selected -u OW_module_selected -u NODE_module_selected -u NET_module_selected -u MQTT_module_selected -u I2C_module_selected -u GPIO_module_selected -u FILE_module_selected -u DHT_module_selected -u BIT_module_selected -u ADC_module_selected -Wl,--start-group -lmain user/.output/eagle/debug/lib/libuser.a crypto/.output/eagle/debug/lib/libcrypto.a driver/.output/eagle/debug/lib/libdriver.a platform/.output/eagle/debug/lib/libplatform.a libc/.output/eagle/debug/lib/liblibc.a lua/.output/eagle/debug/lib/liblua.a lwip/.output/eagle/debug/lib/liblwip.a smart/.output/eagle/debug/lib/smart.a spiffs/.output/eagle/debug/lib/spiffs.a fatfs/.output/eagle/debug/lib/libfatfs.a pm/.output/eagle/debug/lib/libpm.a esp-gdbstub/.output/eagle/debug/lib/libgdbstub.a net/.output/eagle/debug/lib/libnodemcu_net.a mbedtls/.output/eagle/debug/lib/libmbedtls.a modules/.output/eagle/debug/lib/libmodules.a smart/.output/eagle/debug/lib/smart.a uzlib/.output/eagle/debug/lib/libuzlib.a dht/.output/eagle/debug/lib/libdht.a mqtt/.output/eagle/debug/lib/libmqtt.a -Wl,--end-group -Wl,--start-group -lgcc -lhal -lphy -lpp -lnet80211 -lsmartconfig -lwpa -lwpa2 -lcrypto -lwps -lc -lm -Wl,--end-group -o .output/eagle/debug/image/eagle.app.v6.out
/home/mw/src/nodemcu/firmware/sdk/esp_iot_sdk_v3.0-e4434aa/lib/libc.a(lib_a-svfprintf.o):(.literal+0x10): undefined reference to `_malloc_r'
/home/mw/src/nodemcu/firmware/sdk/esp_iot_sdk_v3.0-e4434aa/lib/libc.a(lib_a-svfprintf.o): In function `_svfprintf_r':
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/vfprintf.c:733: undefined reference to `_malloc_r'
/home/mw/src/nodemcu/firmware/sdk/esp_iot_sdk_v3.0-e4434aa/lib/libc.a(lib_a-svfiprintf.o):(.literal+0x0): undefined reference to `_realloc_r'
/home/mw/src/nodemcu/firmware/sdk/esp_iot_sdk_v3.0-e4434aa/lib/libc.a(lib_a-svfiprintf.o):(.literal+0x4): undefined reference to `_free_r'
/home/mw/src/nodemcu/firmware/sdk/esp_iot_sdk_v3.0-e4434aa/lib/libc.a(lib_a-svfiprintf.o): In function `__ssprint_r':
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/vfprintf.c:234: undefined reference to `_malloc_r'
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/vfprintf.c:245: undefined reference to `_realloc_r'
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/vfprintf.c:249: undefined reference to `_free_r'
/home/mw/src/nodemcu/firmware/sdk/esp_iot_sdk_v3.0-e4434aa/lib/libc.a(lib_a-svfiprintf.o): In function `_svfiprintf_r':
/home/wjg/Repo/esp-open-sdk-20170622/crosstool-NG/.build/src/newlib-2.0.0/newlib/libc/stdio/vfprintf.c:733: undefined reference to `_malloc_r'
collect2: error: ld returned 1 exit status
make[1]: *** [../Makefile:480: .output/eagle/debug/image/eagle.app.v6.out] Error 1
make[1]: Leaving directory '/home/mw/src/nodemcu/firmware/app'
make: *** [Makefile:380: .subdirs] Error 2

```

TerryE, 2019-11-04:
The SDK and our RTL only support a subset of the printf variants.  You are using ones that aren't and so are pulling in the newlib libc/stdio functions which use the newlib allocators rather than the SDK one.

BTW in-line editing is quite complex and command line history really too much of a luxury when you only have 44Kb RAM available.

wilhelmy, 2019-11-04:
The only printf-style function I call is snprintf. That's supported, right?

Regarding resource constraints, I'm aware. Input history is limited to 2k (4 lines at 512b).
Linenoise is relatively lightweight and simple by cutting corners. I'd say it's probably the best option on embedded systems.

marcelstoer, 2019-11-04:
> I don't think this really belongs into the issue tracker but I don't know what would be a more appropriate way to discuss it, sorry

The GitHub issue template links to http://nodemcu.readthedocs.io/en/dev/support/.

TerryE, 2019-11-04:
Yup until you prepare and frame this as a PR, then the best place for more informal discussion is on our [Gitter channel](https://gitter.im/nodemcu/nodemcu-firmware).

wilhelmy, 2019-11-04:
I see. Thanks!
Closing this then.

closed 2019-11-04 by wilhelmy