nodemcu/nodemcu-firmware #2956: httpd server module in c

NiklausS, 2019-11-03:
8<------------------------ FEATURE REQUEST ------------------------------------
### I am usding this httpd server now for quite some time, and thought i might be a good idea to let others also use it if they like so.

### Justification
I've read it in the posts every now and then that is is something others would also use.

### Workarounds
i know there are some httpd servers written in lua. I don't want to judge on which one is better, as this one is quite simple and works for me ...

I have provided a git diff to the master branch, including documentation and some samples. I'd be happy if some of you can have a look at it, and if let me know what i have to change to have this in the master.

Thank you
 Nik
[httpd.patch.gz](https://github.com/nodemcu/nodemcu-firmware/files/3802099/httpd.patch.gz)

8<------------------------ END FEATURE REQUEST --------------------------------

TerryE, 2019-11-04:
@NiklausS It would be far better to comply with out contribution guideline by forking the repo and issuing a PR rather than a bare patch.  Also I see that you are making a lot of changes to the net module.  We would be reluctant to do this without compelling reasons and decent backwards compatibility.

marcelstoer, 2019-11-04:
> i know there are some httpd servers written in lua. I don't want to judge on which one is better

We normally state that anything that doesn't have C-level timing requirements (e.g. hardware protocol) should be done in Lua. Therefore, we also started to deprecate and remove C modules for which there is a suitable Lua alternative.

We have a [Lua HTTP server module](https://nodemcu.readthedocs.io/en/master/lua-modules/httpserver/) in this repo and https://github.com/marcoskirsch/nodemcu-httpserver is also quite popular.

closed 2019-11-04 by marcelstoer

seregaxvm, 2019-11-13:
> We have a Lua HTTP server module in this repo and https://github.com/marcoskirsch/nodemcu-httpserver is also quite popular.

It seems abandoned. It would be nice to have a community fork.

marcoskirsch, 2019-11-14:
> It seems abandoned. It would be nice to have a community fork.

I don't work on this anymore but would be happy to transfer ownership to somebody interested and willing to keep working on it.