nodemcu/nodemcu-firmware #2958: Current build process only supports x86-64 hosts out of the box

TerryE, 2019-11-08:
### Expected behaviour
The NodeMCU make should work with ARM class processors such as the RPi4

### Actual behaviour
Follow the build documentation but on an RPi4.  You get this error on the make:
```
<buildroot>/tools/toolchains/esp8266-linux-x86_64-20181106.0/bin/xtensa-lx106-elf-ar: 1: <buildroot>/tools/toolchains/esp8266-linux-x86_64-20181106.0/bin/xtensa-lx106-elf-ar: Syntax error: "(" unexpected
make: *** [Makefile:337: <buildroot>/sdk/.pruned-3.0-e4434aa] Error 2
make: Leaving directory '<buildroot>'
```
### NodeMCU version
Current Master

### Hardware
RPi4
### Workarounds
The make includes the environment variable `TOOLCHAIN_ROOT` to allow the developer to specify the Xtensa toolchain path, but to simplify the build process for those who are building on a 64-bit x86 platform, the make will fetch @jmattsson's relevant prebuilt toolchain, so this make works fine most build targets.  

Our build documentation at [Linux Build Environment](/nodemcu/nodemcu-firmware/blob/master/docs/build.md#linux-build-environment) does explain that the pre-built toolchain is only supplied for 64-bit x86 platforms, but it should mention the need to set `TOOLCHAIN_ROOT` for others.

Given the increasing availability of powerful ARM-based SBCs (such as the Raspberry Pi) and Chromebooks, I feel that we should consider adding a prebuilt ARM toolchain to Johny's set.

TerryE, 2019-11-08:
I've just built the xtensa toolchain on an RPi4 and it doesn't build "out-of-the-box".  You need to enable the `deb-src` line in `/etc/apt/sources.list` and then `sudo apt-get install bison makeinfo texinfo help2man  python2.7-dev python-serial` plus a couple of other tweaks to get the toolchain to build.  I'll put up a packed tarball online tomorrow so other RPi users can avoid all of this hassle.

TerryE, 2019-11-11:
The tarball is here: https://blog.ellisons.org.uk/misc/esp8266-linux-arm_32-20191111.0.tar.xz

chathurangawijetunge, 2019-11-20:
really appreciate if you can give the steps to use "https://blog.ellisons.org.uk/misc/esp8266-linux-arm_32-20191111.0.tar.xz" with RPI

TerryE, 2019-11-21:
You'll find instructions on how to pack an XZ tar on the internet.  You'll also need some stuff installed on the Pi to do the make; git, build-essential etc., but this is no different from any non-docker build.  When in doubt try the make; if it fails then it will tell you what package is missing and Google is your friend here.  Just follow the build instructions but instead of the default toolchain directory, if you have installed the tarball at `~/xtensa/esp8266-linux-arm_32-20191111.0/' then your basic make will be
```
make TOOLCHAIN_ROOT=~/xtensa/esp8266-linux-arm_32-20191111.0/
```
and the build will pick up the correct xtensa toolchain.

f4grx, 2020-06-24:
Hello,
building this right now.

This is highly inefficient. It took a long time that I needed bison
Then it restarted generating the .in files for several minutes
Then it stopped telling me it needed flex
Then it restarted generating the .in files for several minutes
Then it stopped telling me it needed help2man
Then it restarted generating the .in files for several minutes
Then build failed telling me it needs  bash > 3.1
I have bash 5.0.3

So something else is wrong :)

jmattsson, 2020-06-25:
@f4grx  The toolchains NodeMCU typically uses are built using [this repo](https://github.com/jmattsson/esp-toolchains), which also includes the necessary patch for bash 5 to be detected correctly. And yeah, building the toolchain takes A Long Time(tm). I guess I should see if I can find some time and a spare Pi so I can start providing pre-built ARM toolchains too.

f4grx, 2020-06-25:
on it, thanks for the pointer.

I know that the build is long, but the thing that checks for bison could check its availability before doing some long stuff :)

marcelstoer, 2020-06-25:
Building the firmware (and the toolchain) is a resource/computation intensive operation. I don't understand why you would want to do that on a resource-constrained device like the Pi. I would want to run it on the fastest machine I have access to.

f4grx, 2020-06-25:
I know.
I'm not complaining about the build speed.
I'm complaining about the check for build-required packages which is not done as early as it could.

BTW, the choice of build machine does not belong to you. The pi is the fastest linux machine I have access to. And it works well, I have no issue with that.

marcelstoer, 2020-06-25:
> The pi is the fastest linux machine I have access to.

I didn't say anything about Linux being a requirement (x86 is, Linux isn't). The popular Docker image enables convenitently building on fast non-Linux systems: https://nodemcu.readthedocs.io/en/latest/build/#docker-image.

TerryE, 2020-06-25:
@f4grx don't worry I use RPis a lot and they are perfectly capable of acting as a build host, especially as the RPi4 + SSD is as fast as a typical 5 year old laptop.  But the issue here is really that @jmattsson provides prebuilt gcc ESP32 and ESP8266 toolchains for x86 64bit hardware, but doesn't do the same for RPi, so most x86-based devs don't have to do the toolchain build.

As you can see from my [above post](#issuecomment-551996180), There are precursor libraries that you need to install and also a few workaround tweaks that you need.  Doing a toolchain build is complicated.  This project assumes that you've got a toolchain prebuilt.  We don't have the resources or bandwidth to offer individual developers help in doing this. Sorry

f4grx, 2020-06-25:
Docker is not really a path I want to explore. I would have to deploy a full docker setup just to build a toolchain? that is not reasonable.
I dont want alternatives, I want to work on a pi because it's easier for me, since I'm not in front of my main machine right now and the pi is the only development machine I can access.

I would prefer you to tell me that building the toolchain on a pi is not supported than to hear inapplicable alternatives.

TerryE, 2020-06-25:
> I would prefer you to tell me that building the toolchain on a pi is not supported ...

As I said, we don't support building any toolchain.  This is down to Espressif, etc. not this project.

slorquet, 2020-07-01:
Not even to espressif.
I have narrowed down the issue to this gmp bug:
https://gmplib.org/list-archives/gmp-bugs/2018-January/004282.html

in fact there is a problem building a 32-bit gmp on a raspberry pi with a 64-bit cpu.
Raspbian only comes in 32-bit version so this cannot be worked around.

This is an old bug... and so, only for raspberry pi 3 and 4.

Someone said that using -g in the build cflags made the problem disappear. I am testing that. No idea if the cflags I specify in the CT .config will get used...

closed 2021-07-21 by stale[bot]