nodemcu/nodemcu-firmware #2962: tmr.softwd() Not working in loop

chathurangawijetunge, 2019-11-14:
odeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: master
	commit: 310faf7fcc9130a296f7f17021d48c6d717f5fb6
	release: 3.0-master_20190907
	release DTS: 201909070945
	SSL: false
	build type: integer
	LFS: 0x20000
	modules: file,gpio,mqtt,net,node,pwm,rtctime,sntp,tmr,wifi
 build 2019-10-25 06:20 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)
 

-----
tmr.softwd(5)
while true do print("runing") tmr.delay(1000000) end

this will not reset the device.... is it a bug..?
is there a way to reset when stuck in continuous loop (stuck)

TerryE, 2019-11-14:
[`tmr.delay()`](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/modules/tmr.c#L122) issues a `system_soft_wdt_feed()` every 1 sec to prevent a timout during a deliberate delay. Don't ask me why.  The logic was laid down in the mists of time and we deprecate the use of the delay anyway as this can crash the network stack.  Use a proper CB timer. 

If you don't feed the timer then the processor will reset.  Try:
```Lua
for i=1,1000000000 do j = 0 end
```
This  will trigger the H/W watchdog timer after ~3s, as you aren't feeding the SW watchdog.

@nwf Nathaniel, IIRC you did some work on the S/W timer. Any comment

nwf, 2019-11-14:
I'm afraid you've got the wrong committer.  The only thing I did with timers was #2603.

TerryE, 2019-11-25:
Anyway, I don't think that this is a bug, per se; it is more an issue of "that's how it was designed to work". I'll leave it for other committers to comment first, but my inclination is to close this.

closed 2020-07-21 by chathurangawijetunge