nodemcu/nodemcu-firmware #2964: uart data callback unregister

seregaxvm, 2019-11-16:
NodeMCU attempts to call `nil` value after callback unregister.
I think that this broke [nodemcu-uploader](https://github.com/kmpm/nodemcu-uploader).

### Expected behavior

No call should be made if callback is unregistered.

### Actual behavior

NodeMCU calls `nil`.

### Test code

Executed from NodeMCU REPL (connected via UART):

```Lua
uart.on("data")
print("Hello")
```

On `master` branch this prints:
```
Hello
```

On `dev` branch this prints:
```
attempt to call a nil value
stack traceback:
Hello
```

### NodeMCU version
`dev` branch f3b49849d2

TerryE, 2019-11-16:
@seregaxvm Thanks again for a lovely piece of detective work.  You keep finding "my bad"s :rofl: 
The root issue is that the actual uart line handling is done by the input driver and the uart module calls `input_setup_receive()` to set this up.  It doesn't handle the clear down case properly.  I need to do a code review and make sure that we cover this path functionally.  I'll post back when I've done this.

 If you want to hook up or discuss anything by way of follow up, then my email addr is my commits in `git log`.  I've attached a log of my session (with a cut-down of the O/P to keep the listing short).  
```
nodeMCU $ # rebuild with debug enabled for the uart module
nodeMCU $ touch app/modules/uart.c
nodeMCU $ make DEBUG=1
make: Entering directory '/home/terry/nodemcu-firmware'
...
LD app/.output/eagle/debug/image/eagle.app.v6.out
NM app/.output/eagle/debug/bin/eagle.app.v6.bin
ESPTOOL app/.output/eagle/debug/image/eagle.app.v6.out ../bin/
esptool.py v2.6
Creating image for ESP8266...
make[1]: Leaving directory '/home/terry/nodemcu-firmware/app'
make: Leaving directory '/home/terry/nodemcu-firmware'
nodeMCU $ # Flash the new image
nodeMCU $ esptool --port /dev/ttyUSB0 --baud 460800 --after no_reset write_flash -fm dio 0x00000 bin/0x00000.bin 0x10000 bin/0x10000.bin;  tools/nodemcu-partition.py -ls 256K -sa 1M -ss 1M
...
  Partition          Start   Size 
  ------------------ ------ ------
  EAGLEROM           000000 00b000
  RF_CAL             00b000 001000
  PHY_DATA           00c000 001000
  SYSTEM_PARAMETER   00d000 003000
  IROM0TEXT          010000 07c000
  LFS0               08c000 040000
  SPIFFS0            100000 100000
...
nodeMCU $ # Reset the ESP and then connect to it
nodeMCU $ miniterm /dev/ttyUSB0  115200
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
...
cannot open init.lua: 
> -- Lines are compiled as a chunk which is then executed so you can use this to break 
> -- and step through a piece of Lua code
> gdbstub.pbrk() return uart.on("data")
$T05#b9
--- exit ---
nodeMCU $ elf-gdb
GNU gdb (crosstool-NG crosstool-ng-1.22.0-60-g37b07f6f) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
...
lgdbstub_pbreak (L=0x3ffeff00) at gdbstub.c:38
38	  asm("break 0,0" ::);
(gdb) bt 5
#0  lgdbstub_pbreak (L=0x3ffeff00) at gdbstub.c:38
#1  0x4024d596 in luaD_precall (L=0x3ffeff00, func=0x3fff01c0, nresults=0) at ldo.c:335
#2  0x4024b372 in luaV_execute (L=0x3ffeff00, nexeccalls=1) at lvm.c:649
#3  0x4024d811 in luaD_call (L=0x3ffeff00, func=0x3fff01b4, nResults=-1) at ldo.c:395
#4  0x40239195 in f_call (L=0x3ffeff00, ud=0x3ffffc4c) at lapi.c:830
(More stack frames follow...)
(gdb) upto l_uart_on
Hardware assisted breakpoint 1 at 0x402542d3: file uart.c, line 25.

Breakpoint 1, l_uart_on (L=0x3ffeff00) at uart.c:25
25	  int stack = 2, data_len = -1;
(gdb) upto 31
Hardware assisted breakpoint 2 at 0x40254312: file uart.c, line 31.

Breakpoint 2, l_uart_on (L=0x3ffeff00) at uart.c:31
31	  if (lua_type( L, stack ) == LUA_TNUMBER)
(gdb) p L->top[-1]
$1 = {value = {gc = 0x3fff0518, p = 0x3fff0518, n = 8.1166307269597102e-313, b = 1073677592}, tt = 4}
(gdb) # This is a TString TValue, whose value is ...
(gdb) p (char*)L->top[-1].value.gc+16
$2 = 0x3fff0528 "data"
(gdb) p stack
$3 = 2
(gdb) p L->base[2]
$4 = {value = {gc = 0x3fff1210, p = 0x3fff1210, n = 5.304688532146986e-315, b = 1073680912}, tt = 0}
(gdb) # In this case the ToS is also pos [1] (need to offset by one for C arrays)
(gdb) p L->base[0]
$8 = {value = {gc = 0x3fff0518, p = 0x3fff0518, n = 8.1166307269597102e-313, b = 1073677592}, tt = 4}
(gdb) upto 57
Hardware assisted breakpoint 3 at 0x402543d8: file uart.c, line 57.

Breakpoint 3, l_uart_on (L=0x3ffeff00) at uart.c:57
57	    luaL_unref(L, LUA_REGISTRYINDEX, uart_receive_rf);
(gdb) l
52	    lua_pushvalue(L, stack);
53	    luaL_unref(L, LUA_REGISTRYINDEX, uart_receive_rf);
54	    uart_receive_rf = luaL_ref(L, LUA_REGISTRYINDEX);
55	
56	  } else {
57	    luaL_unref(L, LUA_REGISTRYINDEX, uart_receive_rf);
58	    uart_receive_rf = LUA_NOREF;
59	  }
60	  input_setup_receive(uart_on_data_cb, data_len, end_char, run_input);
61	  return 0;
(gdb) upto 60
Hardware assisted breakpoint 4 at 0x402543ee: file uart.c, line 60.

Breakpoint 4, l_uart_on (L=0x3ffeff00) at uart.c:60
60	  input_setup_receive(uart_on_data_cb, data_len, end_char, run_input);
(gdb) p uart_on_data_cb
$10 = {void (const char *, size_t)} 0x40254278 <uart_on_data_cb>
(gdb) # OK, the CB is set up correctly BUT 
(gdb) p uart_receive_rf
$11 = -2
(gdb) # is LUA_NOREF
(gdb) # the root BUG is that we should properly close down the setup receive.
```

TerryE, 2019-11-16:
OK, the ESP hardware uses a 128-byte FIFO to buffer in Rx byte data.  This is flushed by the UART driver operating as an ISR to a secondary 256byte FIFO in `UartDev` in the RAM allocated to the bot ROM.  As this starts to fill a high priority SDK task is posted (`input_readline()`) which empties the realtime buffer into a 3rd fifo.  (The reasons are lost in history).   How `input_readline()` processes this Rx byte stream depends on four internal state parameters: `uart_cb`, `data_len`, `end_char` and `run_input`, but broadly it runs in one of two modes depending on `run_input`.
-  If `true` then complete lines are passed to the Lua interpreter using `lua_input_string()` which queues them for  processing in the Lua input pipe.
-  If `false` they are passed to the application CB according to the stop conditions.

The reason for the input pipe is that in SDK.2.x versions or prior,  pasting text into the Lua interpreter would overrun if more than 128 bytes or so where pasted in one operation.  You can now paste in dozens of lines and the pipe handles these without drop.

These four input state config variables can be modified by `input_setup_receive()`.  We should restore them to the Lua interpreter defaults when the `uart.on('data')` is reset by calling `input_setup_receive(NULL, 0, 0, 1)`

Having just tried this with ESPlorer, this is still hanging.  There is some edge problem in the on / off data handshake with the current version that I need to test further.

TerryE, 2019-11-16:
@seregaxvm if you don't mind: since I have found this specific bug and put a fix in my current PR, the remaining issues are covered in the overlapping #2963, I will close this one and continue the discussions on the other thread.  You input will be very welcome.

closed 2019-11-16 by TerryE

seregaxvm, 2019-11-16:
Sure thing.

TerryE, 2019-11-16:
OK, I picked up two bug fixes so far.  This first is that we are leaving `end_char` uninitialised on the fixed length `uart.on()` code path and the second that I was not clearing down the input driver state correctly.  See below.  This being said, at the moment `nodemcu-tool` works but the binary upload in `ESPlorer` and `nodemcu-uploader` does not. :unamused:  Creating a test harness here is a total PITA, since the `gdbstub` takes over UART Rx on initialisation, so I can't do any Rx diagnostics whilst in remote `gdb`.

Clearly fixing the buffering and marshalling issues for interactive and telnet use has broken some implicit assumption used by `ESPlorer` and `nodemcu-uploader`.  What I propose doing is getting a stripped-down version of the  `nodemcu-uploader` Python code as a test harness and debug that to find the failure mode -- which will hopefully be common to both these uploaders.  Once I have a robust binary uart uploader, I can put in fixes into these two code bases with minimal support of their authors, if needed.  Sorry guys.
 
```patch
index 632348b0..8ab5cdb1 100644
--- a/app/modules/uart.c
+++ b/app/modules/uart.c
@@ -31,6 +31,7 @@ static int l_uart_on( lua_State* L )
   if (lua_type( L, stack ) == LUA_TNUMBER)
   {
     data_len = luaL_checkinteger( L, stack );
+    end_char = (int16_t) ~0;    
     luaL_argcheck(L, data_len >= 0 && data_len < LUA_MAXINPUT, stack, "wrong arg range");
     stack++;
   }
@@ -57,7 +58,10 @@ static int l_uart_on( lua_State* L )
     luaL_unref(L, LUA_REGISTRYINDEX, uart_receive_rf);
     uart_receive_rf = LUA_NOREF;
   }
-  input_setup_receive(uart_on_data_cb, data_len, end_char, run_input);
+  if (uart_receive_rf == LUA_NOREF)
+    input_setup_receive(NULL, 0, 0, 1);
+  else
+    input_setup_receive(uart_on_data_cb, data_len, end_char, run_input);
   return 0;
 }
```

seregaxvm, 2019-11-17:
@TerryE 
> This being said, at the moment nodemcu-tool works but the binary upload in ESPlorer and nodemcu-uploader does not.

I tried to upload `tar.gz` archive using upstream version of `nodemcu-uploader` tool with python3 and it worked (after applying patch).