nodemcu/nodemcu-firmware #2965: hx711 read takes 430ms

mamert, 2019-11-19:
Data reading is very slow

### Expected behavior
read() takes 0-100ms
Same module connected to Arduino (ATmega328P), using bogde/HX711 library, consistently reads in 95-96ms (0-1ms if we sleep(100))

### Actual behavior
A single call to hx711.read() takes 430ms
Below code, in a loop,  produces values: 430088, 430097, 430113, 430228...
Running 10 reads immediately one after the other takes ~4302ms

### Test code
```Lua
hx711.init(2, 1) -- also tried with 7, 6
while true do
	local start = tmr.now()
	hx711.read() -- values are correct, but irrelevant here
	print(tmr.now() - start)
end
```
### NodeMCU version
NodeMCU 3.0.0.0 built on nodemcu-build.com provided by frightanic.com
	branch: master
	commit: 310faf7fcc9130a296f7f17021d48c6d717f5fb6
	release: 3.0-master_20190907
	release DTS: 201909070945
	SSL: false
	build type: float
	LFS: 0x0
	modules: adc,bit,file,gpio,hx711,i2c,net,node,pcm,pwm,sigma_delta,sjson,tmr,uart,websocket,wifi,ws2812
 build 2019-11-18 23:43 powered by Lua 5.1.4 on SDK 3.0.1-dev(fce080e)

### Hardware
black WeMos board, or a NodeMCU V1 board, 4MB, DIO Flash mode, powered via USB
hx711 connected to 2(GPIO4), 1(GPIO5), GND and 5V

pjsg, 2019-11-19:
There is a PR (which I just fixed) #2915 which helps this. It allows you to run the HX711 in streaming mode and has a callback whenever the requisite number of readings have been made. It might help solve your problem....

mamert, 2019-11-20:
@pjsg Neat, thanks :)

Should I leave this issue open until then?

closed 2019-12-09 by marcelstoer