nodemcu/nodemcu-firmware #2966: u8g2/ucg: allow to get list of available fonts

marcelstoer, 2019-11-20:
### Missing feature
u8g2/ucg don't tell me which fonts are available i.e. which ones I compiled into the firmware. Something like `getAvailableFonts()` would be handy.

### Justification
It's easy to forget which fonts you selected at the time you built the firmware. If you built it yourself (rather than using the cloud builder) you maybe luck in still find the relevant `.h` files - and hope you didn't tinker with them in the meantime.

### Workarounds
None - at least not unless you build the firmware yourself.

@devsaurus is this feasible?

Michal78S, 2020-01-16:
I dig into hex file for _Font_. It isn't wery comfortable, but works.
[image]

HHHartmann, 2021-01-10:
Still seems to be a good feature to me.

closed 2022-04-30 by stale[bot]