nodemcu/nodemcu-firmware #2968: Add WebServer on top of httpserver for authentication handling of several routes

HHHartmann, 2019-11-22:
### Missing feature
There have been several calls for a WebServer lately.
Reference to our httpserver.lua or https://github.com/marcoskirsch/nodemcu-httpserver or #2957 

Our httpserver.lua is sophisticated in using output buffering and having a flexible interface.
But it does not do anything out of the box, so I aim to have a WebServer which can be configured to serve static files with one line. Equally simple it can be configured to call dynamic APIs.

It will be easy to write and integrate individual filters, authentication, redirects or whatever middle-ware is needed.

Configuration will be similar to nodemcu-espress. I might also use their plugins and adapt them to our API.

### Justification
Easy way to serve some pages.
@TerryE @marcelstoer @pjsg Do you think this is the right repo or should I put it in my own.
This would include several files for the core WebServer and some plugins.

### Workarounds
Everybody write their own

marcelstoer, 2019-11-22:
Amazing (or scary?) how many HTTP servers there exist for NodeMCU - I had no idea. This despite [our FAQ stating](https://nodemcu.readthedocs.io/en/latest/lua-developer-faq/#how-do-i-minimise-the-footprint-of-an-application) 

> If you are trying to implement a user-interface or HTTP webserver in your ESP8266 then you are really abusing its intended purpose.

I don't have any preference as to where you are going with this code. Not having to maintain an HTTP module at all would also be fine i.e. we remove ours and link to yours instead.

NiklausS, 2019-11-23:
@Gregor,

.But it does not do anything out of the box, so I aim to have a WebServer which can be configured to serve static files with one line. Equally simple it can be configured to call dynamic APIs.

This was the idea when i wrote this.
https://github.com/NiklausS/nodemcu-firmware/blob/Webserver/docs/modules/httpd.md

I've also done some simple test with authentication. But as it is not secure, i didn't follow this further.

HHHartmann, 2020-11-20:
I will eventually continue my work. Please have a bit more patience.

HHHartmann, 2022-04-18:
Need to make a pr for this