nodemcu/nodemcu-firmware #2972: PANIC: unprotected error in call to Lua API (attempt to concatenate a nil value)

o07sai, 2019-11-29:
There is a timer, period is 5 sec.output cmd: 2 	{"cmd":"null"}
I send "a=" to the console, then send "1" after esp32 output cmd:	2	{"cmd":"null"}.
I got a panic.

Log:
cmd:	2	{"cmd":"null"}
cmd:	2	{"cmd":"null"}
a=
>> cmd:	2	{"cmd":"null"}
1
PANIC: unprotected error in call to Lua API (attempt to concatenate a nil value)
[0;31mE (477036) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:[0m
[0;31mE (477036) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (477036) task_wdt: Tasks currently running:[0m
[0;31mE (477036) task_wdt: CPU 0: main[0m
[0;31mE (477036) task_wdt: CPU 1: IDLE1[0m
[0;31mE (477036) task_wdt: Aborting.[0m
abort() was called at PC 0x400d27c7 on core 0

o07sai, 2019-11-29:
There is a timer, period is 5 sec. make a http.post  requset,got response and output cmd: 2 {"cmd":"null"}
I send "a=" to the console, then send "1" after esp32 output cmd: 2 {"cmd":"null"}.
I got a panic.

Log:
cmd: 2 {"cmd":"null"}
cmd: 2 {"cmd":"null"}
a=

cmd: 2 {"cmd":"null"}
1
PANIC: unprotected error in call to Lua API (attempt to concatenate a nil value)
�[0;31mE (477036) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:�[0m
�[0;31mE (477036) task_wdt: - IDLE0 (CPU 0)�[0m
�[0;31mE (477036) task_wdt: Tasks currently running:�[0m
�[0;31mE (477036) task_wdt: CPU 0: main�[0m
�[0;31mE (477036) task_wdt: CPU 1: IDLE1�[0m
�[0;31mE (477036) task_wdt: Aborting.�[0m
abort() was called at PC 0x400d27c7 on core 0

TerryE, 2019-12-01:
@o07sai, I assume that this is ESP32-specific.  If you can't present this as a bugrep in the correct format with the required content and conforming to #1010 then unfortunately I will have to close this issue.

closed 2019-12-04 by marcelstoer