nodemcu/nodemcu-firmware #2977: SmartConfig is disabled by default and can be enabled by setting WIFI_SMART_ENABLE in user_config.h before you build the firmware

momkey02, 2019-12-08:
wifi.c:20:31: fatal error: smart/smartconfig.h: No such file or directory
 #include "smart/smartconfig.h"
                               ^
compilation terminated.
../../Makefile:422: recipe for target '.output/eagle/debug/obj/wifi.o' failed
make[2]: *** [.output/eagle/debug/obj/wifi.o] Error 1
make[2]: Leaving directory '/opt/nodemcu-firmware/app/modules'
../Makefile:380: recipe for target '.subdirs' failed
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory '/opt/nodemcu-firmware/app'
Makefile:380: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
make: Leaving directory '/opt/nodemcu-firmware'

marcelstoer, 2019-12-08:
That works as documented and designed: https://nodemcu.readthedocs.io/en/latest/modules/wifi/#wifistartsmart

closed 2019-12-08 by marcelstoer

HHHartmann, 2019-12-09:
@momkey02 what branch are you building on? 
Did you do a clean build?

z1521183003, 2020-02-13:
> @momkey02 what branch are you building on?
> Did you do a clean build?

I also encountered the same problem,

wifi.c:20:31: fatal error: smart/smartconfig.h: No such file or directory
 #include "smart/smartconfig.h"
                               ^
compilation terminated.
../../Makefile:422: recipe for target '.output/eagle/debug/obj/wifi.o' failed
make[2]: *** [.output/eagle/debug/obj/wifi.o] Error 1
make[2]: Leaving directory '/home/zjh/Desktop/nodemcu-firmware/app/modules'
../Makefile:380: recipe for target '.subdirs' failed
make[1]: *** [.subdirs] Error 2
make[1]: Leaving directory '/home/zjh/Desktop/nodemcu-firmware/app'
Makefile:380: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
make: Leaving directory '/home/zjh/Desktop/nodemcu-firmware'